ant-thomas/zsgx1hacks #5: Hi! Great project!

laurileemet, 2017-11-05:
Hi! I also bought 2 of these cameras ZS - GX1 1080P WiFi IP Camera Webcam from GearBest (also known as GUUDGO GD-SC03 Snowman 1080P WIFI) .
 I have tested one camera and seems to be good quality for the money. I have also tried to find out root password for telnet and tried to sniff packets with WireShark. Only thing I discovered was that "eyeplusiot.com" equals to "icloseli.com". 

Fortunately my camera already had rtsp://admin:20160404@cameraip/onvif1 active.
Thanks for sharing your work!

ant-thomas, 2017-11-07:
That's interesting, because that username/password combo doesn't work with my camera.
Only admin/admin.
Is it possible that you have a different firmware. Do you have the ability to dump the full firmware and upload it?

kkm0614, 2017-11-12:
Hey guys!

I ordered a Gearbest camera: ZS - GX1 1080P WiFi IP Camera Webcam.
This can only be used with eyepluspilot.com or by phone. I would like to use it as described, but I do not have the knowledge to install it. Please make a "how-made" video for users, which everyone uses for their own responsibility.
I'm sorry about English, I'm writing about Hungary.

All my respect, Mark!

kotfarepra, 2017-11-17:
Looking at that password someone could create a bruteforce script to test all the password combinations for 2016-2017?

Presuming the username would always be admin it would be a combination of 
2 years X 365 days = 730 combinations?

Nothing complex for today computers.. 

Am I wrong?

sanscorp1, 2017-11-17:
Try admin without pass.

ant-thomas, 2017-11-18:
Just tried to brute force the telnet with username ```root``` and passwords from ```20160101``` to ```20171231```
It failed.

kotfarepra, 2017-11-19:
Good job 
The example above though has the username as admin 
Can you try with that username ?

ant-thomas, 2017-11-19:
The telnet login will only use ```root``` as the username.
admin is for RTSP/onvif

laurileemet, 2017-11-20:
Sorry for not getting all the info out. rtsp://admin:20160404@cameraip/onvif1 address I found on the net and it still asks for password. admin / admin works for my camera. I did not know that 20160404 should be password :)

closed 2018-02-13 by ant-thomas