ant-thomas/zsgx1hacks #6: Help my camera! :)

kkm0614, 2017-11-17:
Hey guys!

I ordered a Gearbest camera: ZS - GX1 1080P WiFi IP Camera Webcam.
This can only be used with eyepluspilot.com or by phone. I would like to use it as described, but I do not have the knowledge to install it. Please make a "how-made" video for users, which everyone uses for their own responsibility. I tried Telnet, but I do not know the password.
Features:
Camera Version
Firmware: 3.1.1.0908
Embedded Application: 3.1.1.0908
Device ID: ******************
MAC Address: ******
Model Type: CloudCamera
Anti Flicker: 60Hz

I'm sorry about English, I'm writing about Hungary.

All my respect, Mark!

kkm0614, 2017-11-17:
Hello!

It works : rtsp://admin:admin@<camera_ip_address>:8001/onvif1

kotfarepra, 2017-11-17:
What firmware version  is your camera?

kkm0614, 2017-11-17:
ZS - GX1 1080P
Firmware: 3.1.1.0908

kotfarepra, 2017-11-17:
Can you confirm it works without internet from your modem?

kkm0614, 2017-11-17:
Network can be used without the use of the internet. (No internet=Network it works) :D

kkm0614, 2017-11-17:
the ispy program can not use the camera. it would be nice if PTZ were to go too.

kotfarepra, 2017-11-17:
Great, dont update the firmware. I heard that the new version doesn't has
access

Regards

sanscorp1, 2017-11-17:
Rtsp / onvif 1.2 are enabled by default on the mentioned firmware. Onvif device manager (sourceforge) can access the livestream and ptz.

Today I recieved a usb to ttl interface, I'm going to rip the filesystem from a tenvis camera with embedded web interface.

I also replied under the initial readme here, but didn't know if it was the right place. I have no experience with hacking or linux, just trying stuff.

kkm0614, 2017-11-17:
Ispy opensource program work. The ptz option is still in question.

ant-thomas, 2017-11-17:
I can confirm that PTZ does work via onvif!

I tried it using the tinyCam Free app on Android

ant-thomas, 2017-11-19:
Updated instructions

closed 2018-02-13 by ant-thomas