ant-thomas/zsgx1hacks #8: 2017-11-17 - Update 9

Kev0123, 2017-11-18:
How to access these settings without previously connectiong cam to network?

Is this only possible via serial?

ant-thomas, 2017-11-18:
I will hopefully make some updates to the debug_cmd.sh script and improve the instructions today

ant-thomas, 2017-11-19:
Updated instructions

felixsteghofer, 2018-01-16:
I have a somewhat similar cam, DIGOO DG-M1X 960P.
I was able skip the inital app setup by telneting in and setting the creds in `/rom/wpa_supplicant0.conf`. I'm pretty sure you already tried that but it took me an hour or so to finally find the _persistent_ mnt point (e.g. `/etc/wpa_supplicant0.conf` was not).

As the cams, this repo is targeted at, seems to run a (much) different version of goke this might me irrelevant…