ant-thomas/zsgx1hacks #9: How to extract firmware ?

kotfarepra, 2017-11-19:
Tried to extract this firmware via binwalk but it doesnt recognize any partition/formats 

http://files.banggood.com/2016/06/GUUDGO%20ONVIF.rar

I see the header PK2 which seems compressed kinda like ZIP 

Any idea ?

kotfarepra, 2017-11-19:
I got the firmware extracted using this 

https://github.com/karcaw/ipr3tools

Got 

`[root@localhost Desktop]# ./tenvis_pack  -u -f  firmware.bin
TENVIS firmware unpacker V0.1 - leecher@dose.0wnz.at

Firmware created: 2017-08-24 15:03:51
Camera type: 1003
Version: 0.0.0.0
Sections: 8
CMD rm -f /home/p2pcam*.t*
Processing file /home/p2pcam.tar.gz
Processing file /home/tees
Processing file /home/wpa_supplicant
Processing file /home/start.sh
Processing file /home/cloud.ini
Processing file /home/cloud_oversea.ini
CMD chmod -R 777 /home/
All sections processed.
`

sanscorp1, 2017-11-19:
I tried zipping all folders and transferring them to the sd card. Got a lot but not all...

Are you able to zip the unpacked firmware and upload it?

kotfarepra, 2017-11-19:
I dont have the camera itself :)
Its still at post office

sanscorp1, 2017-11-19:
But you managed to unpack the fw, right? This is only an update but perhaps with my files and yours we have a complete package of files.

kotfarepra, 2017-11-19:
My bad , the tool i used is this 
http://dose.0wnz.at/ipcams/tenvis_pack.c

kotfarepra, 2017-11-19:
Did a test extracting with 

```./tenvis_pack -u -f firmware.bin -d out -c kontrollfile```

And then compressing the same output directory with 

``` 
./tenvis_pack -p -f firmware2.bin  -d out  -c kontrollfile
```

The new/compressed file firmware2.bin is the same in size as the original . 

```
-rw-rw-r-- 1 asus asus 3334410 Nov 19 19:53 firmware2.bin
-rw-r--r-- 1 asus asus 3334410 Nov 19 19:15 firmware.bin
```

Doing a file comparison everything is the same aside this   http://prntscr.com/hcjaq6

Eventually the md5sum is different 

```
5306cdeb7927dbf63b8af07cf5156d25  firmware.bin
067422f5f85b29b213120606307af958  firmware2.bin
```

leecher1337, 2018-02-15:
Sure, this DWORD is different as this is creation timestamp 
uint32_t tCreation;