ant-thomas/zsgx1hacks #12: Permanent Hack - without SD

ant-thomas, 2017-11-26:
Working on trying to figure out a permanent hack without using an SD card.

Currently got one working where you only need to use an SD card once. The hack is copied across and the SD card can be removed once the hack is applied. 

This is possible because everything in ```/home``` is persistent across reboots. Meaning you can use ```debug_cmd.sh``` to copy a new ```start.sh``` file to ```/home``` and copy all the required files (dropbear, busybox, hosts etc) to a folder on ```/home```.

Within a new ```start.sh``` the commands are added to apply the hack on each boot. No SD required.

deorder, 2017-11-27:
I already modified my cams without the use of an sdcard. The following ```start.sh```:

[start.tar.gz](https://github.com/ant-thomas/zsgx1hacks/files/1504372/start.tar.gz)

It boots normally when there is a sdcard inside the slot and runs the ```debug_cmd.sh``` if available. If there is no sdcard in the slot it will continue with ```/home/boot.sh```.

ant-thomas, 2017-12-02:
Thanks for that.

Does ```wpa_supplicant``` actually work on an unconfigured camera?

deorder, 2017-12-03:
It works but as soon as ```p2pcam``` starts running it will connect to WiFi as specified in ```devParam.dat```. Still have to figure out what checksum algorithm is being used so ```p2pcam```  will accept it. While testing I remove ```p2pcam``` from the ```boot.sh``` and add a endless loop containing a sleep at the end of the script.

int2018, 2018-01-13:
So far it is working when things are available on the SD card. But things do not survive the reboot, even when I put everything in /home. 
Anyway, now the device seems messed up, it does not finish the boot sequence.  It sounds as if it tries to connect again and again. Any hints? Downgrade firmware?