ant-thomas/zsgx1hacks #15: Does this works with GUUDGO GD-SC02

kotfarepra, 2017-12-02:
Title says it all 
Does this works for GUUDGO GD-SC02 https://www.banggood.com/GUUDGO-GD-SC02-720P-Wifi-IP-Camera-PanTilt-IR-Cut-Night-Vision-Two-way-Audio-Motion-Detection-Alarm-p-1176850.html

thanks in advance

kotfarepra, 2017-12-02:
Tested a bit the following and they works 
1- opening stream with vlc rtsp://admin:admin@IP_OF_CAMERA:8001/onvif1
2- telnet works with root/cxlinux . Files seems the same as on this github we are 
3- tinycam for android works 
4- onvif device manager ( sourceforge ) works and it shows a username administrator with password 12345 ( dont know where thats used ) 

Regards

shark1981, 2018-01-09:
Hello!
I am the owner of the camera GUUDGO GD-SC02, after 1 month of good work, the camera does not connect to the router using cable or wi-fi.
Help me please. How do I save the current firmware 3.2.0.1118?
How do I reset the camera to factory defaults or flash another working firmware?

DLCDavid, 2018-08-07:
Howdy, first time here leaving comment/question.  I too have the GUUDGO GD-SC02.  I have tried to install the firmware found in readonlyhack-v0.1.zip since I have the 3.2.8.0121 fw.  For the likes of me, I can not get the cam to load that firmware.  It just performs a reset and I'm back to the GUUDGO gui on android and in Chrome on W7.  What am I doing wrong?  I think there are many steps missing from your instructions found in the README.md file.

SyM3N, 2019-02-03:
Hey Folks, 

as of today, i can confirm the SC02 with Firmware **3.1.1.0908** working with [v0.4](https://github.com/ant-thomas/zsgx1hacks/raw/master/zsgx1hacks-v0.4.zip)