ant-thomas/zsgx1hacks #16: Remove rtsp authentication

kotfarepra, 2017-12-02:
hello 
Trying to set up a server with ffmpeg to convert the RTSP of the camera to hls . 
Ffmpeg doesnt like how the authentication is setup on the RTSP server of camera.

Is there a way to make the rtsp server accessible without user/pass ?

meaning rtsp://192.168.1.15:8554 instead of rtsp://admin:admin@192.168.1.15:8554 ?

Thanks

nestoroa, 2017-12-02:
Hi, I take screenshots from the RTSP with this shell script:
`ffmpeg -i "rtsp://admin:admin@192.168.0.241:554" -y -f image2  -frames 1 /home/USER/snapshot.jpeg`
Maybe it will help you.

kotfarepra, 2017-12-02:
I tried that too but still get the same error 

Using admin/admin
[rtsp @ 0x16e0570] method DESCRIBE failed: 401 Unauthorized
rtsp://admin:admin@192.168.1.115:554: Server returned 401 Unauthorized (authorization failed)

Using just username admin and blank password
[rtsp @ 0x16e0570] method DESCRIBE failed: 401 Unauthorized
rtsp://admin@192.168.1.115:554: Server returned 401 Unauthorized (authorization failed)

kotfarepra, 2017-12-02:
Even tried with ffprobe and still same problem

ffprobe rtsp://admin@192.168.1.15:554

nestoroa, 2017-12-02:
try rtsp://admin:@192.168.1.15:554

nestoroa, 2017-12-02:
Also share here your firmware version

kotfarepra, 2017-12-02:
That last command did it :)

Thanks

ant-thomas, 2017-12-02:
Glad to see you've sorted it. On a camera that hasn't been configured by the app there is no password, just a username of ```admin```

kotfarepra, 2017-12-02:
Yeah , i did the configuration via the app and was just playing with the options ( telnet ) that are available without flashing the camera with the "hack" file on this github repo. 

Thought since the others ( vlc , live555 ) were connecting OK , then even FFMPEG would be ok .... but hey you learn something new everyday :)

Thanks once again

kotfarepra, 2017-12-02:
By the way , referring to  what you said above 

i did the configuration via the app and it was still without password ( just username admin ) for RTSP 

My camera is "GUUDGO GD-SC02" and the infos from the APP are as below 

model type : 
eyeplus_ipc_gk_002

Firmware : 
3.1.1.0908

Embedded Application
3.1.1.0908

closed 2017-12-03 by kotfarepra