ant-thomas/zsgx1hacks #19: PTZ control (rotation control)

jimdaft, 2017-12-04:
I m starting to work on how the camera rotation is controlled. Any ideas are welcome.

kotfarepra, 2017-12-04:
Good job
I played a bit with this myself https://github.com/futomi/node-onvif
It has also PTZ controls

jimdaft, 2017-12-04:
This cam does not support Onvif

kotfarepra, 2017-12-04:
Maybe my bad but i tested that script and i was able to get the information from camera.

kotfarepra, 2017-12-05:
This is the results using that nodejs library. Whats strange though is that the PTZ min/max are null but like @ant-thomas  said it works with many PTZ apps for android and also the usual[ ONVIF DEVICE MANAGER ](https://sourceforge.net/projects/onvifdm/)

```
[nodemon] starting `node camera.js`
The OnvifDevice object has been initialized successfully.
{
  "Manufacturer": "EYEPLUS",
  "Model": "EYEPLUS_DEV",
  "FirmwareVersion": "3.1.1.0908",
  "SerialNumber": "12345679890",
  "HardwareId": "88"
}
{
  "token": "Profile_2",
  "name": "subStream",
  "snapshot": "http://192.168.66.102/onvif/snapshot",
  "stream": {
    "udp": "rtsp://192.168.66.102:554/0/av1",
    "http": "rtsp://192.168.66.102:80/0/av1",
    "rtsp": "rtsp://192.168.66.102:554/0/av1"
  },
  "video": {
    "source": {
      "token": "VideoSourceToken",
      "name": "VideoSourceConfig",
      "bounds": {
        "width": 1280,
        "height": 720,
        "x": 0,
        "y": 0
      }
    },
    "encoder": {
      "token": "VideoEncoderToken_2",
      "name": "VideoEncoder_2",
      "resolution": {
        "width": 640,
        "height": 320
      },
      "quality": 10,
      "framerate": 12,
      "bitrate": 0,
      "encoding": "H264"
    }
  },
  "audio": {
    "source": {
      "token": "AudioSourceConfigToken",
      "name": "AudioSourceConfig"
    },
    "encoder": null
  },
  "ptz": {
    "range": {
      "x": {
        "min": null,
        "max": null
      },
      "y": {
        "min": null,
        "max": null
      },
      "z": {
        "min": null,
        "max": null
      }
    }
  }
}
```

kotfarepra, 2017-12-05:
Playing a bit with ONVIF DEVICE MANAGER and doing a bit of wireshark capture i was able to replicate the PTZ by sending a POST ( via POSTMAN )  to the camera and it moved  [https://prnt.sc/hjf44a](https://prnt.sc/hjf44a)

Here the code for the nodejs library i mentioned above 

I had to change the XADDR to the one you see on the code below and it worked

```
const onvif = require('node-onvif');

// NOTICE THE URL which is different from the examples
let device = new onvif.OnvifDevice({
  xaddr: 'http://192.168.66.101/onvif/PTZ',
  user : 'admin',
  pass : '12345'
});

device.init().then(() => {
  // Move the camera
  return device.ptzMove({
    'speed': {
      x: 1.0, 
      y: 0.0,
      z: 0.0
    },
    'timeout': 1 // seconds
  });
}).then(() => {
  console.log('Done!');
}).catch((error) => {
  console.error(error);
});
```

jimdaft, 2017-12-05:
Good Job. I was doing same thing yesterday and I confirm that with ONVIF DEVICE MANAGE it works (onvifer app didn't recognize it as onvif device).

DrakoZ900, 2018-02-05:
With IP Cam for Android get the picture very nice, but the PTZ dosent work.

:(

erwill2, 2018-02-10:
@DrakoZ900 

Make sure your connected to ONVIF port 80. If you connect directly to the stream via rtsp 554 you wont get PTZ control.

gokhansaroglu, 2018-02-27:
Using this project https://github.com/futomi/node-onvif I have managed to run PTZ using onvif :) Sample project written very clearly. Only case is server application couldn't get a snapshot from the camera. Whenever I disable "this.fetchSnapshotCallback(data);" function from the client (sample/html/onvif.js),  you can move the camera using keyboard direction buttons :) Only thing is library can't get a snapshot from camera. Right now I'm working on figuring out why server can't get a snapshot from the live stream. Somehow cameras are not giving proper response and client-server socket connection dropping :(

gokhansaroglu, 2018-02-27:
https://github.com/ant-thomas/zsgx1hacks/issues/33