ant-thomas/zsgx1hacks #22: Just a note

kotfarepra, 2017-12-07:
Tried the files on the zip on my camera and they work ok . 

The only "issue" is that the card needs to be there all the time . 

Thought it would be an "update file" but the camera is using the old files as soon as i reboot without sdCard.

So maybe just add a note on the readme.md that the sdcard needs to be on camera always for the "hack" to work 

Thanks again for great work

ant-thomas, 2017-12-07:
I'll mention it in the readme.

I have worked on a script that doesn't need an SD card so I'll release that soon.

lobmarib, 2017-12-19:
Hi guys,

To avoid create a new issue I'll reuse this one.

First, congrats for the great work done in here.

Now, the question/comment itself. Can you update the readme file? It seems there're things in the TODO list that are done.

I'll start using this and try to contribute to improve it.

Cheers

closed 2018-02-13 by ant-thomas