ant-thomas/zsgx1hacks #23: Works on GD-SC01

davidlb, 2018-01-02:
Just a short message to tell you, it's also working on GD-SC01 outdoor camera.
The RTSP default password is just different: dg20160404

I'm now working on the wifi part.

Thanks for your good job!

saintee, 2018-01-22:
Hi David -
I have tried on my GD-SC01 with no joy :-( 

I suspect that the debug_cmd.sh is running.  I am on firmware 1.3.1.0510 I wonder if that is the issue ?

Are you able to telnet with root/cxlinux ?

Thanks Sandy

davidlb, 2018-01-23:
Hi Sandy, mine is running firmware 22.x.x.x. It seems there is different hardware for the same model.. :(
Ssh and telnet are working on mine with root/cxlinux

AndyDHill, 2018-03-05:
I used the following 

rtsp://192.168.1.68:554/0/av1

rtsp://192.168.1.68:554/onvif1

with username  : admin
pass : gd20160404

bkmill, 2018-04-13:
Can't get my GD-SC01 working either. Also on firmware 1.3.1.0510

sasadara, 2018-05-17:
Works for me also.

saintee, 2018-07-19:
I got the firmware from the Guudgo app - preferences - device details

saintee, 2018-07-19:
I can telnet in  with root and 1234 as the password

saintee, 2018-07-19:
and get a still on http://ip:8001/snapshot

wexman, 2018-09-15:
How does one find out if the hack worked or not? I'm on firmware 3.2.8.0208 and I can telnet with root/cxlinux, but all the other features don't seem to work (ssh, onvif server, webui,...)
(I'm not a linux guru btw ;-)