ant-thomas/zsgx1hacks #26: Works on Digoo DG-W01F Camera

kaotix, 2018-01-23:
Not an issue, just tested on the Digoo DG-W01F camera and these scripts work.
It appears to run damn near identical code to the cameras you've been discussing so I thought I'd test it as I've been doing a bit of hacking of my own.

Scripts run fine. I think even the wifi comes up but I haven't fully tested that yet it's currently wired for testing.
I had looked at compiling the snx_rtsp_server for this camera so I'm going to test that today as an alternative to the p2pcam RTSP server.

Link for this camera is https://www.banggood.com/Digoo-DG-WO1F-Cloud-Storage-3_6mm-Lens-720P-Waterproof-Outdoor-WIFI-Security-IP-Camera-Motion-Detect-p-1194795.html

kaotix, 2018-02-11:
Rtsp fully working. It was just that it relies on the inbuilt p2pcam binaries for rtsp and a proper rtsp binary would be more suited.

Briggsbfc, 2018-03-15:
Hi Kaotix, what id/password combination did you use for the rtsp server?

kaotix, 2018-03-15:
admin
dg20160404

hope that works.

tightwebz, 2018-03-22:
after i enter the url and user and password the page is down or wont load. does it work for you guys??

robwood9, 2018-03-30:
Hi, very new to this, I’m trying to connect to my w01f, using Ipsy, not really getting anywhere, can anyone help with the url, type it up on here so I can confirm what I’m doing is correct? Would be much apreciated

kaotix, 2018-03-31:
I use blueiris for my cameras so setup might be a bit different.
URL is http://192.168.0.1/onvif1
Obviously replace the IP with what ever your camera is on.

Alternatively (since blueiris asks for username and password) you can use http://admin:dg20160404@192.168.0.1/onvif1
if that doesnt work, use rtsp:// instead of http://

robwood9, 2018-04-01:
Thats great thanks, i was doing it right jjst hadnt started at the begining, then found the readme section! All made sence from there!!

davidlb, 2018-05-15:
I just received my DG-W01F and I confirm it works well on it!
Good job :)

spolenak, 2018-05-20:
Hi guys,
does anybody know how to detect motion detected events from camera on remote system like Openhab2 ? I have digoo dg-f01f camera which works great and it's really easy to play rtsp on iOS, Android, Linux and Windows.

megagenio, 2018-05-21:
Hi all,
I try script on my dg-w01f. 
I copied the contents of the sdcard folder into a sd fat32.
Inserted the sd in the cam, turned off and on the cam.
After booting the camera is on the network at the address I assigned it from the router, but it does not function as ssh or telnet.
Where am I wrong?
Thks.

raf1que, 2018-06-10:
@megagenio - i think i have the same problem as you, were you able to get this working?

raf1que, 2018-06-10:
now working, just needed to reformat sdcard

megagenio, 2018-06-12:
@raf1que - doesn't work for me. 
Can you tell me what kind of formatting did you use?
What did you copy in the SD? the only content of the file zsgx1hacks-v0.4.zip? 
Did you create a directory in the SD or did you put the zip file in the root od SD itself?
How did you reboot camera? Simply with turn OFF and ON camera power?

davidlb, 2018-06-12:
Format as Fat32, extract zsgx1hacks-v0.4.zip directly on the root of the SDcard and that's all.
Insert SDCard, plug the camera.

EdKo66, 2018-06-12:
Will this work on W02f too?

jcobreros, 2018-06-27:
I'm also trying to get motion detection through ONVIF to work. Any idea where to start looking? Is that p2pcam responsibility?

ant-thomas, 2018-08-12:
Make sure you've got tx and rx the correct way round.
Baud is usually 115200
3.3v isn't usually needed, just rx, tx, gnd.

Briggsbfc, 2018-08-16:
I have Version 3.1.1.0908 on a Digoo DG-W01F.
I'm trying to update to Version 3.2.8.0121, but the version on the camera remains at 3.1.1.0908.
The sdcard with the contents of readonlyhack-v0.1.zip now has a file called INFO.TXT containing "**eye.conf already existed !**" 
I renamed eye.conf and tried booting again, but the version reported by Onvifer remains stubbornly at 3.1.1.0908.
Any suggestions?

wimpie007, 2018-09-02:
@johnnt 
did you manage to fix it? I'm in the same situation. Altering /home/start.sh killed it...

johnnt, 2018-09-04:
Hi wimpie007,  Sorry didn't see your message.

Yeah i managed to get it working, in the end it was actually a split cable from bending the camera mount to 90degrees. That's why the serial interface wasn't working.

wimpie007, 2018-09-04:
Thanks for the info.
I managed to "relocate" the record-to-sd-card to a NFS mount on my nas. (a usb-stick on my nas to be exactly)
If someone is interested, give me a yell.

johnnt, 2018-09-04:
On another note, I am not sure if that has been covered already but in order to get wifi working without using the app you need to update both wpa_supplicant.conf  and devParam.dat files which are both located in /home.

The devParam.dat file is a binary file so i just copied one from another camera i had already setup but it shouldn't be too big an issue to determine the format.

Cheers
J

uglagwox, 2018-09-08:
@wimpie007 
Hi, trying NFS mount on a Digoo M1Q, failing with "operation not supported" :( . 
Could you share your NFS mount cmd, pls? Thanks much!

uglagwox, 2018-09-10:
@wimpie007
Yeah, confirm it's working. Seems my nfs kernel service was somehow stuck as well. Thanks again!

wimpie007, 2018-09-10:
without the port, it doesn't seem to work:)

sanket15n, 2018-09-30:
I used the  readonlyhack-v0.1  on digo dg-myq  and its working.
webUI , ftp and rtsp without pass  is looking good.

It would be awesome if there can be local video recording function on the SD card itself so we can get these recording via ftp or view the same on WebUI.

wimpie007, 2018-11-05:
"SHOW US!!"
Because you ask it so 'kind'...
I don't use it anymore myself, but is was: 
on startup, execute 
"
umount /mnt/mmc01/1 
mount -o port=2049,nolock,proto=tcp -t nfs IP_NUMBER:/PATH /mnt/mmc01/1/ 
"

mpelletier92, 2018-12-29:
> On another note, I am not sure if that has been covered already but in order to get wifi working without using the app you need to update both wpa_supplicant.conf and devParam.dat files which are both located in /home.
> 
> The devParam.dat file is a binary file so i just copied one from another camera i had already setup but it shouldn't be too big an issue to determine the format.
> 
> Cheers
> J
> 
> EDIT: Oh, just realised there is a file for creating devParam.dat Doh!

By trying to edit wpa_supplicant.conf I accidentally erased all the content of the file. Could someone send a reply of the proper syntax? That'd be really appreciated.

Everything working well so far with my Digoo-W01F but still trying to connect to Wifi without app.

bortcene, 2019-02-17:
Benim cihaz da 3.4.0.1012 yazılım yüklü bunun son sürüm yazılımı hangisi nasıl yüklerim

rrosancrans, 2019-03-12:
i purchased a Digoo DG-W01f heres a link to the cam https://www.walmart.com/ip/DIGOO-Indoor-Outdoor-720P-Wireless-WiFi-Network-IP-Camera-Baby-Home-Security-Monitor-Waterproof-CCTV-Cloud-Storage-Pan-Tilt-Night-Vision-Motion-Detec/422622298

back on Mar 30, 2018 kaotix to use with blueiris 

also i have the cam on wifi not sure if that will play into not connecting or not any help would be great

so i downloaded blueiris and all i get is 
HTTP
Opening 192.168.1.148 port 80...
HTTP Get / request...
Failed with HTTP 12029
Opening 192.168.1.148 port 5000...
ONVIF GetSystemDateAndTime
HTTP 12029
Checking for common cameras...
Foscam FI86xx/98xx compatible?
Foscam FI89xx compatible?
Foscam FI9821 V2 compatible?
Foscam FI9821 media port compatible?
RTSP port open?
No automatic configuration made
Done

RTSP
Opening 192.168.1.148 port 5000...
ONVIF GetSystemDateAndTime
HTTP 12029
Checking for common cameras...
Foscam FI86xx/98xx compatible?
Foscam FI89xx compatible?
Foscam FI9821 V2 compatible?
Foscam FI9821 media port compatible?
RTSP port open?
No automatic configuration made
Done

gabrielcba1982, 2019-03-13:
Same here, no success with blue iris, try GO1984 works super! 

you can download it here! If you like it, but it! ;)
https://drive.google.com/file/d/1nbVEL4y9gciZQjn4uOrZZKlEyXje8YtN/view?usp=sharing

Thatwouldbemetoo, 2019-03-16:
Hi all. Is there any way to have the Digoo W01F running on Synology using the Surveilance app? I've been mudding around with it, but cannot get it to work. 
Synology does detect it, connection checkmarks as OK, but the connection gets lost. 

I used the ReadOnly version first and now also the zsgx1hacks-v0.4.zip
I would really like it to work with onvif, and with the IR leds as well.

Did anyone succeeded in this?

Let me know!

johnnt, 2019-03-17:
> i purchased a Digoo DG-W01f heres a link to the cam [...]

I have 3 W01f and they all work with Blueiris - ignore the auto detection and just manually configure the rtsp streams.

I configure mine as follows:

set http://<ip address>
set Username and password
Make: Generic/ONVIF
Model: RTSP H.264/H.265/MJPG/MPEG4

Video:
Path /onvif1
Params <blank>

Media/video/RTSP port: 554
Discovery/ONVIF: 5000

J

spolenak, 2019-03-18:
Johnnt, are you able also to receive motion detection events on Blueiris software ?

Thatwouldbemetoo, 2019-03-18:
Thanks Johnnt !

I haven't succeeded so far to have synology Surveillance get the camera to work. It is detected by the system as an ONVIF cam, but when I want to activate it, connection gets lost. 

[image]

Is there anyone who can take a look at this and help me to get this cam to work on Synology Surveillance?

tonbor, 2019-03-25:
see discussion #104: user defined, path: admin:password@ipaddress:554/onvif/snapshot

chrisvor, 2019-04-09:
Hi all,
I successfully applied the RO hack and everything was working fine through ETHERNET. 
While I was trying to make the WLAN0 work, I must have messed with the wrong things and now ETH0 does not come up.... While it is still booting, I can see the ETH0 LED blinking, and then suddently it stops.
I saw that the AP was up (dont know why it got configured without pressing the button next to the sdcard), and so I connected to that and then through telnet on the wifi address (192.168.200.1), and then I did "ifconfig eth0 up" and ETH) was up again and working properly. Until I reboot, then it gets "auto-disabled".
In start.sh the commands are still there:
ifconfig wlan0 up
ifconfig eth0 up

Any ideas why eth0 gets disabled on its own?
Thanks for any hints
Chris

chrisvor, 2019-04-09:
> Hi all,
> I successfully applied the RO hack and everything was working fine through ETHERNET.
> While I was trying to make the WLAN0 work, I must have messed with the wrong things and now ETH0 does not come up.... While it is still booting, I can see the ETH0 LED blinking, and then suddently it stops.
> I saw that the AP was up (dont know why it got configured without pressing the button next to the sdcard), and so I connected to that and then through telnet on the wifi address (192.168.200.1), and then I did "ifconfig eth0 up" and ETH0 was up again and working properly. Until I reboot, then it gets "auto-disabled".
> In start.sh the commands are still there:
> ifconfig wlan0 up
> ifconfig eth0 up
> 
> Any ideas why eth0 gets disabled on its own?
> Thanks for any hints
> Chris

Answering my own question, I found the issue after looking at this page:
https://github.com/ant-thomas/zsgx1hacks/issues/104

The problem was with the /home/hwcfg.ini file, there is a line :
support_ap_mode = 2

I changed the 2 to 1, and now the AP does not get auto-started. It seems that when AP is on, ETH0 gets disabled (down)

Supplied the solution in case somebody else has the same issue
Chris

4j17h, 2019-04-15:
hey @johnnt , I was trying to connect the board to uart interface. are the rx, tx & gnd pins are connected to wifi module or else the infrared led port. Tried the wifi module pin out but couldn't get anything on the screen. Camera doesn't seem to work when I have added the ptz setting in hwcfg file. What are the exact rx tx & gnd pins on the board?

4j17h, 2019-04-28:
hey @johnnt @wimpie007 , I was trying to connect the board to uart interface. I tried everything but yet there is nothing I get on the screen. Can anyone please tell me the exact location of rx tx gnd pins?

4j17h, 2019-04-28:
hey @johnnt @wimpie007 , I was trying to access the board via uart interface, but couldn't get anything on the screen. Can anyone help me in locating the exact pins of rx tx & gnd on the board. 
[Isn't this supposed to be the pins?](https://imgur.com/a/QT941r2)

wimpie007, 2019-05-01:
[image]
This is the only picture i took... Was kinda obvious... (label rx, tx, gnd?)

4j17h, 2019-05-01:
w01f board seems to be different. Anyways thanks for the photo. Are there any indications on w02f board like rx, tx & gnd. W01f doesn't have any. I have tried all the combinations though.

wimpie007, 2019-05-01:
yeah, there were indications of the rx/tx/gnd... didnt have to try combinations.

4j17h, 2019-05-01:
Sadly w01f doesn't have any. @johnnt seems to have figured out the pins successfully. I just gave up unbricking the camera.

lordgotham, 2019-05-05:
hi, 
i'm french , so i don't understand all your are saying
nobody find a soluce for dg-w01F on qnap 
it's work during 1 min and shutdown automaticaly after the camera reboot work 1 min and  reboot

nobody know an equivalent of this camera? because i have many model on qnap maybe if i found a model similar i can try it

thank you

Zinstigator, 2019-09-17:
Thanks for writing this code. I have applied to my Digoo DG-W01F and can now telnet into it. :-)

The main reason for applying the hack was so that I can fix the IP address of the camera also to redirect the recording to another location.  Network Share or USB drive plugged directly into my router.

Onvif does not allow me to change the IP address - it states that the Underlying connection was closed.  I have tried looking into start.sh within the home directory but cannot see anything  specific for Ethernet IP address.

Also I have an issue trying to mount a Share.

I created a new mount point of 1 as zero is currently used and tried to mount to it using the command below.

The shares I am connecting to are completely open and set to Everyone R/W. 

'mount -o port=2049,nolock,proto=tcp -t nfs 192.168.0.3:\T_Drive /mnt/mmc01/1/' (USB Drive inrouter)

'mount -o port=2049,nolock,proto=tcp -t nfs 192.168.0.23:\open /mnt/mmc01/1/'  (FileShare)

I receive mount: 'mounting 192.168.0.23:open on /mnt/mmc01/1/ failed: Connection refused'

any help or assistance would be greatly appreciated.

Thanks.

Zinstigator, 2019-09-17:
> Thanks for writing this code. I have applied to my Digoo DG-W01F and can now telnet into it. :-)
>
> The main reason for applying the hack was so that I can fix the IP address of the camera also to [...]

I have set the IP address using ifconfig.  I am sure you have to save it in an etc location though for it to be permanent.  My Unix is very rusty its been years since i have used it.

Zinstigator, 2019-09-17:
I have edited the start.sh file within the /home directory  setting the ip address. I can see on bootup it starts to use the new address for a few seconds it then seems to revert back to the old DHCP address. A

I have managed to get it to keep the correct TZ now with the echo statement within start.sh

hr1179, 2019-11-10:
After a long time I tried again to get my camera working. The first time it was working, but now I'm not able anymore to connect to the rtsp stream. 
My first assumption was a network issue but then I recognized that there was an issue with the sdcard. So I used another one and also enable the permanent hack.
But still connection was not possible.
So I checked if the rtsp ports are listening at all, but it seems they are not:

```
# netstat -anp
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 0.0.0.0:21              0.0.0.0:*               LISTEN      137/tcpsvd
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      135/dropbearmulti
tcp        0      0 0.0.0.0:7101            0.0.0.0:*               LISTEN      179/p2pcam
tcp        0      0 0.0.0.0:7103            0.0.0.0:*               LISTEN      179/p2pcam
tcp        0      0 0.0.0.0:3201            0.0.0.0:*               LISTEN      157/tees
tcp        0      0 192.168.1.22:22         192.168.1.21:49277      ESTABLISHED 225/dropbearmulti
tcp        0      0 :::8080                 :::*                    LISTEN      130/busybox
tcp        0      0 :::22                   :::*                    LISTEN      135/dropbearmulti
tcp        0      0 :::23                   :::*                    LISTEN      185/telnetd
udp        0      0 0.0.0.0:8001            0.0.0.0:*                           179/p2pcam
Active UNIX domain sockets (servers and established)
Proto RefCnt Flags       Type       State         I-Node PID/Program name    Path
unix  2      [ ACC ]     STREAM     LISTENING       1033 179/p2pcam          /tmp/aaa
unix  3      [ ]         DGRAM                       837 157/tees            /var/run/tees.sock
unix  2      [ ]         DGRAM                      1045 179/p2pcam
```

Shouldn't it listen on port 554 and 5000?
Any idea?

hr1179, 2019-11-26:
I've connected a UART interface now and there is this error message:
`ERROR: can not get mac from net interface, please check hardware...`

But ifconfig shows that hwaddr for eth0 (wlan not used up to now) is set correctly.
Could this be the reason for the issue in my post above?

tonbor, 2019-11-26:
tcp        0      0 192.168.1.155:554       192.168.1.249:50264     ESTABLISHED                                                                                                                                                              210/p2pcam
tcp        0      0 192.168.1.155:554       192.168.1.249:50243     ESTABLISHED                                                                                                                                                              210/p2pcam
No only 554

tonbor, 2019-11-26:
settings in cls.conf
[cls_server]
fixip = eth0
ip = 192.168.X.XXX
gw = 192.168.X.X

hr1179, 2019-11-30:
If created the cls.conf in /home directory and it is used, but the camera is still not listining on port 554 and I'm therefore not able to connect.
Any other ideas.
Is there a way to restore the image or at least the /home directory?

hr1179, 2019-12-22:
Still no success. Does anyone have a firmware.bin for Digoo DG-W01F?

tonbor, 2019-12-22:
Wrong directory for cls.conf

hr1179, 2019-12-22:
Ok, put onto sdcard, correct?
But still does not work. Even if I restore hack, camera is not connecting to any external cloud service.

tonbor, 2019-12-22:
In the process, I found out that it is possible in / home or on a USB flash drive (checked / mnt / mmc01, vfat and exfat are supported) to put the cls.conf file with something like this
  [cls_server] 
  fixip = eth0 
  ip = 192.168.1.123 
  gw = 192.168.1.1 
  netmask = 255.255.255.0 
  dns = 192.168.1.1 
  ssid = wifiname 
  passwd = 129345678 

to set the network configuration. The same parameters are copied to devParam.dat.
If you remove fixip = eth0, then wifi will be activated.

hr1179, 2019-12-22:
Yes, I saw it also via UART interface that cls.conf was found on /home. 
But network is setup correctly. The issue is, that the camera is not listing on port 554.

tonbor, 2019-12-22:
rtsp://192.168.1.XXX:554/ch0_0.h264

[11 more replies]