ant-thomas/zsgx1hacks #27: ZS-GX1 - Firmware

erwill2, 2018-01-27:
Anyone have backup firmware for the ZS-GX1 I can download? My camera is stuck loop.

purduephotog, 2018-02-03:
I have two that I haven't been able to get registered with the app or wired. I could use a chip dumper and dump the firmware. You'd have to use the bin file though and tear it apart as needed... at least I'm assuming I can dump the firmware.

erwill2, 2018-02-03:
I'll try anything at this point. I'd be happy to try.

purduephotog, 2018-02-03:
[gd25q128.zip](https://github.com/ant-thomas/zsgx1hacks/files/1691364/gd25q128.zip)

25q127cs1g is what is on the chip; the reader saw it as something different but it looked OK

erwill2, 2018-02-04:
Thanks for the firmware. Will see what I can do. I did receive a firmware from the seller if anyone wants to take a look. It is attached. It didn't fix my problem. 

[firmware.zip](https://github.com/ant-thomas/zsgx1hacks/files/1693024/firmware.zip)

I did figure out what happened to my camera. The GUUDGO ONVIF firmware.bin changed the version of p2pcam that is incompatible. I was able to keep it running by modifying the debug_cmd.sh to replace the version of p2pcam running. Still unable to get a stream going though.

purduephotog, 2018-03-06:
I can't help much as I still can't get the cams to recognize the app or do anything. Tried the SD Card and no luck. Was hoping someone had a good set of firmware with an IP so that I could at least burn it to the chip manually and use the thing.

erwill2, 2018-03-06:
Thanks for the help! I got mine to stream again. I used the attached firmware and while attached to serial I noticed when booting up I was missing one of the voice files it was trying to play. Which caused it to reboot immediately. I added the missing file and it booted right up.