ant-thomas/zsgx1hacks #28: Few issues

thehijacker, 2018-01-28:
Hello,

Great little hack to prevent it sending video to their cloud.

I see some problems that could make this perfect cam for my newborn room and to be used as baby monitor.

1. RTSP stream has no sound.
2. IR LED should turn on/off automatically depending on the brightness level in the environment.
3. Command ntpd -q -p uk.pool.ntp.org did not set correct time for me. I had to use my ISP NTP server address. Maybe we need more "general" NTP server.

Is there a solution already available for this issues?

Thank you and best regards.

thehijacker, 2018-01-28:
Found another "issues". When camera boots I get a voice message from it to "Please wait for wired connection". When the sound is heard I can already connect via SSH. Can we disable this voice permanently?

deorder, 2018-02-04:
You can try preventing the boot script from unpacking the voice files in /tmp. My cameras do not play the voice files as I wrote a custom boot that does not unpack them.

thehijacker, 2018-02-04:
Excelent. Thank you. How about RTSP with sound?

deorder, 2018-02-08:
Could not get that to work. The p2pcam software crashes when enabling the audio stream. The best solution would be to write something that can p2pcam of course, but without documentation of the hardware it is something I will not take upon me.

ant-thomas, 2018-02-13:
Found RTSP with sound
`rtsp://ipaddress/0/av0`

`av0` gives a high res stream with sound
`av1` gives a low res stream with sound

thehijacker, 2018-02-13:
I noticed that IR LED in fact does turn on automatically on low light. Not sure why this did not work before. I must have played with GPIO commands at some point.

Time is also correct on camera. I had to do a delay so that network is initialized. Like so:

```
# update the time
(sleep 20 && /usr/sbin/ntpd -q -p time.ijs.si ) &
```

Last issue that I had was RTSP without sound and to my great suprise ant-thomas came with correct URL that includes sound.

Only issue that I have is with Android application (TinyCam). I use ONVIF protocol since with that I can use PTZ. But if I enable sound the cam will freeze and reboot itself. Any idea about that?

Thank you!

ant-thomas, 2018-02-13:
I've just tried to enable to sound with TinyCam and had the same problem. Camera crashed and rebooted, Odd behaviour.