ant-thomas/zsgx1hacks #30: Cannot connect to RTSP

cptX, 2018-02-13:
Hi to all,

I successfully logged in through telnet and through ssh, but it is completely impossible to find a proper rtsp stream to get the image from. I have tried hundreds of combinations without success.

I'm using Linux Mint 18.2 and I have tried these rtsp links either with VLC and mplayer...

Any help would be much appreciated!

erwill2, 2018-02-13:
run a netstat -a to find out if 554 is open.

gilbertman77, 2018-02-13:
What camera are you trying to install?   I purchased 2 get cheap digoo cloud locked cameras from banggood, and this worked perfectly to better secure them and access the streams.

- Verify default user and pw for camera (not telnet login necessarily)
- Verify default rstp stream address for camera
- Verify port availability with netstat per above.

cptX, 2018-02-13:
This is what I get from netstat
`
# netstat -a
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       
tcp        0      0 0.0.0.0:5050            0.0.0.0:*               LISTEN      
tcp        0      0 0.0.0.0:7101            0.0.0.0:*               LISTEN      
tcp        0      0 0.0.0.0:7103            0.0.0.0:*               LISTEN      
tcp        0      0 0.0.0.0:8001            0.0.0.0:*               LISTEN      
tcp        0      0 0.0.0.0:3201            0.0.0.0:*               LISTEN      
tcp        0      0 0.0.0.0:554             0.0.0.0:*               LISTEN      
tcp        0      0 0.0.0.0:843             0.0.0.0:*               LISTEN      
tcp        0      0 0.0.0.0:6670            0.0.0.0:*               LISTEN      
tcp        0      0 localhost:9008          0.0.0.0:*               LISTEN      
tcp        0      0 0.0.0.0:www             0.0.0.0:*               LISTEN      
tcp        0      0 0.0.0.0:telnet          0.0.0.0:*               LISTEN      
tcp        0      0 ipc:60802               ec2-52-52-190-141.us-west-1.compute.amazonaws.com:5222 ESTABLISHED 
tcp        0   3126 ipc:34971               ec2-54-93-188-188.eu-central-1.compute.amazonaws.com:https ESTABLISHED 
tcp        0     20 ipc:telnet              (my computer):38712          ESTABLISHED 
udp        0      0 0.0.0.0:7998            0.0.0.0:*                           
udp        0      0 0.0.0.0:8001            0.0.0.0:*                           
udp        0      0 0.0.0.0:8002            0.0.0.0:*                           
Active UNIX domain sockets (servers and established)
Proto RefCnt Flags       Type       State         I-Node Path
unix  3      [ ]         DGRAM                       560 /var/run/tees.sock
unix  2      [ ACC ]     STREAM     LISTENING        754 /tmp/aaa
unix  2      [ ]         DGRAM                       768 
`

What is the proper rtsp link I should use in vlc to get the image stream?
How can I use it with credentials?
For example I have tried the following without success:

rtsp://admin@192.168.2.31
rtsp://admin@192.168.2.31/
rtsp:/192.168.2.31
rtsp://192.168.2.31/0/av1
rtsp://admin:admin@192.168.2.31:8001/onvif1
rtsp://admin:@192.168.2.31:8001/onvif1
rtsp://admin@192.168.2.31:554/onvif1

What program, what link and what credentials should I use?

ant-thomas, 2018-02-13:
Try `nmap`

The result I get for an ZS-GX1 is which shows port 554 is open
```
nmap 192.168.1.174

Starting Nmap 7.40 ( https://nmap.org ) at 2018-02-13 19:01 GMT
Nmap scan report for ipc (192.168.1.174)
Host is up (0.012s latency).
Not shown: 994 closed ports
PORT     STATE SERVICE
22/tcp   open  ssh
23/tcp   open  telnet
80/tcp   open  http
554/tcp  open  rtsp
7103/tcp open  unknown
8001/tcp open  vcom-tunnel

Nmap done: 1 IP address (1 host up) scanned in 1.22 seconds```

If you try and open `rtsp://ipaddress` in VLC it should present you with a login

cptX, 2018-02-15:
@ant-thomas This is the result of nmap:

Starting Nmap 7.01 ( https://nmap.org ) at 2018-02-15 09:34 EET
Nmap scan report for ipc (192.168.2.31)
Host is up (0.0032s latency).
Not shown: 993 closed ports
PORT     STATE SERVICE
23/tcp   open  telnet
80/tcp   open  http
554/tcp  open  rtsp
843/tcp  open  unknown
5050/tcp open  mmcc
7103/tcp open  unknown
8001/tcp open  vcom-tunnel

Strange is that ssh is no longer available and I cannot connect through it.
I never succeeded in using vlc, and it never presented me with a login...

What am I doing wrong?

cptX, 2018-02-15:
I reflashed the files in the SD and now ssh is again functional but vlc still cannot open the rtsp:// link

cptX, 2018-02-15:
Tried using mplayer but still no luck

`mplayer rtsp://192.168.2.31
Creating config file: /home/myron/.mplayer/config
MPlayer 1.2.1 (Debian), built with gcc-5.3.1 (C) 2000-2016 MPlayer Team
mplayer: could not connect to socket
mplayer: No such file or directory
Failed to open LIRC support. You will not be able to use your remote control.

Playing rtsp://192.168.2.31.
Resolving 192.168.2.31 for AF_INET6...

Couldn't resolve name for AF_INET6: 192.168.2.31
Connecting to server 192.168.2.31[192.168.2.31]: 554...

librtsp: server responds: 'RTSP/1.0 401 Unauthorized'
libavformat version 56.40.101 (external)
libavformat file format detected.
[rtsp @ 0x7f4c4ad3ad80]CSeq 1 expected, 0 received.
[rtsp @ 0x7f4c4ad3ad80]method OPTIONS failed: 401 Unauthorized
LAVF_header: av_open_input_stream() failed

Exiting... (End of file)
`

`mplayer rtsp://admin@192.168.2.31
MPlayer 1.2.1 (Debian), built with gcc-5.3.1 (C) 2000-2016 MPlayer Team
mplayer: could not connect to socket
mplayer: No such file or directory
Failed to open LIRC support. You will not be able to use your remote control.

Playing rtsp://admin@192.168.2.31.
Resolving 192.168.2.31 for AF_INET6...

Couldn't resolve name for AF_INET6: 192.168.2.31
Connecting to server 192.168.2.31[192.168.2.31]: 554...

librtsp: server responds: 'RTSP/1.0 401 Unauthorized'
libavformat version 56.40.101 (external)
libavformat file format detected.
[rtsp @ 0x7f97cbda7d80]CSeq 1 expected, 0 received.
[rtsp @ 0x7f97cbda7d80]method OPTIONS failed: 401 Unauthorized
LAVF_header: av_open_input_stream() failed

Exiting... (End of file)
`

erwill2, 2018-02-15:
Try this rtsp://admin:@192.168.2.31.

Or rtsp://admin:admin@192.168.2.31.

There are a few password combinations out there.

cptX, 2018-02-15:
Tried this already, no luck!

cptX, 2018-02-16:
I just tried Onvif Device Manager in Windows again with no success. Something is really wrong with my Ip camera, or I'm missing something really basic here...
I also searched for passwords in the file system. I discovered a password in /etc/passwd which was never used. I tried this in rtsp protocol again without success...

erwill2, 2018-02-18:
Try removing from the cloud service and resetting. One of my IP Cameras shutoff most ports once connected to the cloud.

cptX, 2018-02-20:
Thanks erwill2, your idea was interesting so I tried it but without success.
This is what I get in any case:

`mplayer -user admin rtsp://192.168.2.31
MPlayer 1.2.1 (Debian), built with gcc-5.3.1 (C) 2000-2016 MPlayer Team
mplayer: could not connect to socket
mplayer: No such file or directory
Failed to open LIRC support. You will not be able to use your remote control.

Playing rtsp://192.168.2.31.
Resolving 192.168.2.31 for AF_INET6...

Couldn't resolve name for AF_INET6: 192.168.2.31
Connecting to server 192.168.2.31[192.168.2.31]: 554...

librtsp: server responds: 'RTSP/1.0 401 Unauthorized'
libavformat version 56.40.101 (external)
libavformat file format detected.
[rtsp @ 0x7fe476ca0d80]CSeq 1 expected, 0 received.
[rtsp @ 0x7fe476ca0d80]method OPTIONS failed: 401 Unauthorized
LAVF_header: av_open_input_stream() failed

Exiting... (End of file)
`
Also I noticed that in every failed attempt actually the camera opens an new socket on 554 port. So after several attempts there is a long list of open connections ...
`
# netstat -n
Active Internet connections (w/o servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       
tcp        0      0 192.168.2.31:554        192.168.2.2:49924       ESTABLISHED 
tcp        0      0 192.168.2.31:554        192.168.2.2:49916       ESTABLISHED 
tcp        0      0 192.168.2.31:554        192.168.2.2:49906       ESTABLISHED 
tcp        0      0 192.168.2.31:554        192.168.2.2:49922       ESTABLISHED 
tcp        0      0 192.168.2.31:554        192.168.2.2:49950       ESTABLISHED 
tcp        0      0 192.168.2.31:554        192.168.2.2:49942       ESTABLISHED 
tcp        0      0 192.168.2.31:554        192.168.2.2:49918       ESTABLISHED 
tcp        0      0 192.168.2.31:554        192.168.2.2:49954       ESTABLISHED 
tcp        0      0 192.168.2.31:554        192.168.2.2:49948       ESTABLISHED 
tcp        0      0 192.168.2.31:554        192.168.2.2:49952       ESTABLISHED 
tcp        0      0 192.168.2.31:554        192.168.2.2:49926       ESTABLISHED 
tcp        0      0 192.168.2.31:554        192.168.2.2:49908       ESTABLISHED 
tcp        0      0 192.168.2.31:554        192.168.2.2:49912       ESTABLISHED 
tcp        0      0 192.168.2.31:554        192.168.2.2:49944       ESTABLISHED 
tcp        0      0 192.168.2.31:22         192.168.2.2:39948       ESTABLISHED 
tcp        0      0 192.168.2.31:554        192.168.2.2:49946       ESTABLISHED 
tcp        0      0 192.168.2.31:554        192.168.2.2:49914       ESTABLISHED 
tcp        0      0 192.168.2.31:554        192.168.2.2:49910       ESTABLISHED 
tcp        0      0 192.168.2.31:554        192.168.2.2:49920       ESTABLISHED 
udp        0      0 192.168.2.31:39318      192.168.2.1:53          ESTABLISHED 
Active UNIX domain sockets (w/o servers)
Proto RefCnt Flags       Type       State         I-Node Path
unix  3      [ ]         DGRAM                       841 /var/run/tees.sock
unix  2      [ ]         DGRAM                      1057 
`

I think the whole problem starts from the fact that I don't know the credentials on the rtsp server...
How can I set my own credentials or replace the rtsp server with my own?

cptX, 2018-02-20:
Could be that the firmware of the camera is different? Can I install another firmware?

cptX, 2018-04-05:
Any help? Still not solved!

cptX, 2018-04-05:
By the way, Nemo in linux Mint cannot connect through ssh, although I can connect with SSH through terminal successfully. Does this show something?

cptX, 2018-04-05:
VLC responds "access_realrtsp access warning: only real/helix rtsp servers supported for now"
but I successfully openned a test rtsp stream I found in internet, both with vlc and mplayer. So it looks like the stream my ipcam is producing is not understood by my players. Or the url I'm using is wrong.
How can I find the correct url?

cptX, 2018-04-05:
So I installed  zsgx1hacks-v0.4.zip this time.
First of all I can access the webinterface, and I can turn on and off the IR LEDs but I cannot move the camera. PTZ is not working.

Second, no progress regarding getting the RTSP stream.
The only successful authorized url is the following:

rtsp://192.168.2.31:5050

but mplayer always fails saying "LAVF_header: av_open_input_stream() failed"
I think is a matter of encoding of the stream.
Any clues?

cptX, 2018-04-05:
OK, installed openRTSP and this is what I got:

openRTSP rtsp://192.168.2.31:5050
Opening connection to 192.168.2.31, port 5050...
...remote connection opened
Sending request: OPTIONS rtsp://192.168.2.31:5050 RTSP/1.0
CSeq: 2
User-Agent: openRTSP (LIVE555 Streaming Media v2016.02.09)

Opening connection to 192.168.2.31, port 5050...
...remote connection opened
Sending request: DESCRIBE rtsp://192.168.2.31:5050 RTSP/1.0
CSeq: 3
User-Agent: openRTSP (LIVE555 Streaming Media v2016.02.09)
Accept: application/sdp

Failed to get a SDP description for the URL "rtsp://192.168.2.31:5050": liveMedia0

cptX, 2018-04-05:
Also tried ffmpeg

ffmpeg -i rtsp://192.168.2.31:5050
ffmpeg version 2.8.11-0ubuntu0.16.04.1 Copyright (c) 2000-2017 the FFmpeg developers
  built with gcc 5.4.0 (Ubuntu 5.4.0-6ubuntu1~16.04.4) 20160609
  configuration: --prefix=/usr --extra-version=0ubuntu0.16.04.1 --build-suffix=-ffmpeg --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --cc=cc --cxx=g++ --enable-gpl --enable-shared --disable-stripping --disable-decoder=libopenjpeg --disable-decoder=libschroedinger --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmodplug --enable-libmp3lame --enable-libopenjpeg --enable-libopus --enable-libpulse --enable-librtmp --enable-libschroedinger --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxvid --enable-libzvbi --enable-openal --enable-opengl --enable-x11grab --enable-libdc1394 --enable-libiec61883 --enable-libzmq --enable-frei0r --enable-libx264 --enable-libopencv
  libavutil      54. 31.100 / 54. 31.100
  libavcodec     56. 60.100 / 56. 60.100
  libavformat    56. 40.101 / 56. 40.101
  libavdevice    56.  4.100 / 56.  4.100
  libavfilter     5. 40.101 /  5. 40.101
  libavresample   2.  1.  0 /  2.  1.  0
  libswscale      3.  1.101 /  3.  1.101
  libswresample   1.  2.101 /  1.  2.101
  libpostproc    53.  3.100 / 53.  3.100
rtsp://192.168.2.31:5050: Invalid data found when processing input

cptX, 2018-04-06:
So my conclusion is that the camera is producing an invalid stream.

lobmarib, 2018-05-31:
Anyone had success with the rtsp url?

cptX, 2018-05-31:
I personally never succeeded getting live video from my camera. The url should be rtsp://192.168.2.31:5050 but my camera probably makes something wrong with the rtsp protocol.
Also I never got any answer regarding using another firmware, nobody answered my questions here or here: https://github.com/ant-thomas/zsgx1hacks/issues/48

davidlb, 2018-06-01:
@cptX, what is your camera model?

To access mine, I have to set this password: dg20160404

lobmarib, 2018-06-07:
@cptX I have tried every url I found in here and none works. 
I notice also that the camera reboots time to time so I assume that is something wrong with this "code". My camera model is zs-gx1.

lobmarib, 2018-06-07:
@ant-thomas this "code" was developed to which camera model? Do you know if there are more than one version of the zs-gx1 model? Thanks for your time and work

cptX, 2018-11-20:
Hi guys I'm so happy today!! After so many months I have successfully succeeded in getting the live stream. First of all, I upgraded the firmware of the camera throught the Android app. New firmware is 3.2.8.0121. Secondly, I installed the new readonlyhack-v0.1.zip code from here. And last but not least the address for me was: rtsp://admin@192.168.2.31:8001, which was never referenced in any forum before... Looks like I have a special model. I used vlc to open the above address.