ant-thomas/zsgx1hacks #35: GIBSSI 1080P Wireless IP Camera

chillophil, 2018-03-08:
Got a GIBSSI 1080P Wireless IP Camera with same SoC (Goke GK7102S) but different PCB. Telnet/root credentials are equivalent, even RTSP (on different port: 8001) works with VLC but can not control it with onvif (tried iSpy) ...

Any suggestions?

```
# uname -a
Linux IP CAMERA 3.4.43-gk #61 PREEMPT Wed May 17 19:07:50 CST 2017 armv6l GNU/Linux

```
```
# netstat -tulpen
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 0.0.0.0:843             0.0.0.0:*               LISTEN      175/p2pcam
tcp        0      0 0.0.0.0:6670            0.0.0.0:*               LISTEN      175/p2pcam
tcp        0      0 127.0.0.1:9008          0.0.0.0:*               LISTEN      175/p2pcam
tcp        0      0 0.0.0.0:23              0.0.0.0:*               LISTEN      178/telnetd
tcp        0      0 0.0.0.0:5050            0.0.0.0:*               LISTEN      175/p2pcam
tcp        0      0 0.0.0.0:7101            0.0.0.0:*               LISTEN      175/p2pcam
tcp        0      0 0.0.0.0:7103            0.0.0.0:*               LISTEN      175/p2pcam
tcp        0      0 0.0.0.0:8001            0.0.0.0:*               LISTEN      175/p2pcam
tcp        0      0 0.0.0.0:3201            0.0.0.0:*               LISTEN      149/tees
udp        0      0 0.0.0.0:7998            0.0.0.0:*                           175/p2pcam
udp        0      0 0.0.0.0:8001            0.0.0.0:*                           175/p2pcam
udp        0      0 0.0.0.0:8002            0.0.0.0:*                           175/p2pcam
```

```
# pwd
/home
# ls -la
drwxrwxrwx    5 root     root             0 Mar  8 18:39 .
drwxr-xr-x   18 root     root             0 Jan  1  1970 ..
-rwxrwxrwx    1 root     root          1200 Jan  1  1970 1080_mem_cfg.bin
-rwxrwxrwx    1 root     root       1137700 Sep  8 02:41 8188fu.ko
-rwxrwxrwx    1 root     root        361845 Jan  1  1970 VOICE.tgz
-rwxrwxrwx    1 root     root        334850 May  5  2017 VOICE.tgz.yuan
-rwxrwxrwx    1 root     root         59746 Jan  1  1970 ca-bundle-add-closeli.crt
-rwxrwxrwx    1 root     root           375 Jan  1  1970 check_mem.sh
-rwxrwxrwx    1 root     root         14153 Jan  1  1970 chmemcfg
-rwxrwxrwx    1 500      500            634 Jan  1  1970 cloud.ini
-rwxrwxrwx    1 500      500            609 Jan  1  1970 cloud_oversea.ini
-rw-r--r--    1 root     root           753 Mar  8 18:39 config.cfg
-rwxrwxrwx    1 root     root           753 Mar  8 18:39 config.cfg.bak
-rwxrwxrwx    1 root     root           141 Mar  7 12:33 config.json
-rwxrwxrwx    1 root     root          3659 Mar  8 18:40 config.xml
-rwxrwxrwx    1 root     root           141 Nov  8 15:19 config_bak.json
-rwxrwxrwx    1 root     root           155 Jun 23  2017 custom_init.sh
-rwxrwxrwx    1 500      500           7376 May 25  2017 debugtool
-rwxrwxrwx    1 root     root          1024 Mar  8 18:39 devParam.dat
drwxrwxrwx    2 1000     default          0 Mar 31  2017 extra
-rwxrwxrwx    1 root     root            32 Jul 28  2017 eye.conf
-rwxrwxrwx    1 500      500           1575 Jun 20  2017 factory_tool.sh
-rwxrwxrwx    1 root     root         14078 May 13  2017 flashwriteMtd
-rwxrwxrwx    1 500      500           4639 May 17  2017 gio.ko
-rwxrwxrwx    1 root     root        677824 Jun 23  2017 gk_fw.bin
-rwxrwxrwx    1 root     root         26094 Jan  1  1970 gkptz-dsa.ko
-rwxrwxrwx    1 root     root         25734 Jan  1  1970 gkptz.ko
-rwxrwxrwx    1 root     root           455 Jun 23  2017 hardinfo.bin
-rwxrwxrwx    1 root     root           137 Sep 12 07:57 hwcfg.ini
-rwxrwxrwx    1 root     root             8 Mar  7 12:34 idx.log
-rwxrwxrwx    1 root     root            30 Jan  1  1970 model.ini
-rwxrwxrwx    1 root     root       2706387 Jan  1  1970 p2pcam.tar.gz
-rwxrwxrwx    1 root     root            20 Mar  8 10:39 psp.dat
-rwxrwxrwx    1 root     root            89 Jun 23  2017 ptz.cfg
-rwxrwxrwx    1 500      500          21489 May  4  2017 sdc_tool
-rwxrwxrwx    1 500      500           1779 Apr 13  2017 sensor.sh
-rwxrwxrwx    1 500      500          38796 May 13  2017 sensordetect
drwxrwxrwx    2 500      500              0 Jan 23  2017 sensors
-rwxrwxrwx    1 500      500           3337 Jan  1  1970 start.sh
[... 175 lines cut ...]
MAC Address: 00:0C:43:xx:xx:xx (Ralink Technology)
Device type: general purpose
Running: Linux 2.6.X|3.X
OS CPE: cpe:/o:linux:linux_kernel:2.6 cpe:/o:linux:linux_kernel:3
OS details: Linux 2.6.32 - 3.13
Uptime guess: 0.002 days (since Thu Mar  8 19:44:12 2018)
Network Distance: 1 hop
TCP Sequence Prediction: Difficulty=254 (Good luck!)
IP ID Sequence Generation: All zeros

TRACEROUTE
HOP RTT     ADDRESS
1   0.53 ms 192.168.2.3

NSE: Script Post-scanning.
Initiating NSE at 19:46
Completed NSE at 19:46, 0.00s elapsed
Initiating NSE at 19:46
Completed NSE at 19:46, 0.00s elapsed
Read data files from: /usr/local/bin/../share/nmap
OS and Service detection performed. Please report any incorrect results at https://nmap.org/submit/ .
Nmap done: 1 IP address (1 host up) scanned in 55.67 seconds
           Raw packets sent: 1026 (45.986KB) | Rcvd: 1014 (41.258KB)

```

alxponom, 2019-04-07:
Hi, there. 
I got a similar device with disabled ONVIF protocol. I've fixed it by inserting a line "support_onvif = 1" in the file "hwcfg.ini" and restaing the device.

Anon0ne, 2019-12-17:
what was telnet login? i cannot get into my cam.

alxponom, 2019-12-18:
try using following: user: root, password: cxlinux

alxponom, 2019-12-18:
try using following: user: root, password: cxlinux

From: 0nezer0
Sent: 18 декабря 2019 г. 2:42
To: ant-thomas/zsgx1hacks
Cc: alxponom; Comment
Subject: Re: [ant-thomas/zsgx1hacks] GIBSSI 1080P Wireless IP Camera - onvif ?(#35)

what was telnet login? i cannot get into my cam.

StevieeG, 2023-05-01:
Hey did anyone have http running on port 80 on their camera? We must have a very similar camera, if so anyone get the creds for HTTP web page??

StevieeG, 2023-05-01:
Hey do you know the creds for the webpage on port 80? Looks like we have the same IP camera but I have port 80 open on mine.