ant-thomas/zsgx1hacks #37: Any way i can contribute?

kirelam, 2018-03-12:
Dear Cheap cam buyers :P 

I also bought a cheap chinese cam (well 2). This is a Lintratek with 3 antenna's. 
All connections are same as always through app and cloud. 
Also telnet is open with the password found here before.

Would really want to help here. 
Also, took apart 1 cam and this seems to be a slightly different chip GK7102s
[image]
[image]
[image]

ant-thomas, 2018-03-12:
Realistically all the simple things have been found.
You are welcome to confirm the script works on your camera so it can be added to the list?
Only real progress would be to create a new RTSP server.

gilbertman77, 2018-03-12:
Web interface wouldn't be bad either..

kirelam, 2018-03-12:
Is this something as a lead?

https://www.unifore.net/ip-video-surveillance/erobot-ip-camera-debug-software-download-and-guide.html

kirelam, 2018-03-15:
There is a way to create a low level RTSP server i found. Is this usefull people?

https://github.com/watson/rtsp-server

kirelam, 2018-03-15:
I will install zsgx1hacks-v0.2.zip tonight on my cam. But the awkard thing is that my version of the firmware allready has Telnet access and the file structure seems off, with the one shown here on the site

Can anyone give me the cmd's over telnet to export the mounted firmware for backup/exploit purposes?

deorder, 2018-03-17:
It would help to find what checksum algorithm is used for devParam.dat, the first few bytes. If that works the settings for p2pcam can be changed without requiring the app. Next it would be useful if someone can reverse engineer the api for video and audio. I created ioctl_debug which can be used to test the device io.

kirelam, 2018-03-19:
So i have to find the file "devParam.dat", is that correct? 
After that look at the first few bytes of it.

deorder, 2018-03-21:
It is located in /home and it contains all the settings for p2pcam in binary format.

kirelam, 2018-03-22:
Found the file and will copy it to a Micro SD card if i can manage. Made allready a spare copy of the file, so i will try to move that one. 
Will upload it then here and also will try to look around in it.

thoompje, 2018-03-28:
Hi, do you maybe have the firmware of the "GK7102s" chip?

I updated my device with the SD-card to  zsgx1hacks-v0.3 but it is not stuck in a bootloop, see:
https://github.com/ant-thomas/zsgx1hacks/issues/44

Can anyone help me with that?

kirelam, 2018-04-16:
Would it be helping, if i would upload all the files from my cam?
If yes, what would be the best way? 
Copy them to the SD card?