ant-thomas/zsgx1hacks #38: Works on ZS-GX1 but sharp whistling

scassarGitHub, 2018-03-15:
Hi,

Thanks for the hacks. That saves me money.

Works very well on 2 ZS-GX1 with latest firmware but I heard a sharp whistling on the 2 devices when booted on SDCard and not on the original firmware.

Could you help me ?

deorder, 2018-03-17:
If you get the sharp whistling it means the 'goke-volume' binary is not run correctly.

ttanel, 2018-03-23:
On mine faint whistling is there even if I set volume to 0 via goke_volume. I just disconnected the speaker from motherboard and now it's completely silent.

scassarGitHub, 2018-03-23:
Oh really ! Was it difficult to disconnect ? Do you need to break some solders or were there connectors to disconnect ?

ttanel, 2018-03-23:
Nope, it's very easy and reversible, both mic and speaker have connectors on main board.

scassarGitHub, 2018-03-23:
Thank you very much, I will try this WE.

ant-thomas, 2018-03-24:
The whistling is often still there, just quieter.
I've also disconnected the speaker because it serves no purpose for me.

Disconnect the attached cable
[image]

reglisse69, 2018-03-27:
Thanks a lot for all your help.

Finally v0.3 (seems) to make the noise disapeared for me (tried quickly this morning).

reglisse69, 2018-03-29:
Ok in fact sound is always there but almost inaudible for some people and depending on the noises in the house.

I will try to do the hardware method.