ant-thomas/zsgx1hacks #39: Does this work on the Digoo M1Q?

WRH2000, 2018-03-16:
http://www.mydigoo.com/Digoo-DG-M1Q-960P-2_8mm-Wireless-Mini-WIFI-Night-Vision-Smart-Home-Security-IP-Camera-Onvif-Monitor-p-38.html

Would be interested if anyone has tried this on the Digoo M1Q cameras? Apparently they run on the Goke GK7102 chip as well.

The current firmware is annoying because it wants to setup a tunnel to a server in China, when I use the firewall to block access to the internet it starts a timer and will reset itself. Also uses the tunnel to get NTP time so the time is always messed up.

ant-thomas, 2018-03-24:
I think the best advice would be to try it - probably try v0.2 so it isn't persistent

roaringbits, 2018-05-04:
It seems to work on mine (webinterface etc), but I can't access the RTSP stream yet. Will try to find out how to access it :)

roaringbits, 2018-05-06:
I can access the RTSP stream with;

URL: rtsp://{ip-address}/onvif2
Credentials: admin / dg20160404

WRH2000, 2018-12-05:
How did everyone get the firmware uploaded? I've tried inserting the SD card and rebooting but nothing happens. Also was able to telnet in and tried to run the debug_cmd.sh (readonly) and the start.sh but neither worked. Any ideas?

nc1618, 2019-02-03:
hey WRH2000 have a look at the below Issue, I was having similar issues with the MYQ and got it all sorted: https://github.com/edsub/Goke_GK7102/issues/1#issuecomment-460042652