ant-thomas/zsgx1hacks #42: Change password?

xenio, 2018-03-24:
I like to connect the camera to a cloud to keep the recordings, is there a way to change the RTSP password? It is now user:admin and blank password.

ant-thomas, 2018-03-24:
So far I don't think a way has been found.

xenio, 2018-03-29:
Any idea where the user/password are? Is it hardcoded in the firmware?

lukasz-glowacki, 2018-03-30:
RTSP Login: admin
RTSP Password: dg20160404

Following links works for me:
rtsp://admin:dg20160404@camera_ip_address/onvif1
rtsp://admin:dg20160404@camera_ip_address/onvif2

xenio, 2018-03-30:
@lukasz-glowacki thanks but my camera has admin with no password, I was asking a way to change/set the password. I have the camera exposed to the internet and this is not a good thing, I have a restrict access with fixed ip via nat on the router.

ant-thomas, 2018-03-30:
I suppose the idea with these cameras is you shouldn't be exposing them to the internet because of the cloud app - which I realise we've disabled.

cptX, 2018-04-05:
@lukasz-glowacki Hi, how did you find the password and the rtsp link.
I have been trying several weeks to connect to rtsp but was not successful.
see the thread I openned https://github.com/ant-thomas/zsgx1hacks/issues/30

lukasz-glowacki, 2018-04-05:
@cptX I found it there:
https://www.youtube.com/watch?v=QDHUVUvunEw

detxm80, 2018-11-20:
hi, there is no way to protect rstp stream with a password?

detxm80, 2019-04-08:
hi guys, news from rstp password stream?

Fietspomp86, 2019-04-08:
Did you not read my comment? You can change the password using my link. Adjust password in ini file and copy this together with bin file to see. Boot and volia, password is set. Boot custom firmware again and password is still set.

Fietspomp86, 2019-08-17:
Another very simple way is to FTP into your camera if this hack is installed, go to /home/ and there is a file called hwcfg.ini.
Open it and change the line 
`passwd = *******`

Of course change the **** into your desired password

HaylanderLandim, 2020-05-20:
If it still works for someone.
I accessed the camera by telnet and added the parameter `passwrd = ******` using `vi /home/hwcfg.ini`

bisewski001, 2021-08-03:
> If it still works for someone.
> I accessed the camera by telnet and added the parameter `passwrd = ******` using `vi /home/hwcfg.ini`

Hi Mrs...Is need to install the hack or we can set with a standard?
Can you explain more how you did this?
Thank you

lukasz-glowacki, 2021-08-03:
> Hi Mrs...Is need to install the hack or we can set with a standard?
> Can you explain more how you did this?
> Thank you

You don't need to install hack to set or change rtsp password. Just telnet to camera with password cxlinux and edit mentioned file by vi editor.

bisewski001, 2021-08-04:
> > Hi Mrs...Is need to install the hack or we can set with a standard?
> > Can you explain more how you did this?
> > Thank you
> 
> You don't need to install hack to set or change rtsp password. Just telnet to camera with password cxlinux and edit mentioned file by vi editor.

Hi Lukasz...I open the file in Telnet but this is the text inside:
model = CloudCamera
main_bps = 768
sensor_position = 0
support_ptz = 1
ptz_mcu = 1
support_eth = 1
sound_detect = 1
ir_detect_type = 2
adc_setting_max = 700
adc_setting_min = -300
support_onvif = 1

As you can see, I dont have the passwd field...Can I put this field without problem? How can I save this file edit ?

Thank you Lukasz....

lukasz-glowacki, 2021-08-04:
> As you can see, I dont have the passwd field...Can I put this field without problem? How can I save this file edit ?

Yes, just add new entry in config file:
passwd = yourpassword

You may google how to use vi editor.