ant-thomas/zsgx1hacks #43: Configurable permanent change to camera

m4no, 2018-03-26:
Hi, great work. 
I don't wont to be rude, but I think it is better to make the zsgx1hacks-v0.3.zip configurable. If someone want to have permanent change to camera or not. Thank you.

ant-thomas, 2018-03-26:
It's not rude at all!
I will put together a non-permanent version soon. But the configurable idea is probably best.

reglisse69, 2018-03-27:
Hi, thanks a lot for this new release. Works great on ZSGX1 and noise has disapeared for me. Thanks.

Just one thing how WebUI works ? Seems to respond on 80 but nothing on the browser (IE, Edge, Chrome, Firefox)... But not really a function needed for me.

ant-thomas, 2018-03-27:
It can't use port 80 as the p2pcam software already opens port 80.
It is port 8080

The address for the webui is `http://ipaddress:8080/cgi-bin/webui`