ant-thomas/zsgx1hacks #45: Compile or repository?

crazygsm, 2018-03-30:
Hi all,

Was also doing some tests do full-fill some needs I have.

Currently using ffserver + ffmpeg to convert RTSP stream into mpeg and could open in any browser.

Are you compiling the bin files (busybox, httpd, etc) or exist any repository with pre-build packs for this armv6?

br
Alex

ant-thomas, 2018-03-30:
Hi,

Are you using ffmpeg on the camera itself? 
Because I've tried that and it caused the camera to crash, I think from a lack of memory possibly.

Busybox precompiled can be found here - https://busybox.net/downloads/binaries/1.21.1/
httpd is part of busybox
I believe I lifted the dropbearmulti binary from fang-hacks - https://github.com/samtap/fang-hacks/blob/0d2913f11b6bcf471f97e7bb29e6d4b5777c1dbf/sbin/dropbearmulti

crazygsm, 2018-03-30:
Hi,

Not in the camera itself, this is next step intend to work for and the reason for my question, if there is any repository to get those packagers instead compile it.

Currently I am using my router (OpenWRT) where ipcam is conected trough WIFI as server.

installed ffserver and ffmpeg in router

/etc/ffserver.conf

Besides basic inputs I am using this config

_<Feed monitoring1.ffm>
File /mnt/usb/monitoring1.ffm  # In my case used external storage coz have not enough space in /tmp
FileMaxSize 50M
ACL allow 127.0.0.1
ACL allow 192.168.0.0 192.168.0.255  # Allow your network
</Feed>

# Define according your needs
<Stream monitoring1.mjpg>
Feed monitoring1.ffm
Format mpjpeg
VideoCodec mjpeg
VideoFrameRate 5
VideoBufferSize 80
VideoSize 620x324
NoAudio
</Stream>_

Start ffserver
_# ffserver_

Start ffmpeg
_# ffmpeg -i "rtsp://admin:@<YOURIP>/0/av0" http://localhost:8090/monitoring1.ffm_

Now you can see stream in browser
_http://<YOUR FFServer IP>:8090/monitoring1.mjpg_

If you have the ffserver+ffmpeg already compiled for armv6l can you share it to do some tests?
Probably I will need to use an SD card to mount there the overlay and additional storage, in terms of RAM and CPU believe is more than capable.

ant-thomas, 2018-03-30:
To give you an idea of the RAM constraints
```             total         used         free       shared      buffers
Mem:         31908        30924          984            0         3676
-/+ buffers:              27248         4660
Swap:            0            0            0
```
Less than 1MB free whilst p2pcam is running.

FFMPEG for ARM - https://johnvansickle.com/ffmpeg/
This one runs - https://johnvansickle.com/ffmpeg/releases/ffmpeg-release-armel-32bit-static.tar.xz

crazygsm, 2018-03-30:
will do some tests.

Now I got surprised, never supposed system runned with more than 10Mb on ram usage, just tested and indeed, 1Mb free, killing p2pcam releases around 9Mb.

From my test in openwrt both take around 16Mb of ram,

crazygsm, 2018-03-30:
Need to found a way to make it run slimmer  :dancing_women: 

Manage at least to do snapshots 

./ffmpeg -i rtsp://admin:@127.0.0.1/0/av1 -vframes 1 -r 1 /mnt/mmc01/0/1111.jpg

crazygsm, 2018-03-30:
yes, using the package from link you posted and the command from my last comment.

Is very slow and I needed to use /0/av1 to have smaller resolution but it created the snapshoot.

Output #0, image2, to '/mnt/mmc01/0/1111.jpg':
  Metadata:
    title           : h264.mp4
    comment         : TAS-Tech Live Cast
    encoder         : Lavf57.83.100
    Stream #0:0: Video: mjpeg, yuvj420p(pc), 640x360, q=2-31, 200 kb/s, 1 fps, 1 tbn, 1 tbc
    Metadata:
      encoder         : Lavc57.107.100 mjpeg
    Side data:
      cpb: bitrate max/min/avg: 0/0/200000 buffer size: 0 vbv_delay: -1
frame=    1 fps=0.0 q=3.9 Lsize=N/A time=00:00:01.00 bitrate=N/A speed=0.0211x    
video:12kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: unknown

ant-thomas, 2018-03-30:
Ah yes, I tried with the full resolution. That must have been too much!

crazygsm, 2018-03-30:
Haven't had yet time to read all research done here but remembered to read something about replace p2pcam in what step is?

Had other idea to explore as a long shoot, replace current Cam engine (p2pcam) for motion?

Only question is if motion can recognize the cam coz if yes is all solved.

ant-thomas, 2018-03-31:
p2pcam hasn't been replaced yet.