ant-thomas/zsgx1hacks #46: HELP :)

mpoupou, 2018-03-30:
Hello ! ... I hope I can find help from you ... I have no idea how to communicate with you and how this site works. I came across your post trying to change the way to register on google drive or ftp instead of the paid cloud. I have a CIPC-Gc13H cloud ip camera.

I was able to record on my SD card only.

I downloaded the last file zsgx1hacks-v0.3.zip ... unzip and put on my SD card, insert in my camera, restart this one

Now ... I do not know what to do !! LOL I'm not even able to come back and use as before! I downloaded Putty and tried to communicate with the camera .. also try onvid app on android.

Can you help me please! ..thank you

I do not even know if I'll be able to read you if you answer me ..

ant-thomas, 2018-03-30:
@crazygsm - I think you're right. I'm working on it now.

mpoupou, 2018-03-30:
Camera Version
Firmware:
3.2.8.0121
Embedded Application:
3.2.8.0121
Device ID:
AJWL171020121S3CNUX84C53CD036029
MAC Address:
00:0C:43:76:17:CE
Model Type:
gk7102s_ytj_A0

thank you for your quick answers !!

good ... ok I was able to connect with my computer to the website eyplusiot and I see the image! .. I can not see recording but only live.

Then for the application..I now see the image, and it records, but if I go in the SD card option, it tells me that there is no card.

Better than nothing.

Now I would like to know if it is possible to record directly on my ftp or google drive and how. :) I'm not super knowing

thank you