ant-thomas/zsgx1hacks #47: How to set up wifi using telent?

franki29, 2018-03-31:
Hi all , great work. 
I like to setup the wifi via telnet. Does anyone know how to do?

THX

ant-thomas, 2018-03-31:
Unfortunately this is currently tied to the p2pcam program, so you need to setup first that way and then install the hack.

crazygsm, 2018-03-31:
Hi,

Unfortunetly I have configured first time using the app, trying to understand how it works, however by now is possible to config the wifi in this way

go to /home/ and edit/add the file 'wpa_supplicant.conf' according to the wifi you wanna connect.

After simply run this command (you need to be at /home/ directory)

wpa_supplicant -Dwext -iwlan0 -c /home/wpa_supplicant.conf

franki29, 2018-04-02:
Hi, thanks for answer. The wpa_supplicant command seems to work but after reboot, the settings are gone. Editing the startup.sh file does not help, as the w-lan is not asking for a DHCP address.

crazygsm, 2018-04-02:
if you use the hack from this git it works even after reboot, tested by me.

tonbor, 2019-03-09:
There is no link to "use the hack from this git " and is it working for the RO version?

crazygsm, 2019-03-12:
I don't think this have any relation between RO version and others.

You can also do it in different way.
Create a script to be called 1min after finish boot to replace the file "wpa_supplicant.conf" by yours and run the command.

kamilmirza, 2019-03-12:
here is another solution to connect to WiFi, works on RO & RW fw version
create cls.conf file on sdcard root
and paste the following with your WiFi SSID & pass

[cls_server] 
ssid = SSiD_Wi-Fi 
passwd = password_WiFi

source: https://4pda.ru/forum/index.php?showtopic=928641&view=findpost&p=82237201

tonbor, 2019-03-12:
Ssid en password plain or between quotes

kamilmirza, 2019-03-12:
plain without quotes 

[cls_server]
ssid = mywifiname
passwd = mywifipassword

tonbor, 2019-03-12:
Thanks, I will try it out.