ant-thomas/zsgx1hacks #51: Second time WiFi setup without application (same camera, same login)

xjikka, 2018-04-06:
During first time setup WiFi over App just make screenshot with QR code, next time just point camera to this screenshot and WiFi is set up. (Content of QR is crypted, maybe someone found a way to generate it).

lukasz-glowacki, 2019-01-16:
Isn't QR Code just MAC address?

xjikka, 2019-01-16:
No, its not so simple. QR Code has 80 byte (non ASCII 8bit numbers)
Here my QR example: 43 b5 25 f0 a5 e4 c2 c6   a5 43 04 25 82 d2 55 a3   ... plus another 64 bytes like these...

lukasz-glowacki, 2019-01-17:
In my case (Digoo DG-W01f) QR Code generated from MAC address (eg. 1122334455) allowed me to pair again camera with my account using Digoo app (I lost original sticker). Morover QR code with MAC address 11:22:33:44:55 Digoo app interpreted as set of 5 addresses divided by colon and tried to find it.

I have another IP camera where QR code is plaintext serial number. So maybe it depends on brand.