ant-thomas/zsgx1hacks #53: V0.4 works on GD-SC02

hitjock, 2018-04-10:
V0.4 permanent installed on my GD-SC02, works perfectly.
Only issue, I cannot connect to SSH by using putty.... gives me a key algorithm error...still trying to find a solution.
Thanks for the hard work. I will contribute if I find anything helpful.
Cheers!

xjikka, 2018-04-10:
On DG-SC03 same issue, cannot connect to SSH (22).  I'm getting PuTTY Fatal Error "Server unexpectedly closed network connection". Over telnet (23) works OK.

fbertone, 2018-05-08:
for me it doesn't seem to be working on GD-SC02.
in serial console during boot I see this log:

```
[...]
exec factory_tool.sh .......
ls: /mnt/*-hwcfg.ini: No such file or directory
ls: /mnt/*-VOICE.tgz: No such file or directory
ls: /mnt/*-ptz.cfg: No such file or directory
ls: /mnt/*-hardinfo.bin: No such file or directory
[...]
---------open /mnt/mmc01/cls.conf failed
[...]
rd storage = /mnt/mmc01/0
local_create_root, root_path = /mnt/mmc01
MountLocalPartitions......
try mount t=vfat dev=/dev/mmcblk0p1 mp=/mnt/mmc01/0...---------open /opt/upgrading failed
[...]
local_create_root, root_path = /mnt/mmc01/0
open .recording: /mnt/mmc01/0/.recording
```
the SD get correctly mounted and is used for recording, I can see the files downloaded from this repo but apparently they don't get called.
which script is supposed to run the firmware from SD?

I'll try checking inside my bash scripts to see if I can find any reference to mounted FSs