ant-thomas/zsgx1hacks #54: File devParam.dat

xjikka, 2018-04-10:
I've found, that devParam.dat is one of files, which are changed when WiFi changed over App. In this binary file is stored some data and some parts contain WiFi SSID + Pass and file checksum. Camera (tested on SC03) on start checks checksum and if data in file changed [e.g. SSID], crc is not OK and file is deleted. Therefore I thought, that WiFi is initialized from this file. I've found a way, how to calculate checksum for changed WiFi data in file. After data change and with new calulated crc, file is no more deleted, but unfortunately nothing is changed. Old WiFi connection is still used. If somebody want, how to calculate checksum, please let me know.

kotfarepra, 2018-04-10:
Interesting . 
Please post how to calculate checksum so it would be accessible to anyone that is interested

xjikka, 2018-04-11:
Here the CRC calculation (delphi)
```
program datcrc;

{$APPTYPE CONSOLE}

{$R *.res}

uses
  System.SysUtils;

procedure BufWriteData(buf:PByte; bufsz, from, sz:Int64; const data:string);
var rb:RawByteString;
begin
  if bufsz<from+sz then raise Exception.Create('Bad data size');
  fillchar(PByte(buf+from)^,sz,0);
  rb:=RawByteString(data);  //Unicode string to 8bit string
  if length(rb)<sz then sz:=length(rb);
  Move(rb[1],PByte(buf+from)^,sz);
end;

//devParam.dat  size 3*4 + 1024 = 1036
// 0x0000   UINT       0x20150910     date ???
// 0x0004   UINT       data crc       data word sum - 32bit sum of 16bit words
// 0x0008   UINT       data size      0x00000400  = 1024
// 0x000C   char(size) data
// ...
// 0x0128   pansichar  SSID
// ...
// 0x0154   pansichar  password
// ...
// 0x01FC   pansichar  username      (admin)???
// ...

procedure UpdateWiFi(const fname,ssid,pass:string);
var fsz,bsz,cnt,i,crc:longint;
  fh:THandle;
  buf:PByte;
const
  CRCPOS    = $0004;
  SIZEPOS   = $0008;
  DATASTART = $000C;
  SSIDPOS   = $0128;
  SSIDSZ    = $20;  // not 100% sure, maybe more
  PASSPOS   = $0154;
  PASSSZ    = $20;  // not 100% sure, maybe more
begin
    fh:=FileOpen(fname, fmOpenReadWrite);
    if (fh> 0) then begin
    try
      fsz:=FileSeek(fh, 0, 2{soFromEnd});
      buf:=AllocMem(fsz);
      try
        FileSeek(fh,0,0{soFromBeginning});
        bsz:=FileRead(fh,buf^,fsz);
        //change data
        BufWriteData(buf,bsz,SSIDPOS,SSIDSZ,ssid);
        BufWriteData(buf,bsz,PASSPOS,PASSSZ,pass);
        //get crc
        cnt:=bsz div 2; //word count
        crc:=0;
        for i := (DATASTART div 2) to cnt do begin
          crc:=crc + (buf[i*2]+buf[i*2+1]*256);
        end;
        //update crc
        //CopyMemory(PByte(buf+CRCPOS),@crc,4);
        Move(crc,PByte(buf+CRCPOS)^,4);
        //update file
        FileSeek(fh,0,0{soFromBeginning});
        if FileWrite(fh,buf^,bsz)=bsz then begin
          writeln('File: '+fname);
          writeln('CRC:  '+IntToHex(crc));
        end;
      finally
        freemem(buf);
      end;
    finally
      FileClose(fh);
    end;
  end else raise Exception.Create('Cannot open file');
end;

var fname:string;
begin
  try
[devParamDatCRC.zip](https://github.com/ant-thomas/zsgx1hacks/files/1897751/devParamDatCRC.zip)

    if ParamStr(1)='' then begin
      writeln('Usage datcrc.exe SSID password [filename]');
      writeln('(filename is optional, if not defined, devParam.dat used)');
      exit;
    end;
    fname:=paramstr(3);
    if fname='' then fname:='devParam.dat';
    UpdateWiFi(fname,ParamStr(1),ParamStr(2));
  except
    on E: Exception do
      Writeln(E.ClassName, ': ', E.Message);
  end;
end.
```

xjikka, 2018-04-11:
[devParamDatCRC.zip](https://github.com/ant-thomas/zsgx1hacks/files/1897756/devParamDatCRC.zip)

deorder, 2018-04-19:
Cool. Thanks. If you want you can add it to:
https://github.com/ant-thomas/zsgx1hacks/blob/master/source/goke_p2pcam_param.c

xjikka, 2018-04-19:
I'm pascal developer, maybe someone with C experiences helps? It is simple (fortunately) 32 bit sum of 16bit words from 0x000c to end (size 0x0400).   
crc = word0+word1+ word2+word3+...+word1023
  word1 = data[0x000c]+0x0100 * data[0x000d]    
    _(or word1 = data[0x000c]+ data[0x000d] shl 8 )_  
  word2 = data[0x000e]+0x0100 * data[0x000f]
  ....
  word1023 = data[0x040a]+0x0100 * data[0x040b]

CRC stored:

data[0x0004]=(crc & 0x000000ff)            [lower byte of crc]
data[0x0005]=(crc & 0x0000ff00) shr 8
data[0x0006]=(crc & 0x00ff0000) shr 16
data[0x0007]=(crc & 0xff000000) shr 24  [higher byte of crc]

deorder, 2018-04-20:
Okay. I will add it to the C code then. Thanks for figuring that out.

Briggsbfc, 2018-11-05:
I got an EOutOfMemory error when running the datcrc.exe from the command line on a 32-bit Windows 7 machine. Is there an issue with the binary?

Briggsbfc, 2018-11-05:
I created an empty file and ran the program with the file name as the third parameter and now get a message: Exception: Bad data size

Briggsbfc, 2018-11-05:
Leave this with me, I think I understand now, the program doesn't create a new devParam.dat file, it modifies an existing file.

xjikka, 2018-11-05:
Briggsbfc has right. If file doesn't exists or is not specified, Exception Out of Memory is thrown.

xjikka, 2018-11-05:
Updated source and binary, added exception file not found 
[devParamDatCRC_2.zip](https://github.com/ant-thomas/zsgx1hacks/files/2548955/devParamDatCRC_2.zip)

Damien0505, 2019-06-18:
Hi, maybe slightly off topic, but can goke_p2pcam_param be used to set the wifi password?
Thanks in advance,