ant-thomas/zsgx1hacks #56: Is the rtsp connection disabled after the trial period?

Hagiman2000, 2018-04-20:
There were a lot of comments on websites like "mydealz" where users posted older firmwares for a GUUDGO GD-SC03 where rtsp was enabled (unlike in the original firmware of the shipped version). So far so good. But after the trial period of 14 days this access was disable (even if you blocked all outside communications).

Is this hack different? Do I need a special firmware version?

The result should be, that I can use a rtsp stream for a unlimited amount of time. Is this the case?

deorder, 2018-04-21:
For me this is not the case. If not you can probably reset it using the files provided here.

cptX, 2018-05-31:
I also have a problem with RTSP. Although I suspect the problem to be on the encoding of the RTSP that my camera does. How can I reflash another firmware?