ant-thomas/zsgx1hacks #60: Didn't work in ZS-GX1 1080p

ducapedia, 2018-05-08:
Hi..
I've been trying this hack on my ZS-GX1 (in the box says it's 1080p and 2Mp with 1/2.7, so I'm guessing GK7101 and SC2135 ?), but when I've booted with 0.4 version in my 64gb microSD card (well,, windows10 don't allow to format in FAT32 anymore, but exFAT it's almost the same), none of these functions worked (WebUI, FTP, Telnet, SSH, all of this had no response from camera IP).
When I toke off the SDcard and read it in my notebook, I could saw many video files (extensio .x264, but I couldn't play these in any player, including VLC).
Your or anyone who is reading, have some hint to install in this kind of camera ?
Thanks in advanced.

ducapedia, 2018-05-09:
Oh well... I've tried again last night, but now I formated sdcard in Linux... (mkfs.vfat), and copy zsgx1hack-0.4 uncompressed to it.
In this try, FTP (with no password), webgui with PTZ (but no image)... but, RTSP didnt work at all... and website eyeplusiot.com, shows a message that WebCam was turned off, and didnt show me any image or recording.
Also, my fianceè says that camera was emitting a very pitch noise (but I didnt hear it, maybe a very high frequency witch only younger can hear).
Anyone had similar problems with this model ?
In Camera details show "Camera Model" as "CloudCamera"

kotfarepra, 2018-05-09:
@ducapedia At least you know what to turn on when you get angry with her :)

ducapedia, 2018-05-09:
LOL