ant-thomas/zsgx1hacks #62: Doesnt work on latest GUUDGO GD-SC03 Snowman

potchin, 2018-05-10:
Looks like the firmware shipped on the GUUDGO GD-SC03 Snowman 1080P has some changes which prevent the hacks from loading.

My first camera (bought February) works fine but the one I ordered at the end of April has the root filesystem mounted readonly..

```
# mount
rootfs on / type rootfs (rw)
/dev/root on / type squashfs (ro,relatime)
proc on /proc type proc (rw,relatime)
tmpfs on /dev type tmpfs (rw,relatime)
tmpfs on /tmp type tmpfs (rw,relatime)
sysfs on /sys type sysfs (rw,relatime)
tmpfs on /tmp type tmpfs (rw,relatime)
devpts on /dev/pts type devpts (rw,relatime,mode=600,ptmxmode=000)
/dev/mtdblock3 on /home type jffs2 (rw,relatime)
/dev/mmcblk0p1 on /tmp/mnt type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=cp437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p1 on /media type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=cp437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p1 on /tmp/mnt/mmc01/0 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=cp437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
```

Its cant be remounted read-write..

```
# touch /test
touch: /test: Read-only file system
# mount -o remount,rw /
# touch /test
touch: /test: Read-only file system
# mount -o remount,rw /dev/root /
# touch /test
touch: /test: Read-only file system
# 
```

Until this is sorted I would recommend that nobody update the firmware of their camera, or buy a new one :(

potchin, 2018-05-10:
Oh dear. I ran `./persistenthack.sh` as /home was mounted rw but its not gone well. No errors but the camera appears to be in a reboot loop (can hear the clicking as it restarts) and its not appearing on the network.

dwarf2k, 2018-05-10:
Re: [ant-thomas/zsgx1hacks] Doesnt work on latest GUUDGO GD-SC03 Snowman (#62)

Hello potchin,

Thanks for the warning, but it is too late!  I ordered 2 days ago since I wanted a cheap camera I could keep off the cloud and hit directly.  Please keep us informed as to what you find and good luck.

-Scott

Thursday, May 10, 2018, 2:39:21 PM, you wrote:

Looks like the firmware shipped on the GUUDGO GD-SC03 Snowman 1080P has some changes which prevent the hacks from loading.
My first camera (bought February) works fine but the one I ordered at the end of April has the root filesystem mounted readonly..
# mount
rootfs on / type rootfs (rw)
/dev/root on / type squashfs (ro,relatime)
proc on /proc type proc (rw,relatime)
tmpfs on /dev type tmpfs (rw,relatime)
tmpfs on /tmp type tmpfs (rw,relatime)
sysfs on /sys type sysfs (rw,relatime)
tmpfs on /tmp type tmpfs (rw,relatime)
devpts on /dev/pts type devpts (rw,relatime,mode=600,ptmxmode=000)
/dev/mtdblock3 on /home type jffs2 (rw,relatime)
/dev/mmcblk0p1 on /tmp/mnt type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=cp437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p1 on /media type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=cp437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p1 on /tmp/mnt/mmc01/0 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=cp437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
Its cant be remounted read-write..
# touch /test

touch: /test: Read-only file system

# mount -o remount,rw /

# touch /test

touch: /test: Read-only file system

# mount -o remount,rw /dev/root /

# touch /test

touch: /test: Read-only file system

# 

Until this is sorted I would recommend that nobody update the firmware of their camera, or buy a new one :(

-- 
Best regards,
 Scott                            mailto:sav@scottv.net

potchin, 2018-05-26:
Dont flash the firmware in the link above.. seems to have bricked my camera. It no longer connects to my network, even via ethernet.

Does anyone have a link for the original firmware? Hoping I can rescue it

roleoroleo, 2018-05-26:
I flasked that firmware several times without problems.
After flashing I only had to repeat the installation procedure.

potchin, 2018-05-26:
Not entirely sure how but ive got a little further - it appeared back on the network. I forced flashed the firmware from the link above..

```
# /bin/sdc_tool -d $BOARD_ID -c ruishi-xiaoxueren-hwcfg.ini  firmware.bin
Update from firmware.bin ...
PUBTIME: 0x599ECEB7
# echo $?
0
```
But / is still mounted read only. rtsp has also stopped working and I dont see the camera connecting to the cloud - so the app on my phone is useless :(

roleoroleo, 2018-05-26:
Probably it depends on the previous firmware version.
Are you able to access in telnet?

potchin, 2018-05-26:
Thankfully yes, just need a way to flash everything back to defaults. Any suggestions?

potchin, 2018-05-27:
```
# ps
  PID USER       VSZ STAT COMMAND
    1 root      1064 S    {linuxrc} init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW   [kworker/u:0]
    6 root         0 SW<  [khelper]
    7 root         0 SW<  [netns]
    8 root         0 SW   [sync_supers]
    9 root         0 SW   [bdi-default]
   10 root         0 SW<  [kblockd]
   11 root         0 SW   [khubd]
   12 root         0 SW   [kworker/0:1]
   13 root         0 SW<  [cfg80211]
   14 root         0 SW<  [rpciod]
   15 root         0 SW   [kswapd0]
   16 root         0 SWN  [ksmd]
   17 root         0 SW   [fsnotify_mark]
   18 root         0 SW<  [nfsiod]
   19 root         0 SW<  [crypto]
   32 root         0 SW   [mtdblock0]
   33 root         0 SW   [mtdblock1]
   34 root         0 SW   [mtdblock2]
   35 root         0 SW   [mtdblock3]
   36 root         0 SW   [kworker/u:1]
   39 root         0 SW<  [deferwq]
   40 root         0 SW   [kworker/u:2]
   73 root         0 SWN  [jffs2_gcd_mtd3]
   86 root         0 SW   [gk_debug]
   87 root         0 SW   [encoder]
  169 root      1064 S    {start.sh} /bin/sh /home/start.sh
  173 root      168m S    ./p2pcam
  174 root      1072 S    -sh
  176 root      1064 R    /usr/sbin/telnetd
  197 root      1060 S    udhcpc -i eth0 -b -s /usr/share/udhcpc/default.script -h ipc
  200 root         0 SW   [flush-mtd-unmap]
  219 root      1072 S    -sh
  223 root      1064 R    ps
```

```
# cat /home/start.sh
#!/bin/sh

getHwInfo()
{
        grep $1 /home/hardinfo.bin | awk -F '>'  '{print $2}' | awk -F '<' '{print $1}'
}

getHwCfg()
{
        grep $1 /home/hwcfg.ini | awk '{printf $3}'
}

#check if stop app auto run
read -t 1 -p "Press 'q' in 1 seconds to exit: " q
if [ $? -eq 0 -a "$q" = "q" ]; then exit; fi

rm /bin/wpa_supplicant
ln -s /home/wpa_supplicant /bin/wpa_supplicant
cp -f /home/tees /bin
chmod -R 777 /home

#init isp
/home/sensor.sh

#check if old sc1135
#or if ii2 3229 is 2
mirror_type=$(getHwCfg gk_mirror_type)
if [ "$mirror_type" = "1" ]; then   
echo "sensor do not support vi mirror! check sensor_hw.bin"
mv -f /home/sensors/sc1135_hw.bin_old_10bits /home/sensors/sc1135_hw.bin
fi

#check if 1080, modify uboot args
/home/check_mem.sh

#drivers
insmod /home/gio.ko && mdev -s

#run custom init for board OEM
/home/custom_init.sh

BOARD_ID=$(getHwInfo BoardType)

#mount SD card
if [ -b /dev/mmcblk0p1 ]; then
        mount -t vfat /dev/mmcblk0p1  /mnt
elif [ -b /dev/mmcblk0 ]; then
        mount -t vfat /dev/mmcblk0 /mnt
fi

#upgrade firmware
if [ -f /mnt/firmware.bin ]; then
        sdc_tool -d $BOARD_ID -c /home/model.ini /mnt/firmware.bin

        #check upgrade from OTA or factory test
        if [ -f /mnt/OTA ]; then
                rm /mnt/firmware.bin
                rm /mnt/OTA
        else
                touch /opt/upgrading
        fi
fi

#Run facoty_tool.sh for burn id and change voice and change hwcfg.ini
/home/factory_tool.sh

#Run debug_cmd.sh
if [ -f "/mnt/debug_cmd.sh" ]; then
        echo "find debug cmd file, wait for cmd running..."
        /mnt/debug_cmd.sh
fi
umount /mnt

#update form flash
if [ -f /home/firmware.bin ]; then
        /bin/sdc_tool -d $BOARD_ID /home/firmware.bin
        if [ $? -eq 0 ]; then
                echo "upgrade success."
        else
                echo "upgrade failed."
        fi
        rm -f /home/firmware.bin
fi

#network init 
wifi_type=$(getHwCfg wifi_type)
if [ "$wifi_type" = "7601" ]; then 
insmod /home/mt7601Usta.ko
else
insmod /home/8188fu.ko
fi
sleep 1
ifconfig lo 127.0.0.1
ifconfig wlan0 up
ifconfig ra0 up
ifconfig eth0 up

#check if old sc1135
#or if ii2 3229 is 2
mirror_type=$(getHwCfg gk_mirror_type)
if [ "$mirror_type" = "1" ]; then
echo "sensor do not support vi mirror! check sensor_hw.bin"
mv -f /home/sensors/sc1135_hw.bin_old_10bits /home/sensors/sc1135_hw.bin
fi

#run tees for debug info
tees -s -v -b 20 -e ps -e 'ifconfig; route -n' -e 'wpa_cli status' -e 'mount' -e 'uptime' -e 'df' -e 'netstat -napt' -e free -a /tmp/closelicamera.log -o /mnt/mmc01/1/ipc.log -o /mnt/mmc01/0/ipc.log -O /tmp/upipc.log & (cat /proc/kmsg | /tmp/tees) &

#upzip files
tar -zxf /home/p2pcam.tar.gz -C /tmp
tar xzf /home/VOICE.tgz -C /tmp
if [ -f /tmp/VOICE/OVERSEA ]; then
cp /home/cloud_oversea.ini /tmp/cloud.ini
else
cp /home/cloud.ini /tmp
fi
cp /home/ca-bundle-add-closeli.crt /tmp

#init ptz
ptz_mcu=$(getHwCfg ptz_mcu)
has_ptz=$(getHwCfg support_ptz)
if [ "$ptz_mcu" = "1" ]; then
        mv -f /home/gkptz-dsa.ko /home/gkptz.ko
fi

if [ "$has_ptz" = "1" ]; then
        if [ -f /home/silent_reboot ]; then NO_SLFCK=1; rm /home/silent_reboot; else NO_SLFCK=0; fi
        insmod /home/gkptz.ko cfg_file=/home/ptz.cfg psp_file=/home/psp.dat no_selfck=$NO_SLFCK
fi

#wpa_supplicant -B -iwlan0 -c /home/wpa_supplicant.conf_EYERD &

mdev -s

cd /tmp
(
export CLOSELICAMERA_LOGMAXLINE=1000
./p2pcam; killall -10 tees)&
(
sleep 5
rm -f p2pcam

sync; echo 3 > /proc/sys/vm/drop_caches
free
)&
```

roleoroleo, 2018-05-27:
Strange.
p2pcam is running, so the start procedure completes.
We could try to re-enable the internet connection and try to update the firmware with the phone app.
Back up your /etc/hosts and delete all rows except
127.0.0.1       localhost
127.0.1.1       goke

potchin, 2018-05-28:
The hosts file is unmodified - the rootfs id readonly ;)

roleoroleo, 2018-05-28:
I don't know it it can help, but I could share a dump of my mtdblocks.

potchin, 2018-06-03:
Its ok, I pulled a dump of them off my working camera. No luck. Its dead, doesnt even boot anymore. Will try cracking it open and seeing if I can get serial access to it.

closed 2018-06-03 by potchin

potchin, 2018-06-03:
Just thought I'd comment to say I managed to salvage my dead camera. Curiosity got the better of me and I hooked up to the internal serial port (its labelled on the board).

The camera runs u-boot unsurprisingly so I flashed a copy of mtdblock1 from my working camera and it was able to boot. I couldnt get the mtdblock2 to flash from uboot so just restored that via a tarball on the sd card since its mounted rw.

For anyone else who gets stuck into this mess, heres my uboot commands..

```
GK7102 # set ipaddr 192.168.0.42
[PROCESS_SEPARATORS] set ipaddr 192.168.0.42
GK7102 # set serverip 192.168.0.2
[PROCESS_SEPARATORS] set serverip 192.168.0.2
GK7102 # sf probe 0
[PROCESS_SEPARATORS] sf probe 0
SF:    16 MiB [page:256 Bytes] [sector:64 KiB] [count:256] (XM25QH128)

GK7102 # sf erase 0x000000050000 0x0000002b0000
[PROCESS_SEPARATORS] sf erase 0x000000050000 0x0000002b0000

Erasing:
    100%

GK7102 # tftp mtdblock1
[PROCESS_SEPARATORS] tftp mtdblock1
TFTP from server 192.168.0.2; our IP address is 192.168.0.42
Filename 'mtdblock1'.
Load address: 0xc1000000
Loading: #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 #################################################################
	 ################################
done
Bytes transferred = 2490368 (260000 hex)

GK7102 # sf write 0xc1000000 0x000000050000 0x2B0000
[PROCESS_SEPARATORS] sf write 0xc1000000 0x000000050000 0x2B0000

Programming:
    100%

Verifying:
    100%
```

And used [this post](https://felipe.astroza.cl/hacking-hi3518-based-ip-camera/) as inspiration.

@ant-thomas It may be worth warning people about this. If they get a camera with a readonly root filesystem its a bit of a pain to fix. Id also recommend not updating the firmware via the app :)

roleoroleo, 2018-06-04:
Good job!

ljs, 2018-07-11:
Hi @potchin, @ant-thomas and the other guys taking part in the discussion. 

Great project and nice work has been done tilll now by @ant-thomas and contributors.

Normally the manufacturer now tries to prevent such workarounds. But I wanted to share my experience.

I also got a new GUUDGO GD-SC03 and it is already with new firmware, which has the above mentioned "limitations", in particular the read-only root /. This can be solved easily with mount binds for example. So, I did:

`mount --bind /media/hack/hosts.new /etc/hosts`
`mount --bind /media/hack/busybox-armv6l /bin/busybox`

This prevented the cloud connection and enabled me to use the webUI with ptz control. So, I can watch live the rtsp stream. One can play more with bindings (of whole dirs) and tweak and fine tune, I wanted just a PoC.

The real problem for me is that the camera does not record anything on the sd card, unless there is cloud connection. Moreover if the camera has the cloud access disabled, but you start the WiFi app, the wifi app takes care to upload the sd card files into the cloud, and I think the camera is again in recording mode. 

I don't know if this was the behavior also with the previous firmwares? I would like to use it for event based recording on SD without cloud. Seems if for some reason one does not have Internet connection (e.g. link is down on a remote location), there is not much sense of the SD card, as it won't be used...  (OR, I haven't performed these tests yet, might be that the manufacturer checks for the hacks, e.g. dns resolution to 127.0.0.1 of his cloud servers etc, and only then it does not record to sd..).

My current firmware is:
```
Firmware
3.2.8.0208
Embedded Application
3.2.8.0208
````

Is there a way to downgrade the firmware to another version that can do event based records on the SD card? 

If this is not clear how to be done the other way would be to make the camera believe it is connected to the cloud, or at least make the recording feature running again.

Would be happy to hear if there is somebody alive in this discussion :)

Regards,

P.S. I don't know since this issue is closed if ppl participating in the discussion and also the tagged ones will get notification for new posts (e.g. mine).

P.P.S. Would somebody of you recommend a similar product/camera that has the same functionalities (wifi, event based sd record, IR sensor for night mode, etc), but is not so vendor locked, even at a higher price. 

Before somebody gave me this product I was thinking for options of DIY cameras based on raspberry pi for example, but it seems as too much hassle and time dedication that I don't currently seem to have. 

Thanks in advance if some ideas would pop-out!

sensegsm, 2018-08-05:
Good job #potchin.   Can you upload mtdblock1 for GK7102 ? I have same problem but i dont find anywhere  mtdblock1. Thanks in advance.

ant-thomas, 2018-08-05:
@ljs Thanks for pointing out the mount binds trick. I have just bought another IP camera (CIPC-GC15HE) which has the same type of setup as the hacks I created, but is running an updated firmware and the root is read-only.

I've discovered the GPIO for IR cut, but not IR leds yet, not sure if they are controllable.

@potchin Does your reflashing downgrade the firmware and make it r/w again?

I've played with a totally different camera that turned out to also be read-only. Managed to dump the flash partitions and edit files on a computer, repack them and write them back.

Ideally my next step is to figure out what needs changing to be able to mount the new firmwares as r/w.

potchin, 2018-08-05:
@sensegsm Heres the mtdblock1 for my camera (GUUDGO GD-SC03) with the rw rootfs. Obviously make sure this is suitable for your camera before flashing..

https://www.dropbox.com/s/z9bw91et0eyjkm7/mtdblock1?dl=0

@ant-thomas I just flashed a backup from my old camera (which had the rw rootfs, link above). Have you tried editing the fstab on your custom filesystem? I dont think its anymore complicated than that

sensegsm, 2018-08-05:
Thanks very much @potchin . Unfortunately my cam is a chinese unbranded. I think is a ESCAM G02, but i `m not sure. After power on my cam, this test Pan/Tilt and after reboot itself. I can not connect to it by lan or wifi. Only way to connect to her is through the serial.
If you can help me in any way I would be grateful.

This is the serial log: 
U-Boot 2012.10 (Feb 24 2016 - 22:33:54) for GK7102 rb sc1045 v2.00 (GOKE)

HAL:  20151223
DRAM:  64 MiB
Flash: 32 MiB
32 MiB
NAND:  SPINAND MID = 0xff, DID = 0xffff, 1]: No support this SPI nand!
SF: Detected W25Q256FV with page size 256 B, sector size 64 KiB, total size 32 MiB
In:    serial
Out:   serial
Err:   serial
Net:   arm_freq(600MHz)..............0x112032
use int MII..............
gk7101
Hit any key to stop autoboot:  0
[PROCESS_SEPARATORS] run sfboot
[PROCESS_SEPARATORS] setenv bootargs console=${consoledev},${baudrate} noinitrd mem=${mem} rw ${rootfstype} init=linuxrc ;sf probe 0 0;sf read ${loadadSF: Detected W25Q256FV with page size 256 B, sector size 64 KiB, total size 32 MiB
put param to memory
mem size (41)
bsb size (2)

the kernel image is zImage or Image
entry = 0xc1000000
## Transferring control to Linux (at address c1000000)...

Starting kernel ...

machid = 3988 r2 = 0xc0000100
Uncompressing Linux... done, booting the kernel.
[    0.000000] Booting Linux on physical CPU 0
[    0.000000] Linux version 3.4.43-gk (bruce@ubuntu) (gcc version 4.6.1 (crosstool-NG 1.18.0) ) #33 PREEMPT Thu May 19 20:55:21 PDT 2016
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: VIPT aliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine: Goke GK7102 RB_SC1045 board V2.00
[    0.000000] Memory policy: ECC disabled, Data cache writeback
[    0.000000] AHB: 0x90000000  0xf2000000  -- 0x1000000
[    0.000000] APB: 0xa0000000  0xf3000000  -- 0x1000000
[    0.000000] PPM: 0xc0000000  0xc0000000  -- 0x200000
[    0.000000] BSB: 0xc2b00000  0xf5000000  -- 0x200000
[    0.000000] DSP: 0xc2d00000  0xf6000000  -- 0x1300000
[    0.000000] hal version = 20151223
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 10414
[    0.000000] Kernel command line: console=ttySGK0,115200 noinitrd mem=41M rw rootfstype=jffs2 root=/dev/mtdblock3 init=linuxrc
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Memory: 41MB = 41MB total
[    0.000000] Memory: 36636k/36636k available, 5348k reserved, 0K highmem
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
[    0.000000]     DMA     : 0xff600000 - 0xffe00000   (   8 MB)
[    0.000000]     vmalloc : 0x83000000 - 0xff000000   (1984 MB)
[    0.000000]     lowmem  : 0x80000000 - 0x82900000   (  41 MB)
[    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
[    0.000000]       .text : 0x80008000 - 0x8044d000   (4372 kB)
[    0.000000]       .init : 0x8044d000 - 0x8046e000   ( 132 kB)
[    0.000000]       .data : 0x8046e000 - 0x80497260   ( 165 kB)
[    0.000000]        .bss : 0x80497284 - 0x804c9888   ( 202 kB)
[    0.000000] NR_IRQS:128
[    0.000000] >> gk7101 init irq vic1...
[    0.000000] >> gk7101 init irq vic2...
[    0.000000] gk7101 init vic...
[    0.000000] mach gk7101 init timer...
[    0.000000] sched_clock: 32 bits at 100 Hz, resolution 10000000ns, wraps every 4294967286ms
[    0.000000] Console: colour dummy device 80x30
[... 268 lines cut ...]
HI_Media_SDKInit: HI_SDK_Init() error!
HI_Media_Init: init sdk failed!
HI_Websvr_Init: init media failed!
workthread: websvr start failed!
ipc_server stop: 1970-01-01 02:00:38
/mnt/mtd/ipc/platform.sh: line 14: ps: not found
/mnt/mtd/ipc/platform.sh: line 14: ps: not found
/mnt/mtd/ipc/sd.sh: line 14: ps: not found
umount: can't forcibly umount /mnt/mtd/ipc/tmpfs/sd: Invalid argument
[   42.200000] iFlag: 1, iCurStep: 255
/mnt/mtd/ipc/platform.sh: line 14: ps: not found
/mnt/mtd/ipc/platform.sh: line 14: ps: not found
/mnt/mtd/ipc/platform.sh: line 14: ps: not found
/mnt/mtd/ipc/platform.sh: line 14: ps: not found
/mnt/mtd/ipc/platform.sh: line 14: ps: not found
/mnt/mtd/ipc/platform.sh: line 14: ps: not found
/mnt/mtd/ipc/platform.sh: line 14: ps: not found
/mnt/mtd/ipc/sd.sh: line 14: ps: not found
/tmpfs/sd: Invalid argument
/mnt/mtd/ipc/platform.sh: line 14: ps: not found
found
/mnt/mtd/ipc/platform.sh: line 14: ps: not found
/mnt/mtd/ipc/platform.sh: line 14: ps: not found
/mnt/mtd/ipc/platform.sh: line 14: ps: not found
/

leeablett, 2018-09-29:
Does anyone have the latest firmware that works with this ?

Gymnae, 2019-01-27:
Having updated the firmware of my new GD-SC03, which only has 8MB of Flash, I ended up with a semi brick. I have access via telnet and UART, but no proper funtion.

Can anyone zip up their /home folder? I assume that it contains all files necessary for proper operation, inlcuding wifi drivers. My /home folder contains very little files, no drivers subfolders and no kernel modules.