ant-thomas/zsgx1hacks #63: Save the recordings to SD Card?

xlucasfelix, 2018-05-11:
How to set up to save the recordings to the SD Card?

miguipda1, 2018-05-19:
Hi,

I finaly get it work on my new Digoo DG-W01F Camera.

I have the same question. How to get motion detection recorded on the SD card ? Or where does it perhaps already record them because I did not find it.

Sincerely thanks.

miguipda1, 2018-06-07:
Hi,

please consider this as a needed request. It will allow the camera to be used offline and sync its content back to an identified external storage (laptop, raspberry, any IP storage) when it received an IP address.

ljs, 2018-07-13:
Hello again, 

Thanks for your reply too.

No problem, but I mistakenly assumed that you own a Guudgo SC-03, because of the first poster in this thread, so I was asking for firmware version of the Guudgo. I also own a SC-03, and it does not record to SD card when no connection to internet. Which contradicts with the advertised functions actually.... anyway.

I'm afraid I won't be able to assist in regards to your model Digoo DG-W01F, I hope there will be some solution for you too.

Have a nice day too!

L.

miguipda1, 2018-07-13:
Thank you Ljs

ttl76543, 2018-07-17:
I am using the ZS-GX1, and after booting from the hacked SD card, it also does not record locally to the SD card. Any suggestions? Without the cloud service, local recording is important.

miguipda1, 2018-08-13:
Hi everybody,

I suppose that without an answer from the developer we perhaps could find a part of our need [at this page](https://github.com/ant-thomas/zsgx1hacks/issues/80). It [WSDL](https://www.onvif.org/ver10/device/wsdl/devicemgmt.wsdl) point 14 GetCapabilities show it has a Capabilities - Extension - Recording and at the point 82 SetStorageConfiguration it also seems it could allow us to define where to store the motion detection.

So if someone found the way to set our desired store to the local DS card it should be appreciate to share it to us.

Have a nice day.

Miguipda

Cruiser79, 2018-08-15:
@Miguipda The idea was nice, but my cam don't send a Capabilities - Extension - Recording in the _GetCapabilities_ request. It is optional, says the WSDL. Also the _GetStorageConfigurations_ method is not implemented in my cam. So i thing, sadly this idea is dead.

sanket15n, 2018-10-01:
It would be awesome if there can be local video recording function on the SD card itself so we can get these recording via ftp or view the same on WebUI. 

It use to create the file " .recording" in SD card the content of the file was the name of latest ongoing video recording media file.  In readonly hack this file is empty and no recording is captured.

DBaluxa, 2018-12-19:
Hi everybody,

I've implemented the video save function via ffmpeg. You can find the modified code in my pull request: https://github.com/ant-thomas/zsgx1hacks/pull/93

Hope You find it usefull.

DBaluxa

pristis, 2020-11-30:
> Hi everybody,
> 
> I've implemented the video save function via ffmpeg. You can find the modified code in my pull request: #93
> 
> Hope You find it usefull.
> 
> DBaluxa

Hello! 

I was trying to record something, but nothing happens. Can you do some tutorial how to do it. Thanks!

Best regards,
Martin

DBaluxa, 2020-12-01:
Hello Pristis,

If You check out my modified version (it hasn't been merged yet, https://github.com/DBaluxa/zsgx1hacks), it will work automatically. After You start the camera with the modified code on the SD card, itt will start immadiatelly recording on the SD card.

Best regards,
Balazs

pristis, 2020-12-02:
> Hello Pristis,
> 
> If You check out my modified version (it hasn't been merged yet, https://github.com/DBaluxa/zsgx1hacks), it will work automatically. After You start the camera with the modified code on the SD card, itt will start immadiatelly recording on the SD card.
> 
> Best regards,
> Balazs

Hello!

Thanks for your time! I saw your modified code and on the SD card I put the content from "readonlysd" folder.
Unfortunately there are still no recordings. I can see the folder with today's date in Videos, but it's empty.
ffmpeg.log has many rows with: Wed Dec  2 18:26:48 GMT 2020 Killed

Also the cam reboots continuously after few minutes of working.

Best regards,
Martin