ant-thomas/zsgx1hacks #64: Snapahot

roleoroleo, 2018-05-14:
Is there a url to get a jpeg snapshot?
I tried onvif/snapshot, snapshot and other typical path but nothing.
I tried also at 8001 port.

roleoroleo, 2018-05-18:
Using a disassembler I found that the correct size is 320x180.
I patched p2pcam and now it works.

davidlb, 2018-05-18:
I (and probably others) would be interested by your patched p2pcam for taking snapshot. 
Can you provide it?

roleoroleo, 2018-05-18:
Sure.
I think p2pcam depends on the firmware version.
But I will share my version as soon as possible.

roleoroleo, 2018-05-18:
Please, download the archive quickly. I need to remove the server in a few days.
https://roleo.zapto.org:91/s/xvSbNCGS2tYkj41
Feel free to share it.
Fw version: 3.2.0.1118.

davidlb, 2018-05-18:
Great, it works for me on Digoo DG-W01F (don't know how to get the Fw version)
Thanks !

roleoroleo, 2018-05-19:
You can read the firmare version in the log of onvifer, at the bottom of the page.
Add Onvif camera, select name and IP, select user and password, select TCP transport, overwrite snapshot port with 554 and run.

Cruiser79, 2018-08-07:
@roleoroleo What did you patched in the p2pcam binary? Can you give me some more infos? My firmware is 3.1.0.0824, so i can't use your version of binary. I will try to patch my p2pcam on my own.

ant-thomas, 2018-08-07:
Have you got any pointers for patching p2pcam?
I'd like to look in to increasing the bitrate of the RTSP stream.

roleoroleo, 2018-08-08:
I simply changed the wrong resolution 0xF0 with 0xB4 at binary offset 0x00035EFC (see my file above).
F0 10 A0 E3   ---> B4 10 A0 E3
MOV R1, #0xF0   ---> MOV R1, #0xB4

Cruiser79, 2018-08-08:
Ok, i will search the position in my file. Which disassembler did you used, to find this position?

driesvancraen, 2019-07-16:
> Great, it works for me on Digoo DG-W01F (don't know how to get the Fw version)
> Thanks !

How do you get the snapshot exactly?