ant-thomas/zsgx1hacks #68: PTZ  Synology NAS (Survillence Station)

kkm0614, 2018-06-09:
Hi!

Thanks to you, there is sound and picture on your NAS. I have the ipadress: 8001/0 / av0 address right. Unfortunately I do not know how PTZ can work. If someone does this, please report it. Sorry for English, google is my friend.

Mark

[image]

sanscorp1, 2018-06-14:
Which settings and profile did you use for QNAP surv. station?

kkm0614, 2018-06-15:
Sorry, Hungarian is the text. Source [individually adjustable]. This is the Synology NAS Surveillance Station option. I do not know the QNAP settings.

lordgotham, 2018-06-15:
i'm looking for qnap too

sanscorp1, 2018-06-15:
Tried some things and now have an image thanks to your settings.

lordgotham, 2018-06-15:
Sanscorp1 do you find the solution for qnap???

sanscorp1, 2018-06-15:
Copy above settings from screenshot, use generic profile, enter port 8001 and /0/av0 as url. I used admin and no pass.

lordgotham, 2018-06-18:
sanscorp1, sorry buti don't understand your parameter
do you have a screenshot please?
thank a lot

ducapedia, 2018-08-08:
Hi there,
I'm trying to setup PTZ in ZoneMinder, but no matter I've tried, I received a message in log like: "404 Not found"
I've tryied the correct IP (including 'admin:@' to send admin username with no password), and have tryed all known ports (554, 8080, 8001) with and without '/0/av0'
Anyone had look with PTZ link ?
In iSpy program, I only had to choose "ONVIF PTZ" and it worked fine (almost, too slow but I think this I have to blame de cam).
Thanks in advanced

Damien0505, 2019-06-13:
Hi ducapedia,
Did you have any success getting PTZ working with ZoneMinder?
Kind regards,

Aculeasis, 2020-08-10:
Hi @Damien0505, i found solution:
- Control Type: `Netcat ONVIF`
- Control Device: `Profile_1` (I found it in ONVIF Device Manager -> Profiles -> token in mainStream)
- Control Address: `http://<username>:<pass>@<IP>/onvif` ( `http://admin:12345@<IP>/onvif` on my device).

Damien0505, 2020-08-11:
> Hi @Damien0505, i found solution:
> 
> * Control Type: `Netcat ONVIF`
> * Control Device: `Profile_1` (I found it in ONVIF Device Manager -> Profiles -> token in mainStream)
> * Control Address: `http://<username>:<pass>@<IP>/onvif` ( `http://admin:12345@<IP>/onvif` on my device).

Legend!!  I'll have to give it a go in the next couple of days.  Had forgotten all about it until seeing your reply.  
Lost a bit of interest in the camera after a long and prolonged argument with the seller.  Their advertising was false and misleading.  Ended getting a full refund AND got to keep the camera!