ant-thomas/zsgx1hacks #73: possibility to enable disable blue led

pixelecho, 2018-07-19:
does anyone find out how to disable the blue led of the camera (Digoo DG-MYQ) via "gio". I Mean the status LED not the irda LEDs.

edsub, 2018-09-17:
"but be careful, some might not like being changed!"
Defo....
Mine rebooted at some point while executing the loop.
And seems to be stuck.
The error it keeps giving (after applying the hack) is 
"
ERROR: can not get mac from net interface, please check hardware...
"
Tried to restore, no luck.
Finding now how to get back to factory setting (if possible...)

edsub, 2018-09-17:
Solved, after a lot of resetting and trying with the app finally got the camera back to factory.
Hack re-applied... All ok