ant-thomas/zsgx1hacks #75: I have not acces to WEBUI, SSH, FTP

CRACKISH, 2018-08-03:
I have not acces toWebUI.

I go to http://192.168.0.101:8080/cgi-bin/webui but nothing((
When i go just to http://192.168.0.101/ i redirected to http://192.168.0.101/index.asp and nothing((

How can i enable ssh? telnet is work fine.

And what you mean about ftp? It's create ftp server on camera?

Camera: GD-SC03
Firmware 3.2.8.03.08

ant-thomas, 2018-08-05:
Check out the new read-only hack and report back

CRACKISH, 2018-08-06:
Yes, it's works)) thanks

xlucasfelix, 2018-08-06:
@ant-thomas are you alive!!! hahaha when will we get an update?

ant-thomas, 2018-08-07:
@xlucasfelix An update for what?

xlucasfelix, 2018-08-07:
@ant-thomas Performance improvements (fps), adding new features requested here in github
(sd card recording, disable blue led, more robust interface and etc ...) :/

closed 2018-08-07 by CRACKISH

ant-thomas, 2018-08-07:
You do realise that this is a small hack to mainly stop the cameras talking to the cloud?
If you want a camera which produces a better quality picture with all those features you'll likely need to buy a better more expensive camera.