ant-thomas/zsgx1hacks #76: Support for GC15H

theNailz, 2018-08-04:
I've got a [GC15H](https://aliexpress.com/item/KERUI-1080P-Full-HD-Indoor-Wireless-Home-Security-WiFi-Cloud-Storage-IP-Camera-Surveillance-Camera-Home/32849963845.html) camera that's not supported by v0.4. Is there a way I could troubleshoot or debug the hack? 

What's the (supposed) entrypoint of the hack; is that `/mnt/debug_cmd.sh`? Should I dump the filesystem for starters?

theNailz, 2018-08-05:
Alright, the GC15H has a read only filesystem that can't be mounted rw; by running all tools from the sdcard I managed to start the webserver (PTZ works, IR doesn't) and dropbear ssh server. Haven't tested the rest of the tools.
I'll see if I can finish the script RO-style and check the rest if the tools will work as well.

ant-thomas, 2018-08-05:
I've just received the exact same camera and I'm working on making some mods on it.

theNailz, 2018-08-05:
@ant-thomas Are you on Discord? I'd like to share some ideas

ant-thomas, 2018-08-05:
@theNailz I think so (ant#7477)

theNailz, 2018-08-05:
Is now supported 0348815374ce74e800521e7c1d75d112ceba284c

closed 2018-08-05 by theNailz