ant-thomas/zsgx1hacks #77: Limited hack for updated Read-Only systems (CIPC-GC15HE)

ant-thomas, 2018-08-05:
I have just received a Kerui CIPC-GC15HE
Goke GK7102S based.

This has an updated firmware - 3.2.8.0121 - and therefore a read-only filesystem - though ```/home``` is mounted r/w which may help, but having an SD card available makes things workable.

As we know the original hack won't work with the read-only file system.

What I've achieved so far

- telnet access - admin/cxlinux
- rtsp server - admin/12345
- using ```mount --bind``` it is possible to use the hacked files without copying (thanks @ljs) (hosts file etc)
- busybox can be updated but require an alternative install location - this in turn needs to be added to the environment PATH variable to be active without typing the full path. This is possible using the above ```mount --bind``` trick with an edited ```/etc/profile``` file
- ptz works via web interface
- IR cut is GPIO 20

Shouldn't take this long to wrap-up in package for people with RO update cameras.

ljs, 2018-08-05:
Nice progress, but I'm curious do you have offline recording to SD CARD (event based or 24/7) on the new firmwares, when the camera is not connected to the the cloud? Or did you have offline rec on the old RW firmwares in same situation? I'm experience this behaviour..

ant-thomas, 2018-08-05:
Hack uploaded

closed 2018-08-05 by ant-thomas