ant-thomas/zsgx1hacks #79: Help unbrick chinese unbranded GK7012 (GUUDGO GD-SC03)

sensegsm, 2018-08-06:
Thanks very much for this project @ant-thomas. I think my cam is bricked. Unfortunately my cam is a chinese unbranded. I think is a GUUDGO GD-SC03, but i`m not sure. After power on my cam, this test  Pan/Tilt and after reboot itself. I can not connect to it by lan or wifi. SD card recovery not work, not detect card. Only way to connect to her is through the serial.
If you can help me or everyone can help me in any way I would be grateful.  Thanks very much.

This is the serial log:  

<pre><code> 
U-Boot 2012.10 (Feb 24 2016 - 22:33:54) for GK7102 rb sc1045 v2.00 (GOKE)

HAL:  20151223
DRAM:  64 MiB
Flash: 32 MiB
32 MiB
NAND:  SPINAND MID = 0xff, DID = 0xffff, 1]: No support this SPI nand!
SF: Detected W25Q256FV with page size 256 B, sector size 64 KiB, total size 32 MiB
In:    serial
Out:   serial
Err:   serial
Net:   arm_freq(600MHz)..............0x112032
use int MII..............
gk7101
Hit any key to stop autoboot:  0
[PROCESS_SEPARATORS] run sfboot
[PROCESS_SEPARATORS] setenv bootargs console=${consoledev},${baudrate} noinitrd mem=${mem} rw ${rootfstype} init=linuxrc ;sf probe 0 0;sf read ${loadadSF: Detected W25Q256FV with page size 256 B, sector size 64 KiB, total size 32 MiB
put param to memory
mem size (41)
bsb size (2)

the kernel image is zImage or Image
entry = 0xc1000000
## Transferring control to Linux (at address c1000000)...

Starting kernel ...

machid = 3988 r2 = 0xc0000100
Uncompressing Linux... done, booting the kernel.
[    0.000000] Booting Linux on physical CPU 0
[    0.000000] Linux version 3.4.43-gk (bruce@ubuntu) (gcc version 4.6.1 (crosstool-NG 1.18.0) ) #33 PREEMPT Thu May 19 20:55:21 PDT 2016
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: VIPT aliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine: Goke GK7102 RB_SC1045 board V2.00
[    0.000000] Memory policy: ECC disabled, Data cache writeback
[    0.000000] AHB: 0x90000000  0xf2000000  -- 0x1000000
[    0.000000] APB: 0xa0000000  0xf3000000  -- 0x1000000
[    0.000000] PPM: 0xc0000000  0xc0000000  -- 0x200000
[    0.000000] BSB: 0xc2b00000  0xf5000000  -- 0x200000
[    0.000000] DSP: 0xc2d00000  0xf6000000  -- 0x1300000
[    0.000000] hal version = 20151223
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 10414
[    0.000000] Kernel command line: console=ttySGK0,115200 noinitrd mem=41M rw rootfstype=jffs2 root=/dev/mtdblock3 init=linuxrc
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Memory: 41MB = 41MB total
[    0.000000] Memory: 36636k/36636k available, 5348k reserved, 0K highmem
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
[    0.000000]     DMA     : 0xff600000 - 0xffe00000   (   8 MB)
[    0.000000]     vmalloc : 0x83000000 - 0xff000000   (1984 MB)
[    0.000000]     lowmem  : 0x80000000 - 0x82900000   (  41 MB)
[    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
[    0.000000]       .text : 0x80008000 - 0x8044d000   (4372 kB)
[    0.000000]       .init : 0x8044d000 - 0x8046e000   ( 132 kB)
[    0.000000]       .data : 0x8046e000 - 0x80497260   ( 165 kB)
[    0.000000]        .bss : 0x80497284 - 0x804c9888   ( 202 kB)
[    0.000000] NR_IRQS:128
[    0.000000] >> gk7101 init irq vic1...
[    0.000000] >> gk7101 init irq vic2...
[    0.000000] gk7101 init vic...
[    0.000000] mach gk7101 init timer...
[... 271 lines cut ...]
HI_Media_Init: init sdk failed!
HI_Websvr_Init: init media failed!
workthread: websvr start failed!
ipc_server stop: 1970-01-01 02:00:38
/mnt/mtd/ipc/platform.sh: line 14: ps: not found
/mnt/mtd/ipc/platform.sh: line 14: ps: not found
/mnt/mtd/ipc/sd.sh: line 14: ps: not found
umount: can't forcibly umount /mnt/mtd/ipc/tmpfs/sd: Invalid argument
[   42.200000] iFlag: 1, iCurStep: 255
/mnt/mtd/ipc/platform.sh: line 14: ps: not found
/mnt/mtd/ipc/platform.sh: line 14: ps: not found
/mnt/mtd/ipc/platform.sh: line 14: ps: not found
/mnt/mtd/ipc/platform.sh: line 14: ps: not found
/mnt/mtd/ipc/platform.sh: line 14: ps: not found
/mnt/mtd/ipc/platform.sh: line 14: ps: not found
/mnt/mtd/ipc/platform.sh: line 14: ps: not found
/mnt/mtd/ipc/sd.sh: line 14: ps: not found
/tmpfs/sd: Invalid argument
/mnt/mtd/ipc/platform.sh: line 14: ps: not found
found
/mnt/mtd/ipc/platform.sh: line 14: ps: not found
/mnt/mtd/ipc/platform.sh: line 14: ps: not found
/mnt/mtd/ipc/platform.sh: line 14: ps: not found
/
</code></pre>

sensegsm, 2018-08-13:
I try to modify platform.sh, but without success. Does anyone know the root username and password in U-boot mode?

I tried with this, but not working:
<pre><code> 
admin:admin
root:root
root:null
admin:null
root:cxlinux
admin:cxlinux
default:null
admin:123456
root:123456
admin:xmhdipc
root:xmhdipc
</code></pre>

<pre><code> 
$mount SD card
mount: you must be root 

$ mv mnt mnt1
mv: can't rename 'mnt': Permission denied

$init isp
init: must be run as PID 1

$ reboot
reboot: Operation not permitted</code></pre>

<pre><code> $ ls -l mnt/mtd/ipc 
-rwxr-xr-x    1 root     root           881 Jul 13  2016 allexit.sh
-rwxr-xr-x    1 root     root          2048 Jul 13  2016 asc16
-rwxr-xr-x    1 root     root           318 Jul 13  2016 cdisk.sh
-rwxr-xr-x    1 root     root          3468 Jul 13  2016 chg_net.sh
-rwxr-xr-x    1 root     root          3204 Jul 13  2016 chknet
-rwxr-xr-x    1 root     root         12885 Jul 13  2016 chksensor
-rwxr-xr-x    1 root     root          3244 Jul 13  2016 chksock
-rwxr-xr-x    1 root     root          1168 Jul 13  2016 chkwifi.sh
-rwxr-xr-x    1 root     root           116 Jul 13  2016 clear.sh
drwxr-xr-x    4 root     root             0 May  5  2016 conf
-rwxr-xr-x    1 root     root         12824 Jul 13  2016 config_default.zip
-rwxr-xr-x    1 root     root          9712 Jul 13  2016 ddns_update
-rwxr-xr-x    1 root     root           180 Jul 13  2016 dhcp.sh
-rwxr-xr-x    1 root     root           134 Jul 13  2016 exclude.lst
-rwxr-xr-x    1 root     root           725 Jul 13  2016 facddns.sh
-rwxr-xr-x    1 root     root           216 Jul 13  2016 fdisk.sh
-rwxr-xr-x    1 root     root           401 Jul 13  2016 findap.sh
-rwxr-xr-x    1 root     root         37760 Jul 13  2016 finish.g711
-rwxr-xr-x    1 root     root         10304 Jul 13  2016 gerddns
-rwxr-xr-x    1 root     root          3036 Jul 13  2016 getdevmac
-rwxr-xr-x    1 root     root        166152 Mar  1  2016 hichip.d
-rwxr-xr-x    1 root     root        261696 Jul 13  2016 hzk16
-rwxr-xr-x    1 root     root        514527 Jul 13  2016 ipc_server
-rwxr-xr-x    1 root     root        514527 Jul 13  2016 ipc_server.bak
-rwxr-xr-x    1 root     root          5367 Jul 13  2016 libAPILib.so
-rwxr-xr-x    1 root     root          5367 Jul 13  2016 libAPILib.so.bak
-rwxr-xr-x    1 root     root         80063 Jul 1-x    1 root     root         8 
-rwxr-xr-x    1 root     root           376 Jul 13  2016 load_drv
-rwxr-xr-x    1 root     root           723 Jul 13  2016 load_media
-rwxr-xr-x    1 root     root           240 May 26  2016 loadp2p.sh
drwxr-xr-x    2 root     root             0 Jul 13  2016 modules
-rwxr-xr-x    1 root     root          2756 Jul 13  2016 net_detect
-rwxr-xr-x    1 root     root       2174412 Jul 13  2016 onvif
-rwxr-xr-x    1 root     root           576 Jul 13  2016 platform.sh
-rwxr-xr-x    1 root     root         33120 Jul 13  2016 reset.g711
-rwxr-xr-x    1 root     root          6182 Jul 13  2016 run
-rwxr-xr-x    1 root     root           421 Jul 13  2016 runarp
drwxr-xr-x    2 root     root             0 Feb 25  2016 sd
-rwxr-xr-x    1 root     root           170 Jul 13  2016 sd.sh
-rwxr-xr-x    1 root     root          5732 Jul 13  2016 sd_detect
-rwxr-xr-x    1 root     root          4064 Jul 13  2016 sd_first
-rwxr-xr-x    1 root     root           310 Jul 13  2016 sdmore
-rwxr-xr-x    1 root     root         12136 Jul 13  2016 setlinuxtime
-rwxr-xr-x    1 root     root          9944 Jul 13  2016 setnet
-rwxr-xr-x    1 root     root         69036 Jul 13  2016 smartset
-rwxr-xr-x    1 root     root            62 Jul 13  2016 smartset.sh
-rwxr-xr-x    1 root     root         43200 Jul 13  2016 start.g711
-rwxr-xr-x    1 root     root           605 Jul 13  2016 th3ddns.sh
drwxrwxrwt    4 root     root           360 Jan  1 02:00 tmpfs
-rwxr-xr-x    1 root     root          5052 Jul 13  2016 updatewifi
-rwxr-xr-x    1 root     root          6268 Jul 13  2016 upgrade
-rwxr-xr-x    1 root     root         23192 Jul 13  2016 upnp_map
-rwxr-xr-x    1 root     root           796 Jul 13  2016 upnpmap.sh
-rwxr-xr-x    1 root     root          5707 Jul 13  2016 watchdog
drwxrwxrwx   10 root     root             0 Jan  1 02:00 web
-rwxr-xr-x    1 root     root          6352 Jul 13  2016 wfsort
-rwxr-xr-x    1 root     root          1135 Jul 13  2016 wifi_ap.sh
-rwxr-xr-x    1 root     root          1845 Jul 13  2016 wifi_setsmart.sh
-rwxr-xr-x    1 root     root          2375 Jul 13  2016 wifi_setsta.sh
-rwxr-xr-x    1 root     root          3062 Jul 13  2016 wifi_wps.sh
-rwxr-xr-x    1 root     root          5616 Jul 13  2016 wifiget
-rwxr-xr-x    1 root     root          3930 Jul 13  2016 wifiset.sh </code></pre>

ohrid1, 2018-12-01:
sensegsm how you connect to  the serial  TXD RXD GND on PBC please ?????

Jamyz, 2019-06-17:
@sensegsm @ohrid1 how to connect TXD RXD GND(i think is the 4 pin)

closed 2019-06-27 by sensegsm