ant-thomas/zsgx1hacks #80: SOAP interface

theNailz, 2018-08-06:
I've discovered that these devices have a SOAP interface.

The WSDL can be found [here](http://www.onvif.org/ver10/device/wsdl). The endpoint is `http://<camera-ip>/onvif`, haven't figured the authentication out yet, but some call do not require authentication, such as GetHostname.

theNailz, 2018-08-06:
There are multiple SOAP services available as defined by `GetServices`, each with their own endpoint:
```
<env:Envelope xmlns:env="http://www.w3.org/2003/05/soap-envelope" xmlns:soapenc="http://www.w3.org/2003/05/soap-encoding" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xs="http://www.w3.org/2001/XMLSchema" xmlns:tt="http://www.onvif.org/ver10/schema" xmlns:tds="http://www.onvif.org/ver10/device/wsdl" xmlns:trt="http://www.onvif.org/ver10/media/wsdl" xmlns:timg="http://www.onvif.org/ver20/imaging/wsdl" xmlns:tev="http://www.onvif.org/ver10/events/wsdl" xmlns:tptz="http://www.onvif.org/ver20/ptz/wsdl" xmlns:tan="http://www.onvif.org/ver20/analytics/wsdl" xmlns:tst="http://www.onvif.org/ver10/storage/wsdl" xmlns:ter="http://www.onvif.org/ver10/error" xmlns:dn="http://www.onvif.org/ver10/network/wsdl" xmlns:tns1="http://www.onvif.org/ver10/topics" xmlns:tmd="http://www.onvif.org/ver10/deviceIO/wsdl" xmlns:gte="http://www.onvif.org/ver10/gtextension/wsdl" xmlns:wsdl="http://schemas.xmlsoap.org/wsdl" xmlns:wsoap12="http://schemas.xmlsoap.org/wsdl/soap12" xmlns:http="http://schemas.xmlsoap.org/wsdl/http" xmlns:d="http://schemas.xmlsoap.org/ws/2005/04/discovery" xmlns:wsadis="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:xop="http://www.w3.org/2004/08/xop/include" xmlns:wsnt="http://docs.oasis-open.org/wsn/b-2" xmlns:wsa="http://www.w3.org/2005/08/addressing" xmlns:wstop="http://docs.oasis-open.org/wsn/t-1" xmlns:wsrf-bf="http://docs.oasis-open.org/wsrf/bf-2" xmlns:wsntw="http://docs.oasis-open.org/wsn/bw-2" xmlns:wsrf-rw="http://docs.oasis-open.org/wsrf/rw-2" xmlns:wsaw="http://www.w3.org/2006/05/addressing/wsdl" xmlns:wsrf-r="http://docs.oasis-open.org/wsrf/r-2" xmlns:tnsn="http://www.eventextension.com/2011/event/topics">
   <env:Body>
      <tds:GetServicesResponse>
         <tds:Service>
            <tds:Namespace>http://www.onvif.org/ver10/device/wsdl</tds:Namespace>
            <tds:XAddr>http://<camera-ip>/onvif/Device</tds:XAddr>
            <tds:Version>
               <tt:Major>2</tt:Major>
               <tt:Minor>10</tt:Minor>
            </tds:Version>
         </tds:Service>
         <tds:Service>
            <tds:Namespace>http://www.onvif.org/ver10/media/wsdl</tds:Namespace>
            <tds:XAddr>http://<camera-ip>/onvif/Media</tds:XAddr>
            <tds:Version>
               <tt:Major>2</tt:Major>
               <tt:Minor>10</tt:Minor>
            </tds:Version>
         </tds:Service>
         <tds:Service>
            <tds:Namespace>http://www.onvif.org/ver10/events/wsdl</tds:Namespace>
            <tds:XAddr>http://<camera-ip>/onvif/Events</tds:XAddr>
            <tds:Version>
               <tt:Major>2</tt:Major>
               <tt:Minor>10</tt:Minor>
            </tds:Version>
         </tds:Service>
         <tds:Service>
            <tds:Namespace>http://www.onvif.org/ver20/imaging/wsdl</tds:Namespace>
            <tds:XAddr>http://<camera-ip>/onvif/Imaging</tds:XAddr>
            <tds:Version>
               <tt:Major>2</tt:Major>
               <tt:Minor>10</tt:Minor>
            </tds:Version>
         </tds:Service>
         <tds:Service>
            <tds:Namespace>http://www.onvif.org/ver10/deviceIO/wsdl</tds:Namespace>
            <tds:XAddr>http://<camera-ip>/onvif/DeviceIO</tds:XAddr>
            <tds:Version>
               <tt:Major>2</tt:Major>
               <tt:Minor>10</tt:Minor>
            </tds:Version>
         </tds:Service>
         <tds:Service>
            <tds:Namespace>http://www.onvif.org/ver20/analytics/wsdl</tds:Namespace>
            <tds:XAddr>http://<camera-ip>/onvif/Analytics</tds:XAddr>
            <tds:Version>
               <tt:Major>2</tt:Major>
               <tt:Minor>10</tt:Minor>
            </tds:Version>
         </tds:Service>
      </tds:GetServicesResponse>
   </env:Body>
</env:Envelope>
```

Cruiser79, 2018-08-14:
Do you have a small code snipped, which worked on your ipcam?

Cruiser79, 2018-08-15:
Ok, found it by myself. Just sended a post request to my cam  http://admin:@[IP]/onvif/device with body

`<s:Envelope xmlns:s="http://www.w3.org/2003/05/soap-envelope">
<s:Header>
<Security s:mustUnderstand="1" xmlns="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-secext-1.0.xsd">
<UsernameToken>
<Username>admin</Username>
<Password Type="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-username-token-profile-1.0#PasswordDigest">tiOrajBwg6KfUn5yEgo5PV0d18c=</Password>
<Nonce EncodingType="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-soap-message-security-1.0#Base64Binary">aQr+SdY/SEqnHX5nslAnFTUAAAAAAA==</Nonce><Created xmlns="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-utility-1.0.xsd">2018-08-14T20:34:36.859Z</Created></UsernameToken></Security></s:Header>
<s:Body xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema">
	<GetDeviceInterfaces xmlns="http://www.onvif.org/ver10/device/wsdl"/>
</s:Body></s:Envelope>`

and got response

`<?xml version="1.0" encoding="UTF-8"?>
<env:Envelope xmlns:env="http://www.w3.org/2003/05/soap-envelope" xmlns:soapenc="http://www.w3.org/2003/05/soap-encoding" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xs="http://www.w3.org/2001/XMLSchema" xmlns:tt="http://www.onvif.org/ver10/schema" xmlns:tds="http://www.onvif.org/ver10/device/wsdl" xmlns:trt="http://www.onvif.org/ver10/media/wsdl" xmlns:timg="http://www.onvif.org/ver20/imaging/wsdl" xmlns:tev="http://www.onvif.org/ver10/events/wsdl" xmlns:tptz="http://www.onvif.org/ver20/ptz/wsdl" xmlns:tan="http://www.onvif.org/ver20/analytics/wsdl" xmlns:tst="http://www.onvif.org/ver10/storage/wsdl" xmlns:ter="http://www.onvif.org/ver10/error" xmlns:dn="http://www.onvif.org/ver10/network/wsdl" xmlns:tns1="http://www.onvif.org/ver10/topics" xmlns:tmd="http://www.onvif.org/ver10/deviceIO/wsdl" xmlns:gte="http://www.onvif.org/ver10/gtextension/wsdl" xmlns:wsdl="http://schemas.xmlsoap.org/wsdl" xmlns:wsoap12="http://schemas.xmlsoap.org/wsdl/soap12" xmlns:http="http://schemas.xmlsoap.org/wsdl/http" xmlns:d="http://schemas.xmlsoap.org/ws/2005/04/discovery" xmlns:wsadis="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:xop="http://www.w3.org/2004/08/xop/include" xmlns:wsnt="http://docs.oasis-open.org/wsn/b-2" xmlns:wsa="http://www.w3.org/2005/08/addressing" xmlns:wstop="http://docs.oasis-open.org/wsn/t-1" xmlns:wsrf-bf="http://docs.oasis-open.org/wsrf/bf-2" xmlns:wsntw="http://docs.oasis-open.org/wsn/bw-2" xmlns:wsrf-rw="http://docs.oasis-open.org/wsrf/rw-2" xmlns:wsaw="http://www.w3.org/2006/05/addressing/wsdl" xmlns:wsrf-r="http://docs.oasis-open.org/wsrf/r-2" xmlns:tnsn="http://www.eventextension.com/2011/event/topics">
    <env:Body>
        <tds:GetDeviceInformationResponse>
            <tds:Manufacturer>EYEPLUS</tds:Manufacturer>
            <tds:Model>EYEPLUS_DEV</tds:Model>
            <tds:FirmwareVersion>3.1.0.0824</tds:FirmwareVersion>
            <tds:SerialNumber>12345679890</tds:SerialNumber>
            <tds:HardwareId>88</tds:HardwareId>
        </tds:GetDeviceInformationResponse>
    </env:Body>
</env:Envelope>`

But not all onvif commands are supported. _GetStorageConfigurations_ for example returns with an message, included the text _Action Not Implemented_