ant-thomas/zsgx1hacks #81: DG-M1Z not working

ciast, 2018-08-21:
Hello,

I tried to use both the readonly and normal versions. None of them work properly for the firmware version 22.00.00.40. Additionally, using: https://github.com/zzerrg/gmfwtools
I modified the boot script directly in the firmware and is half-successful - only the ftp server does not work.
When I try to connect to ftp, I get the following message in the console on the server side:
tcpsvd: can't execute 'ftpd': No such file or directory

Modified boot script below:

/npc/boot.sh

```
#start npc boot

ulimit -s 2048

# ANT-THOMAS
############
# HACKS HERE

# mount sd card to separate location
if [ -b /dev/mmcblk0p1 ]; then
        mount -t vfat /dev/mmcblk0p1  /media
elif [ -b /dev/mmcblk0 ]; then
        mount -t vfat /dev/mmcblk0 /media
fi

# confirm hack type
touch /home/HACKSD

mkdir -p /home/busybox

# install updated version of busybox
mount --bind /media/hack/busybox /bin/busybox
/bin/busybox --install -s /home/busybox

# set new env
mount --bind /media/hack/profile /etc/profile

# possibly needed but may not be
mount --bind /media/hack/group /etc/group
mount --bind /media/hack/passwd /etc/passwd
mount --bind /media/hack/shadow /etc/shadow

# update hosts file to prevent communication
mount --bind /media/hack/hosts.new /etc/hosts

# busybox httpd
/home/busybox/httpd -p 8080 -h /media/hack/www

# setup and install dropbear ssh server - no password login
/media/hack/dropbearmulti dropbear -r /media/hack/dropbear_ecdsa_host_key -B

# start ftp server
(/home/busybox/tcpsvd -E 0.0.0.0 21 ftpd -w / ) &

# sync the time
(sleep 20 && /home/busybox/ntpd -q -p 0.uk.pool.ntp.org ) &

# silence the voices - uncomment if needed
#if [ ! -f /home/VOICE-orig.tgz ]; then
#    cp /home/VOICE.tgz /home/VOICE-orig.tgz
#fi
#
#cp /media/hack/VOICE-new.tgz /home/VOICE.tgz

#
############

/bin/busybox telnetd

cp /npc/sensors/* /etc/sensors/ -Rf

if [ -e /patch/lib/mt7601Usta_v2.ko ] ; then
insmod /patch/lib/mt7601Usta_v2.ko
elif [ -e /patch/lib/8188fu.ko ] ; then
insmod /patch/lib/8188fu.ko
fi
sync;sync;sync && echo 3 > /proc/sys/vm/drop_caches

```

gecko749, 2018-08-29:
Hi,

try
`/home/busybox/tcpsvd -E 0.0.0.0 21 home/busybox/ftpd -w / &`

regards

gecko

ant-thomas, 2018-09-02:
Is that even a camera with a similar setup?