ant-thomas/zsgx1hacks #84: Work on DG-MYQ

edsub, 2018-09-03:
I am in the process of verifying correct funcioning on a DG-MYQ camera.
The ReadOnly works as described.
It even looks like the R/W version works as well (at least in SD card mode).

verification not finalised.
I am also looking for a way 
1. to use a static IP address on the WiFi
2. to use a snapshot in addition to onvif

To be continued

edsub, 2018-09-06:
Well I managed to get a bit further:
1. The R/W hack works although I have a fairly recent version: 3.2.8.0121
2. Using Linux scripting I am almost there to capture MOTION and SOUND events and call a (Domoticz) webservice to log the event + actual time. The action functioning of the events is a bit puzzling as the config is not very clear about what triggers what.....
3. No real solution available for implementation on the cam to get a snapshot, so will try with ffmpeg on a RaspberryPi. This is a pity because otherwise I could have linked taking a snaphot with the event system.
4. Network connection is setup from within the p2pcam binary. Still no luck in changing from dhcp to static ip....
5. Corrected timezone in config

davidlb, 2018-09-07:
For snapshot, there is a patch to apply: #64

edsub, 2018-09-07:
Cool, will try that.

Some progress from my side: network available detection is now much more reliable (used to be just a sleep in which it work in most cases. But not all....).

Suddenly the camera does not log EVENTs anymore. (not using patched p2pcam). Probably I unintendedly changed something in a config. Anyone with some info on eg
config.cfg
config.xml
hwcfg.ini

edsub, 2018-09-13:
OK, so here is the the first release of my fork of this project: https://goo.gl/pYPF4r
Changes / Improvements:
1. Startup made more robust (especialy waiting for the network to become available)
2. Introduced native Domoticz logging based on Camera EVENTs (probably sensitivity needs tweaking)
3. Included jpeg snapshot
Download https://goo.gl/pYPF4r

NOTE: ONLY the persistent hack (ie. where you do not need the SD card for R/W capable cameras) is tested to be working.

Instructions:  (note 192.168.1.117 is my camera IP, adapt to your own)
Shell login: root / cxlinux
OnVif: rtsp://192.168.1.117 (1280*720 resolution), username admin/no password
Snapshot: http://192.168.1/117:554/snapshot (320*180 resolution)

If you do not need Domoticz logging, you have to comment out two lines in the /home/p2pcam.sh script. These lines are marked with "#Comment out next line if NO Domoticz logging is needed"

Configure Domoticz logging: 
In Domoticz: Create a virtual device of type "Sensor", remember the index of that device!
On your camera: 
Edit /home/p2pcam.sh using vi and modify: 
"770" to the index of your Domotcz Alert sensor (2 locations)
"192.168.1.211" to the ip of your Domoticz (4 locations)
"192.168.1.117" to the ip of your camera (2 locations)

rzr, 2018-09-22:
One of my friend asked about finding UART port on T821-B0 board, I thought that the 3 holes on right edge could be the the serial port ? maybe it's not mandatory to find that to use modded fw ...

edsub, 2018-10-01:
You do not need a telnet session via the uart (which indeed is the 3 pinholes). Just use the sd to hack and start a telnet session to the ip address.

Fietspomp86, 2018-12-11:
I also have a DG-MYQ with the same firmware but unfortunately mine is read-only! Any workaround for this?

TungstenE2, 2019-03-30:
just got my DG-MYQ. Tried your sdcard hack.
While snapshot and rtsp are working, the webui does not.
Any idea why?

Snapshot: 
http://192.168.178.99:554/snapshot

Webui: 
http://192.168.178.99:8080/cgi-bin/webui

AnthonMS, 2019-08-17:
Finally got around to play with my DG-MYQ camera I bought almost 6 months ago.
I can't seem to get it to work. Neither webui or snapshot is working, and it still connects to the cloud.
I have gotten access to it through telnet with user: root / pass:cxlinux.
I have found these RTSP streams to work (with user:admin / pass:empty): 
# Without sound - Small picture
rtsp://admin:@192.168.0.11:8001
# Without sound - Large picture
rtsp://admin:@192.168.0.11:8001/tcp/av0_0
# With Sound - Small picture
rtsp://admin:@192.168.0.11:8001/0/av0

I am new to telnet, so don't really know how I should go about testing if I can do anything useful with it.
I have tried ant-thomas' original zsgx1hacks hacks, both read/write and readonly, but with no luck.

My version is 3.3.0.0803
So it is a bit newer version then what you are using, so I guess that's why nothing is working as it should.

Don't really know what to do. I really don't want to use it with the cloud when I can't change password for neither telnet or rtsp streams. And I can't seem to find the option where I can let it on my WiFi but cut it off from the outside world, but that would be ideal, since I am goind to use RTSP.

Do you have any tips for a novice like myself?

Fietspomp86, 2019-08-17:
You can change the rstp password, with official firmware and an INI file.
Here's the link, usage instructions are in the zip file.
[https://www.mydigoo.com/downfile.html?path=/mydigoo/20181008/5bbc16ced92d5-MYQ%20Onvif%20password%20changing.zip](url)

AnthonMS, 2019-08-17:
Thank you, it changed the password successfully. But all the streams are in black and white now though.

Fietspomp86, 2019-08-17:
> Thank you, it changed the password successfully. But all the streams are in black and white now though.

Maybe that's the night view?

AnthonMS, 2019-08-17:
Might be. Thought the night vision only toggled when it was dark in the room. It is night where I am though, but it's been dark outside since I started fiddling around with it. And I haven't turned the lights out in the room.

AnthonMS, 2019-08-17:
I have updated it with the new password and it got rolled back to an earlier version while doing that. I thought I would try and update it through the app to see what version I would get. I now have version: 3.2.8.0121. Which is the same as you have gotten it to work with, but webui and snapshot still doesn't work, and it still connects to the cloud. The version I got after updating password was something like: 3.1.x something. So I might just try to update the password again to see if it will work with that version.

Fietspomp86, 2019-08-17:
I just figured out they're a more easy way to change the password.
Just FTP in the device with the readonly hack and go to /home/.
There you can edit the hwconf.ini file.

But anyway, what results do you get when you extract the readonly zip from this git to a tf card and place in your camera?

AnthonMS, 2019-08-18:
I have now tried that, and it is still black and white. It is not because it is night, because I have tried to put it in a dark place, and the IR turns on and for some reason it turns back to regular color. But as soon as it is sunny or the light is turned back on, it goes back to black and white. 

these links: 
<IP>/cgi-bin/webui
<IP>/snapshot
now comes up with a popup login form, but I don't know what the credentials are. Do you know anything about that? I have tried root/cxlinux, I have tried admin/<RTSP Pass>, and some other ones.

edsub, 2019-08-18:
If the camera keeps connecting to the cloud, then the hack was not applied.
You can verify that by (eg. via telnet) compare files in the home folder with files you copied to the sd.
I think your sd is not properly created.

AnthonMS, 2019-08-18:
I tried the readonly hack from this git, and that seems to have worked. It does not connect to the cloud but the RTSP streams still work. However, the picture is in black and white. Do you know why this has happened?

Fietspomp86, 2019-08-18:
I just tried to apply new firmware and I got the same result, like night mode is reversed.
What I did now is FTP into the camera and modify hwcfg.ini.
Mine looks like this now:
```
[config]
model = CloudCamera
sound_detect = 1
support_onvif = 1
passwd = xxxxxxx
```

Looks like it works again now!

AnthonMS, 2019-08-18:
Yes, I have also found out it is because the night mode is reversed. I tried following another tutorial on YT that had the same problem but with the M1Q model, but that didn't do the trick.

I'm sorry to be a burden even more, but I haven't really worked with FTP before. Can you give me a quick recap of how to FTP into it and change the hwcfg.ini? And do I have to have the read/write or read-only hack on it, to FTP into it?
I have both a linux and windows pc at my disposal, so whatever is easiest.

Fietspomp86, 2019-08-18:
The m1q tutorial doesn't work, you mean flipping the lens right?
This is purely a software bug.
I always use FlashFXP, for FTP'ing.
Just open flashfxp and enter the ip for the camera.
No need for a user or password.
Navigate to the /home folder.
Rightclick the hwcfg.ini and select edit.
It's that simple!

AnthonMS, 2019-08-18:
Exactly, that is the tutorial I tried, but as you say, it is a software bug.
I just installed FlashFXP, and tried putting in the IP address of my camera after clicking on the connect button on the right side. But says connection failed. I tried doing an nmap scan on the camera, and it doesn't seem to have port 20 or 21 open. Should it not have those ports open for FTP to connect?

Fietspomp86, 2019-08-18:
You should have the readonly hack installed, then it's working fine. Port 21 indeed.

AnthonMS, 2019-08-18:
Thank you. I have done this now. My hwcfg.ini looks the same as what you wrote. But now it doesn't go to night mode. Do you have the same issue? When I put a finger over it, it lights up all the red lights around the lens, but it doesn't make the clicking noise it does when turning to night mode, and it stays in colored picture.

AnthonMS, 2019-08-18:
I have tried to play around with the settings in hwcfg.ini now. I have tried setting 'ir_detect_type' to 0, 1 and 2. On 0 and 1 nothing happens at all, it does not detect night mode, but it is at least not reversed.
When I set it to 2, it detects night mode, but is reversed. So it turns on IR lights at day and off at night.
I have also tried changing the sensor_position to 1, but that didn't seem to do anything neither.
If just the webui worked, I would be able to turn them off/on manually, but that doesn't work neither.

Fietspomp86, 2019-08-19:
You know the webui is at port 8080 right?
Because with the same camera and same hack it works fine here.

AnthonMS, 2019-08-19:
Yes I know. The webui also shows up. But I can't turn on/off the IR with the buttons on the webui.

Fietspomp86, 2019-08-19:
Ok good. Try to use the webui file from edsub
https://github.com/edsub/Goke_GK7102

Just copy the file webui to your sdcard and overwrite, then you get more buttons and these should work.
File webui is located in folder hack/www-cgi

AnthonMS, 2019-08-20:
I thought about that. But I compared the two files. They are almost identical. What shows up when you telnet into it and type gio? Just to know if it is normal that it only responds with gio 49, because the webui turns on/off gio 46.

Fietspomp86, 2019-08-22:
Sorry for the late reply, this is my output of gio

```
gio -di 49
gio -do 49 1
gio -s 49 0
gio -g 49
```

I flashed an older version of p2pcam btw, and night/day works fine for me.
I use this one: 3.1.1.0908 (check version with "ls /tmp | grep -F 3.")
And hwconf is like this:
```
[config]
model = CloudCamera
main_bps = 768
sensor_position = 0
sound_detect = 1
ir_detect_type = 2
adc_setting_max = 700
adc_setting_min = -300
support_onvif = 1
passwd = xxxxxx
```

AnthonMS, 2019-08-22:
I will check the version when I get home. Where did you find the older version? And did you flash it by just putting it on a sdcard like the hack and changing password?

Fietspomp86, 2019-08-22:
It was this one:
[http://files.banggood.com/2016/06/GUUDGO%20ONVIF.rar](url)

Just flash it indeed

AnthonMS, 2019-08-22:
Thank you. I will try it out later tonight and post the version after. Hopefully it fixes the reversed night mode.

AnthonMS, 2019-08-22:
After I used that firmware, something has gone wrong. I took out the sdcard after it said successful, put in my pc again, and it contained a text file with the name ERROR. In this text file it says "Copy hwcfg.ini failed.", it connects to my wifi, and I can use rtsp, but with no password. I have used telnet to look if it indeed did fail the copy, and it did. There are no hwcfg.ini in the /home directory. I can then put the readonly hack on the sd card, and I can conncet over ftp. But when I try to tansfer the hwcfg.ini to the /home folder, it says "1 file failed". Is there a way I can recreate the file directly on the camera over ftp or telnet or something?

I have tried both the MYQ password changing firmware and the GUUDGO firmware you just send, multiple times. Every time I get the error text file returned in the sd card. I have tried resetting the camera, that did not help. So now I am really screwed.

Fietspomp86, 2019-08-24:
Very strange, in my case it worked just fine!
But did you change the filename from that `ruishi-xiaoxueren-hwcfg.ini`? I did not.
Anyway, if you still have acces you can also use vi to make a new hwcfg.ini.
Or put it on a server somewhere and use wget in the home folder.
Or use the other firmware again to see what happens.
The version is now good by the way, with that one the B/W issue is gone. I'm only trying to find a solution to fix snapshot issue.

AnthonMS, 2019-08-24:
Yes I still have access. The telnet works fine, and if I put the Readonly Hack on, it also starts the ftp server. 

Did not know wget and vi was accessible through telnet. Will give that a go.

Fietspomp86, 2019-08-24:
Yes, I was just experimenting and you can also just update the p2pcam by going to the home folder, delete old p2pcam.tar.gz and wget another one from a server you've uploaded it to.
Works great and this also changed firmware version.
Right now I'm using the one from edsub (git above) and seems that day and night works as well as snapshot, version is now 3.2.0.1118

Good luck!

AnthonMS, 2019-08-24:
I think I bricked it. I tried to make the hwcfg.ini with vi, but that said no more room on device when I tried to save it. I then tried the wget, but that said the same thing. I then tried updating it through the app, to see if I could get it back to normal to start over, but it never stopped updating and now it doesn't connect to wifi, and if I hold the reset button it doesn't do anything neither... I tried putting the firmware on the sd card and boot it with that in it, but nothing happens.

So I think this is a lost cause by now. :(

Fietspomp86, 2019-08-24:
Use a serial connection and then you can revive it for sure. I guess you didn't backup your firmware?

AnthonMS, 2019-08-24:
No I didn't back it up. I don't know where the rx, tx and gnd pins are. I was going to try and flash the .bin file from the changing password folder. But can't find the pins.

Fietspomp86, 2019-08-24:
https://github.com/yuvadm/DG-M1Q

Here is the serial connection setup, 3 pins on the side

AnthonMS, 2019-08-24:
My MYQ doesn't have those. It has some smaller ones that look like it, but not exactly where those are located. I found someone saying the three pins next to the sdcard is rx/gnd/tx, but just tried with the ftdi adapter, and it couldn't flash it. 

Not an expert with flashing through ftdi adapter, but do I not need the 3v pin as well? Or can i just connect gnd->gnd / rx->tx / tx->rx and flash it that way?

TimelessNL, 2019-09-09:
The MYQ indeed has the serial port located next to the SD-Card slot (see: [link](https://github.com/yuvadm/DG-M1Q/issues/7)) 3.3v is not needed and may even damage the camera if connected.

GilDev, 2020-03-02:
@AnthonMS Have you tried to make it work again and got any success?