ant-thomas/zsgx1hacks #85: DG03 ONVIF settings

nick9one1, 2018-09-12:
Hi, 

First of all, thanks for all the hard work people have put into this project. I have two GD-SC03's that were in a box until I discovered this!

Unfortunately I seem to be having trouble with ONVIF and 'Home assistant'. 
The RO firmware is successfully running from SD, and I can connect to the camera using ONVIF device manager and tinycam monitor but not Home Assistant using ONVIF. 

I noticed in tinycam monitor there are three protocols;

Snapshot JPEG (does not work)
RSTP over TCP (Works)
RSTP over UDP (Works)

Also ONVIF device manager has the URL rtsp://192.168.1.97:554 under the live video feed, suggesting it has connected over RSTP. 

I wonder if Home assistant is trying to connect using the Snapshot protocol? 
Can anyone confirm if there is a way to get that protocol working?

Thanks!

deorder, 2018-09-28:
Use the following instead of ONVIF:
```
camera:
  - platform: ffmpeg 
    input: rtsp://user:pass@192.168.1.75:554
    name: Camera 1
```

miker5, 2019-09-04:
Works for me, thx