ant-thomas/zsgx1hacks #86: Setup static IP

edsub, 2018-09-18:
Did anyone succeed in setting up a static IP for this cam?

jumblies, 2018-09-30:
I can't address your question specifically, but could u assign in the router? On a lot of router FW, you can bind Mac and IP.  Thanks for your work.  Looking forward to trying it out

edsub, 2018-10-01:
The 4G modem/router the camera will be linked to, does not support forcing a fixed ip based om mac. Hence my question.....

jumblies, 2018-10-01:
Sorry for the unhelpful answer.  

I couldn't find dhcp on the device but there is a way to manually set the IP in the link below.  That file is writeable so it could work to set it manually.  
```/etc/network/interfaces```
https://ubuntuforums.org/showthread.php?t=1426270

gd-99, 2019-03-15:
Bit dated now, but if this is still of interest, I have just posted a possible solution in [issue #104 ](https://github.com/ant-thomas/zsgx1hacks/issues/104#issuecomment-473018884)

The bit that worked for me is setting up a cls.conf file:
**cls.conf contents**
```
[cls_server]
ip = 192.168.1.xx
gw = 192.168.1.1
netmask = 255.255.255.0
dns = 208.67.222.222
ssid = <your_ssid>
passwd = <your_ssid_password>

```
This example fixes the IP address for the wifi. To set a fixed IP address for eth0 add the line:
`fixip = eth0 `

edsub, 2019-03-15:
Cool, coincidently I found the same today, but it does not seem to work on my DG-W01F
Put the cls.conf in the /home folder, is that correct?
Will try to figure out,.

gd-99, 2019-03-15:
@edsub I put my cls.conf in the root directory of the sd-card - alongside the debug_cmd.sh file and the hack directory.