ant-thomas/zsgx1hacks #89: Bricked while trying to remove hack

ducapedia, 2018-10-31:
I think I've bricked my camera, while I was removing the hack.
I've runned remove script and after boot nothing happen (and I even cant acess by ssh or telnet.. nothing)
I also have tried to put a microSD, with hack and in config file, put YES in restore option, but nothing happen either, tried to reboot withou microSD, and press and hold reset button, nothing. The camera had turned on (the leds are turned on), but nothing move (like before when rebooting), official acess web site (eyeplys) says the device is disconected or turned off, and I have no connection detected in my openwrt router (wifi or ethernet, nothing happen).
I think i've read in someware, theres a way to flash the original firmware again, just copying some binary image in microSD. Someone can help me to go throught this, please ?
[image]

Csontikka, 2019-04-01:
Hi!
Could you finally do anything with it? Did you manage to fix?

TungstenE2, 2020-02-10:
check this: https://github.com/ant-thomas/zsgx1hacks/issues/44