ant-thomas/zsgx1hacks #90: Works on IL-HIP291G-2M-AI from INQMEGA

paus56, 2018-11-13:
readonlyhack-v0.1 прекрасно заработал на камере IL-HIP291G-2M-AI
- Версия прошивки: 3.3.0.0817

rtsp://192.168.0.49:554/0/av0  - видео со звуком с высоким разрешением (1920х1088)
rtsp://192.168.0.49:8001/0/av0 - видео со звуком с высоким разрешением (1920х1088)  !!!

-для работы PTZ в tyniCam необходимо в настройки программы tyniCam добавить файл GK7102_CloudCam_vendors.xml
со следующим содержанием:

<?xml version="1.0" encoding="UTF-8"?>
<vendors>
    <vendor name="GK7102_CloudCam_Hacks">
        <model name="GK7102_CloudCam" defaultPort="8080" defaultRtspPort="8001">
            <request name="RTSP">/0/av0</request>
            <request name="PtzMoveRelLeft">/cgi-bin/webui?command=ptzl</request>
            <request name="PtzMoveRelRight">/cgi-bin/webui?command=ptzr</request>
            <request name="PtzMoveRelUp">/cgi-bin/webui?command=ptzu</request>
            <request name="PtzMoveRelDown">/cgi-bin/webui?command=ptzd</request>
        </model>
    </vendor>
</vendors>

были выявлены следующие настройки камеры в файле: hwcfg.ini
- passwd = 12345678       - установка - изменение пароля для ONVIF
- support_mp4record = 1   - запись видео в формате .MP4 на SD карту с оригинальной прошивкой
- main_bps = 768          - битрейт video
- main_fps = 12           - fps video

paus56, 2018-11-13:
Список всех параметров камеры IL-HIP291G-2M-AI из файла:  p2pcam
для записи в файл  hwcfg.ini
`  
  "model"
  "main_bps"
  "sensor_position"
  "support_eth"
  "ir_detect_type"
  "adc_setting_max"
  "adc_setting_min"
  "support_ptz"
   "support_allid"
  "support_onvif"
  "passwd"
  
  "auto_reboot"
  "ptz_one_step"
  "main_fps"
  "sub_bps"
  "sub_fps"
  "support_fisheye"
  "local_quality_auto"
  "voice_prompt"
  "sound_detect"
  "sound_level"
  "motion_level"
  "support_twoway_speech"
  "adc_chan"
  "buzzer_time"
  "cds_lvl_night"
  "ircut_reverse"
  "ircut1_lvl"
  "ircut2_lvl"
  "ir_light_lvl"
  "white_light_lvl"
  "smart_ai_mode"
  "support_433"
  "support_smartfeed",
  "buzzer_mode"
  "support_pir"
  "support_onekeycall"
  "show_logo"
  "support_syncledtime
  "support_wifikit"
  "support_doublelight
  "ptz_mic_mode"
  "support_mp4record",
  "voice_volume"
  "hisi_redCastGain"
  "standard_definition
  "talk_volume"
  "support_autosaturation"
  "fisheye_mode"
  "support_breakpointrecord"
  "night_smear"
  "support_mdzone"
  "agcnight_value"
  "agcday_value"
  "mic_value"
  "power_freq"
  "wifikit_passwd" - "12345678"
  "wifikit_ssid"   - "CLOUD_WIFIKIT"
`

aHVzY2g, 2018-12-15:
I can confirm, the hack ist working. Thank you very much

McGr3g0r, 2018-12-20:
I can confirm it too. However I was able to brick it immediatelly, curious me.
After that decided to open it up and located the 3.3V UART for debuging - [ [image](https://photos.app.goo.gl/BkeW2UKuwCpwHvA39) ]

U-boot output:

```
U-Boot 2012.10 (Dec 26 2017 - 18:17:43) for GK7102 rb-sc1045-v2.0 (GOKE)

HAL:   20160913 
DRAM:  64 MiB
Flash: 8 MiB
NAND:  [No SPI nand] 
SD/MMC: 0
SF:    8 MiB [page:256 Bytes] [sector:64 KiB] [count:128] (XM25QH64)
In:    serial
Out:   serial
Err:   serial
Net:   Int PHY 
Hit any key to stop autoboot:  0 
GK7102 # 
GK7102 # 
GK7102 # printenv
[PROCESS_SEPARATORS] printenv
arm_freq=0x00112032
baudrate=115200
bootargs=console=ttySGK0,115200 mem=36M rootfstype=squashfs root=/dev/mtdblock2 init=linuxrc mtdparts=gk_flash:320K(U),1664K(K),1152K(R),2560K(A),-(H)
bootcmd=sf probe;sf read 0xc1000000 0x50000 0x1A0000;bootm 0xc1000000;
bootdelay=1
bootfile=zImage
bsbsize=1M
consoledev=ttySGK0
ethact=gk7101
ethaddr=xx:xx:xx:xx:xx:xx  <-- this is not real value
filesize=256D78
gatewayip=192.168.0.1
hostname="gk7102s"
ipaddr=192.168.0.27
loadaddr=0xC1000000
mem=36M
netdev=eth0
netmask=255.255.255.0
nfsserver=11.1.4.19
phytype=0
rootfstype=ubi.mtd=3 rootfstype=ubifs root=ubi0:rootfs
rootpath=/opt/work
serverip=192.168.0.29
sfboot=setenv bootargs console=${consoledev},${baudrate} noinitrd mem=${mem} rw ${rootfstype} init=linuxrc ip=${ipaddr}:${serverip}:${gatewayip}:${netmask}:${hostname}:${netdev} mac=${ethaddr} phytype=${phytypem
sfkernel=0x50000
stderr=serial
stdin=serial
stdout=serial
tftpboot=setenv bootargs root=/dev/nfs nfsroot=${nfsserver}:${rootpath},proto=tcp,nfsvers=3,nolock ip=${ipaddr}:${serverip}:${gatewayip}:${netmask}:${hostname}:${netdev} mac=${ethaddr} phytype=${phytype} consolm

Environment size: 1290/65532 bytes
GK7102 # 

```
So anyone can easily dump the SPI FLASH by issuing commands:

```
sf probe;
sf read 0xc1000000 0x0 0x800000
mmc init
mmc parts
fatwrite mmc 0:1 0xc1000000 partall.raw 0x800000

```
it is initializing SPI flash and MMC driver, then reads entire SPI flash to the RAM memory and then writes it to MMC into first partition that has to be FAT type.

Later using the mtdparts=gk_flash:320K(U),1664K(K),1152K(R),2560K(A),-(H)  values
I was able to split the large dump into the chunks and mount partitions 3(root), 4(app), 5(home).
The root and app parition type is squashfs and home partition is jffs2 type.

For my case I had to remove /home/etc/wpa_supplicant.conf file otherwise the app could not connect to the wifi router.

Best regards,
McGr3g0r

paus56, 2018-12-21:
How did you manage to execute these commands?
[image]

McGr3g0r, 2018-12-21:
These commands where issued in **U-boot** bootloader not the Linux console.
You can stop U-boot  from booting the kernel by:
`Hit any key to stop autoboot:  0 `

paus56, 2018-12-21:
and so the received files are different?

```
# dd if=/dev/mtd0 of=/mnt/mtd0.img
640+0 records in
640+0 records out
327680 bytes (320.0KB) copied, 0.146524 seconds, 2.1MB/s
# dd if=/dev/mtd1 of=/mnt/mtd1.img
3328+0 records in
3328+0 records out
1703936 bytes (1.6MB) copied, 0.746780 seconds, 2.2MB/s
# dd if=/dev/mtd2 of=/mnt/mtd2.img
1792+0 records in
1792+0 records out
917504 bytes (896.0KB) copied, 0.405942 seconds, 2.2MB/s
# dd if=/dev/mtd3 of=/mnt/mtd3.img
10624+0 records in
10624+0 records out
5439488 bytes (5.2MB) copied, 5.101560 seconds, 1.0MB/s
```

mtd2.img - squashfs
mtd3.img - jffs2 (Home)

but mtd0 and mtd1 can not be opened, it is unclear what file system ?

McGr3g0r, 2018-12-24:
mtd0 is U-Boot bootloader and mtd1 Linux Kernel, both are plain ARM binaries so mounting of these partitions is impossible.

paus56, 2018-12-28:
```
IL-HIP291G-2M-AI
firmware ver. 3.3.0.0817
#
# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00050000 00010000 "U"
mtd1: 001a0000 00010000 "K"
mtd2: 000e0000 00010000 "R"
mtd3: 00530000 00010000 "A"
#

IL-HIP291G-2M-AI
firmware ver. 3.4.0.1031
#
# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00050000 00010000 "U"
mtd1: 001a0000 00010000 "K"
mtd2: 00120000 00010000 "R"
mtd3: 00280000 00010000 "A"
mtd4: 00270000 00010000 "H"
#

```

paus56, 2019-01-06:
в камере IL-HIP291G-2M-AI  для вывода OSD  даты и времени добавить строку: 
**show_osd_time = 1** 
в файл: **hwcfg.ini** в папке **/home/** камеры

[image]

[image]

Gymnae, 2019-02-04:
@McGr3g0r, you seem to have the same flash chip as I do. I managed to brick my camera when trying to flash a new firmware: https://github.com/ant-thomas/zsgx1hacks/issues/96 - the 8MBiT Flash chip doesn't seem to be so common. 

Do you have a backup of your flash you could share with me, so I can bring my camera back to live by flashing the version you have? Of course, if your camera is in a working state at the moment :)

@paus56, I used deepl.com to translate pages from the thread you are active in on http://4pda.ru - You seem to be pretty deep into the matter, do you have a firmware or flash package for cameras with 8MBiT flash?

paus56, 2019-02-10:
- это прошивка от моей камеры на GK7102, 
- размер флеш моей камеры 8 мегабайт
- с этой прошивкой камера не будет подключаться к облаку
-  [как прошивать описано там:](http://4pda.ru/forum/index.php?s=&showtopic=928641&view=findpost&p=81236013)
- [как подключать, описано там: ](http://4pda.ru/forum/index.php?s=&showtopic=928641&view=findpost&p=82237201)

[Firmware for IL-HIP291G-2M-AI.zip](https://github.com/ant-thomas/zsgx1hacks/files/2848617/Firmware.for.IL-HIP291G-2M-AI.zip)

Gymnae, 2019-02-10:
Thanks, @paus56, I guess then I need to order a programmer.
The flash chip of my guudgo gd-c03 carries the marking of md25q64cs16 and reports as gd25q64. I have a hot air solder station, so taking out the chip and replacing it should be doable. 
---
Спасибо, @paus56, наверное, тогда мне нужно заказать программиста.
Микросхема моего гуудго gd-c03 имеет маркировку md25q64cs16 и сообщает как gd25q64. У меня есть паяльная станция горячего воздуха, поэтому достать чип и заменить его можно.

paus56, 2019-02-12:
если у вас камера отличается от  IL-HIP291G-2M-AI, то прошивка может не заработать

Google translate
if your camera is different from IL-HIP291G-2M-AI, then the firmware may not work

Gymnae, 2019-02-12:
I'm aware, but it's worth a try, I hope. I have no comparable package for the GD-C03.

Я знаю, но, надеюсь, стоит попробовать. У меня нет похожего пакета для GD-C03.

Gymnae, 2019-02-12:
@paus56, I tried your flash just now. It successfully flashed and booted, but you are correct, the GD-SC03 is different and it doesn't work. Also, it really only properly flashes when de-soldering the chip, not when using the SOP8 clamp. 
Now my hope is to find a .bin or collection of mtd fitting to the gd25q64 and/or 8MB Flash from a GD-SC03

deepl.com translate:
@paus56, я только что попробовал твою вспышку. Он успешно прошит и загрузится, но вы правы, GD-SC03 отличается и не работает. Кроме того, он действительно правильно мигает только при депайке чипа, а не при использовании хомута SOP8. 
Теперь я надеюсь найти .bin или коллекцию mtd, подходящую для gd25q64 и/или 8MB Flash от GD-SC03.

slydiman, 2019-02-24:
I have INQMEGA IL-HIP291L-2M-AI (Ethernet + Wi-Fi version) firmware 3.4.1.1212
I have updated the script /media/hack/www/cgi-bin/webui 

1. I used the following command to detect IP address using ifconfig and grep only
ipadd=`/sbin/ifconfig eth0 | grep -oE "inet addr:.*.B" | grep -oE "([0-9]{1,3}\.){3}[0-9]{1,3}"`
because there is no cut and ip commands. Note replace wlan0 with eth0 in case of Wi-Fi only.

2. The file /home/hardinfo.bin contains the line 
<IrCtrl>12_0x00000000_0_1</IrCtrl>
So I have updated  /media/hack/www/cgi-bin/webui (changed 46 -> 12)
 if [ "$command" = "iron" ]; then
  /bin/gio -s 12 1 > /dev/null
 fi
 if [ "$command" = "iroff" ]; then
  /bin/gio -s 12 0 > /dev/null
 fi

3. Please update ptz.md:
I have figured out the following commands
0x71 - see below
0x72 - see below
0x73 arg0 - setPSP (arg0) - store position preset#
0x74 arg0 - callPSP (arg0) - restore position preset#, arg0 must be 0..0x3F
0x74 arg0 - startScan(arg0) - scan, arg0 must be 0x40..0xFC
0x74 0xFD - resetPTZ(0)
0x74 0xFE - resetPTZ(1)
0x74 0xFF - resetPTZ(0)
0x75 arg0 - deletePSP(arg0) - delete position preset#
0x7C arg0 - startScan(0x43) - arg0 ignored
0x81            getRange(ptr)
0x82            getPosition(ptr)
0x84 arg0 - resetPTZ(0) - arg0 ignored
0x83 arg0 - gotoPos(ptr)
0x86 arg0 - setSpeed(ptr)
0x87 arg0 - getSpeed(ptr)
0x89 arg0 - setSavePos()
0x8A arg0 - resetPSP0(0, arg0)
0xFF arg0 - set_pps(arg0)
0x3EA arg0 - printPos()
0x3EC arg0 - switch_dbg(arg0)

There is an internal function driveMotor(direction, _unk2, speed, _unk4=0);
direction is a set of bit flags:
0x00 - stop
0x02 - right
0x04 - left
0x08 - up
0x10 - down
0x20 - command 0x71 (arg0 ignored, _unk4=1)
0x40 - command 0x72 (arg0 ignored, _unk4=1)

hawaltie, 2019-06-13:
> [Firmware for IL-HIP291G-2M-AI.zip](https://github.com/ant-thomas/zsgx1hacks/files/2848617/Firmware.for.IL-HIP291G-2M-AI.zip)

I also own an IL-HIP291G-2M-AI. I really want to close the cloud connection, but I do not understand the steps required for that. Is there some manual that explains the steps I should take?

slydiman, 2019-06-14:
Re: [ant-thomas/zsgx1hacks] Works on IL-HIP291G-2M-AI from INQMEGA (#90)

Hello

Unpack the attached file to the root of the SD card.
I have added the file /hack/hosts.new
and added the following line to the file debug_cmd.sh

mount --bind /media/hack/hosts.new /etc/hosts

Thanks,
Dmitry

Friday, June 14, 2019 12:16:03 AM You wrote:

Firmware for IL-HIP291G-2M-AI.zip

I also own an IL-HIP291G-2M-AI. I really want to close the cloud connection, but I do not understand the steps required for that. Is there some manual that explains the steps I should take?

ejalal, 2019-07-10:
Hi all,

Is there a chance for this to work on other cameras with same external design but from different brands ? I guess it's the same manufacturer (same hardware inside) but with different branding depending on each seller.

I got this one from Amazon as Mamicam brand https://www.amazon.fr/gp/product/B07P1MN2ZJ/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1, but can find plenty of them on Aliexpress using different brands.

ejalal, 2019-07-15:
Well I guess there's only one way to find out.

Catalin84, 2019-11-08:
Hi, I have a big problem. I also tried to get rid of CLOUD from the IP CAMERA and now I believe I was left with a brick.
When I turn on the camera (IL-HIP291G-2M-AI) it tells me I'm waiting for CONFIGURATION. After pressing 2 seconds on the reset button it tells me to enter AP mode. When I log on to the ip on the camera I do not receive any video images.
I got USB-UART, because I thought I would need it if I broke it. Can anyone help me with a solution? i tried to enter it with putty user: root and password cxlinux but it does not allow me to enter HOME. from what I have noticed I have no space on it anymore, and it does not let me delete that it tells me it is just for reading. 
[image]
[image]

Catalin84, 2019-11-08:
> > * это прошивка от моей камеры на GK7102,
> > * размер флеш моей камеры 8 мегабайт
> > * с этой прошивкой камера не будет подключаться к облаку
> > * [как прошивать описано там:](http://4pda.ru/forum/index.php?s=&showtopic=928641&view=findpost&p=81236013)
> > * [как подключать, описано там: ](http://4pda.ru/forum/index.php?s=&showtopic=928641&view=findpost&p=82237201)
> > 
> > [Firmware for IL-HIP291G-2M-AI.zip](https://github.com/ant-thomas/zsgx1hacks/files/2848617/Firmware.for.IL-HIP291G-2M-AI.zip)
> 
> How can I use this file to restore my bricked camera via sdcard?

Did you managed to fix it ? I have a same problem to restore from SD card but it tells that is READ ONLY the home folder.

paus56, 2019-11-08:
- у вас установлена модифицированная прошивка для камеры 
- установите Android приложение Ysee или SAP HD  и после фразы waiting for CONFIGURATION подключите камеру в приложении Ysee...
- подробнее об этом прочитайте там: http://4pda.ru/forum/index.php?s=&showtopic=928641&view=findpost&p=89274008

Catalin84, 2019-11-08:
> * у вас установлена ​​модифицированная прошивка для камеры
> * установите Android приложение Ysee или SAP HD și после фразы în așteptarea CONFIGURĂRII SUPODIMENTARE
> * подробнее об этом прочитайте там: http://4pda.ru/forum/index.php?s=&showtopic=928641&view=findpost&p=89274008

I have a problem. Can i delete some files from home ? 
# df -h | grep home
/dev/mtdblock4            3.4M      3.4M         0 100% /home
/dev/mmcblk0p1          252.0M      8.0M    244.0M   3% /home/hd1
/dev/mmcblk0p1          252.0M      8.0M    244.0M   3% /home/web/sd
#
Is read only ... :(

paus56, 2019-11-08:
нельзя удалять файлы из home... 
нужно полностью перепрошивать камеру оригинальной прошивкой из вашего бэкапа

Catalin84, 2019-11-08:
> нельзя удалять файлы из home...
> нужно полностью перепрошивать камеру оригинальной прошивкой из вашего бэкапа

Yes but the problem is i dont have backup. on the flash it say GK7102 . with what tool i can write from a backup that i found on internet?

paus56, 2019-11-08:
какую прошивку вы заливали в камеру перед этим ?

thiagosouza2000, 2019-11-08:
> * у вас установлена модифицированная прошивка для камеры
> * установите Android приложение Ysee или SAP HD  и после фразы waiting for CONFIGURATION подключите камеру в приложении Ysee...
> * подробнее об этом прочитайте там: http://4pda.ru/forum/index.php?s=&showtopic=928641&view=findpost&p=89274008

The ZIP files are not available anymore - error 404
Any ideia where to find them?

Google Translation:
ZIP-файлы больше не доступны - ошибка 404
Есть идеи, где их найти?

paus56, 2019-11-08:
надо зарегистрироваться и войти на 4PDA, тогда файлы будут доступны...

некоторые файлы есть там: https://github.com/ant-thomas/zsgx1hacks/issues/129

Catalin84, 2019-11-08:
> какую прошивку вы заливали в камеру перед этим ?

i dont remember. i just want to get back the video of camera. Command are working i can open the page local but i dont see video 
[image]

Catalin84, 2019-11-08:
> какую прошивку вы заливали в камеру перед этим ?

in version.ini i found this .    16.06.00.52
1569399530
inn firmware folder i found gk_fw.bin  i can open FTP but all are read only.

thiagosouza2000, 2019-11-08:
> надо зарегистрироваться и войти на 4PDA, тогда файлы будут доступны...
> 
> некоторые файлы есть там: #129

Thanks. I got the files, but it's not working. It returns an error on the SD card: Mod FW programming Unknown target.
Any ideias?

Благодарю. Я получил файлы, но это не работает. Возвращает ошибку на SD-карте: Мод FW программирования Неизвестная цель.
Есть идеи?

paus56, 2019-11-08:
>  Я могу открыть FTP, но все только для чтения.

прикрепите сюда файл  **run_init.sh**  из папки **home**

paus56, 2019-11-08:
> Thanks. I got the files, but it's not working. It returns an error on the SD card: Mod FW programming Unknown target.
> Any ideias?
> 

какой процессор в вашей камере ?  **gk7102**  или  **gk7102S**  ?

Catalin84, 2019-11-08:
> прикрепите сюда файл **run_init.sh** из папки **home**

mount /home/firmware /lib/firmware
export sensorType=gc2033

ko_dir=/home/ko
insmod $ko_dir/hi_reg.ko

mv /dev/random /dev/random.org
ln -s /dev/urandom /dev/random

sysctl -w fs.mqueue.msg_max=256

#2017-07-13 
echo 0 >  /proc/goke/cvbs_switch

mknod /dev/ppp c 108 0
mkdir /dev/mqueue
mount -t mqueue none /dev/mqueue

insmod $ko_dir/goke_gpio.ko

#mount SD card
if [ -b /dev/mmcblk0p1 ]; then
    mount -t vfat /dev/mmcblk0p1  /media
elif [ -b /dev/mmcblk0 ]; then
    mount -t vfat /dev/mmcblk0 /media
fi

# Firmware upgrade or replacement 
if [ -f /media/Hip_Mod_fw.img ];then
	echo "..... Starts Firmware upgrade ....."  
	/home/flashwriterFlash /media/Hip_Mod_fw.img
	mv /media/Hip_Mod_fw.img /media/Hip_Mod_fw.img.upd 
  	echo "..... Firmware upgrade Ok! ....."
	reboot
	sleep 1000 
fi

######### set system var #########
sysctl -w net.ipv4.tcp_mem='3072    4096    2000000' 
sysctl -w net.core.wmem_max='2000000'
sysctl -w net.ipv4.tcp_keepalive_time=60 net.ipv4.tcp_keepalive_intvl=6 net.ipv4.tcp_keepalive_probes=3 
sysctl -w net.ipv4.tcp_syncookies=1
sysctl -w net.ipv4.tcp_tw_reuse=1
sysctl -w net.ipv4.tcp_tw_recycle=1
sysctl -w net.ipv4.tcp_fin_timeout=10

sensorVal=$?
sensorType=gc2033

########## telnetd ######################
#telnetd &

########## lo up ######################
ifconfig lo 127.0.0.1

loadGkSensor()
{
	goke_lib_dir=/home/goke_lib_2.0
	sensor_config_file=$goke_lib_dir/gc2033.bin
	sensorConfigFileName=gc2033.bin
	sensor_hw_config_file=$goke_lib_dir/gc2033_hw.bin	
	tmp=`cat /data/config_info.ini | grep -c sensor_bw=12`
	if [ $tmp = 1 ];then
		sensor_hw_config_file=$goke_lib_dir/gc2033_hw_12bit.bin
	fi 

	# exit
	if [ -f $sensor_config_file ];then
		cp $sensor_config_file /etc/sensors/ -vf
	fi
	
	if [ -f $sensor_hw_config_file ];then
		cp $sensor_hw_config_file /etc/sensors/ -vf
	fi
	
    /home/sensor.sh 
}

loadWifiDrv()
{
	
	cd /usr/ko

	if [ -f rtl8188fu.ko ];then
		insmod rtl8188fu.ko
	fi
}

loadGkSensor

# mount sd card to separate location
if [ -f /media/mod/ipc ]; then
       echo "..... bind /media/mod/ipc ....."
       chmod +x /media/mod/ipc
       mount --bind /media/mod/ipc /home/ipc
fi

if [ -f /media/mod/hosts.new ]; then
       echo "..... bind /media/mod/hosts.new ....."
       mount --bind /media/mod/hosts.new /etc/hosts
fi

if [ -f /media/mod/web ]; then
       echo "..... bind /media/mod/web/ ....."
       mount --bind /media/mod/web/ /home/web/
fi

cd $ko_dir
insmod timer_drv.ko
insmod gpioi2c.ko

loadWifiDrv

######### load process #####
/home/watch &

# start ftp server
if [ -f "/data/ftp" ]; then
   (/bin/tcpsvd -E 0.0.0.0 21 /bin/ftpd -w / ) &
fi

#Run pre_init.sh
if [ -f "/media/pre_init.sh" ]; then
	echo "..... find pre init file, wait for cmd running ....."
    chmod +x /media/pre_init.sh
	/media/pre_init.sh
fi

#*************************************
if [ -f /data/pre_init.sh ];then
	echo "..... run pre init ....."
    chmod +x /data/pre_init.sh
	/data/pre_init.sh
fi

# config_info.ini: modifiable by customer
cfg_ini=`ls /media/*-config_info.ini`
if [ "$cfg_ini" != "" ]; then
  cp -f $cfg_ini /data/config_info.ini
  if [ $? -ne 0 ]; then
    echo "Copy config_info.ini failed." > /media/ERROR.txt
    exit;
  fi
fi 

echo "..... start camera ....."

cd /home/process
./process_init.sh

paus56, 2019-11-08:
у вас старая версия прошивки поддерживает только один сенсор
в новой прошивке есть другие сенсоры
- разархивируйте содержимое архива в корень SD карты
- подайте питание на камеру
- дождитесь загрузки камеры ( 2 минуты )
- настройте подключение к камере...
[Hip_Mod_fw.zip](https://github.com/ant-thomas/zsgx1hacks/files/3825246/Hip_Mod_fw.zip)

Catalin84, 2019-11-08:
> у вас старая версия прошивки поддерживает только один сенсор
> в новой прошивке есть другие сенсоры
>
> * разархивируйте содержимое архива в корень SD карты
> * подайте питание на камеру
> * дождитесь [...]

yes IS WORKING...
THANK YOU VERRY MUCH... THANK YOU

paus56, 2019-11-08:
> Thanks. I got the files, but it's not working. It returns an error on the SD card: Mod FW programming Unknown target.
> Any ideias?
>
> Благодарю. Я получил файлы, но это не работает. Возвращает ошибку [...]

для процессоров gk7102 и gk7102S  разные прошивки...

Catalin84, 2019-11-09:
> у вас старая версия прошивки поддерживает только один сенсор
> в новой прошивке есть другие сенсоры
>
> * разархивируйте содержимое архива в корень SD карты
> * подайте питание на камеру
> * дождитесь [...]

Hi again is there any new version of software/firmware? i have some big delays on camera when i try to control it. i am nex to camera and if i press back,up down i need to wait about 5 seconds.
[image] this is what i have now.

paus56, 2019-11-09:
сделайте настройки видео как  на картинке

[image]

Catalin84, 2019-11-09:
Done this , same lag. i think is the wifi they put inside. :) is there any application that i can use commands with rtsp? i cannot control the moves in rtsp
Best regards

thiagosouza2000, 2019-11-09:
> > Thanks. I got the files, but it's not working. It returns an error on the SD card: Mod FW programming Unknown target.
> > Any ideias?
> 
> какой процессор в вашей камере ? **gk7102** или **gk7102S** ?

Mine is gk7102S. After I tried the gk7102s firmware, the .sh is not executed anymore. I modified the script just to echo anything and it doesn't work. I think it's totally bricked. But thanks anyway.

Мой gk7102S. После того, как я попробовал прошивку gk7102s, .sh больше не выполняется. Я изменил скрипт, чтобы что-то повторить, и он не работает. Я думаю, что это полностью заложено. Но все равно спасибо.

Anon0ne, 2020-01-25:
Anyone else experienced the loss of wifi settings after several days? I have done portion of the hack, and setup blocking cloud info. But after couple days our WiFi settings are gone (the whole devParam.dat is gone from home folder) forcing me to reset or restore it.

aHVzY2g, 2020-01-25:
Yes I have experienced the same problem. But didn't really had time to look in to it so far.

Anon0ne, 2020-01-25:
I put mine up on the ceiling thinking all was good. Now I have to reset it every like 24-48hrs. I need to find out why it's removing connection randomly and deleting devParam.dat sometime. I can login via Ethernet and replace devParam it takes right off so it's definitely rebooting and removing that file.

I have verified it's not the auto ptz test script because I disabled it. Gotta keep looking I guess. Will post my findings

Anon0ne, 2020-02-03:
> Yes I have experienced the same problem. But didn't really had time to look in to it so far.

Managed to get rid of this problem by manually typing in a command I found in the hacks, it was something like "touch /etc/watchdog" and I haven't had a loss of settings since.

yomismo112, 2020-02-23:
Hi paus56, how have you got uncompress the original firmware from the camera? i tried unsquash it but don´t run.
Sorry for my bad english!!!

maartenjd, 2020-03-14:
> So anyone can easily dump the SPI FLASH by issuing commands:
>
> ```
> sf probe;
> sf read 0xc1000000 0x0 0x800000
> mmc init
> mmc parts
> fatwrite mmc 0:1 0xc1000000 partall.raw 0x800000
> ```
>
> it is [...]

Hello McGr3g0r,
My ipcam has the same pcb and memory layout as yours. Some information:

_U-Boot 2012.10 (Dec 26 2017 - 18:17:43) for GK7102 rb-sc1045-v2.0 (GOKE)
HAL:   20160913 
DRAM:  64 MiB
Flash: 8 MiB
NAND:  [No SPI nand] 
SD/MMC: 0
SF:    8 MiB [page:256 Bytes] [sector:64 KiB] [count:128] (XM25QH64)
In:    serial
Out:   serial
Err:   serial
Net:   Int PHY 
Hit any key to stop autoboot:  1 ... 0_

and

```
bootargs=console=ttySGK0,115200 mem=36M rootfstype=squashfs root=/dev/mtdblock2 init=linuxrc mtdparts=gk_flash:320K(U),1664K(K),1152K(R),2560K(A),-(H)
```
and
_[    0.000000] Memory: 36MB = 36MB total
[    0.000000] Memory: 31776k/31776k available, 5088k reserved, 0K highmem
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
[    0.000000]     DMA     : 0xff600000 - 0xffe00000   (   8 MB)
[    0.000000]     vmalloc : 0x82800000 - 0xff000000   (1992 MB)
[    0.000000]     lowmem  : 0x80000000 - 0x82400000   (  36 MB)
[    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
[    0.000000]       .text : 0x80008000 - 0x80412000   (4136 kB)
[    0.000000]       .init : 0x80412000 - 0x80433000   ( 132 kB)
[    0.000000]       .data : 0x80434000 - 0x8045bbe0   ( 159 kB)
[    0.000000]        .bss : 0x8045bc04 - 0x8048e1b8   ( 202 kB)_

After I had pressed the reset button for too long, my camera won't complete the start-up sequence anymore. Previously I already added the serial hack and made dumps of mtd0-4. Unfortunately I did not know this post, otherwise I would have made a dump the way you have shown above. Now I want to write the mtd-images from sd-card using the u-boot fatwrite command. I have to use u-boot because after booting into sh I have an empty /dev-directory so I cannot mount the mtd-blocks. Best I think is to restore mtd1-4, as mtd0 is u-boot and working fine.

Would it work to write "simply back to the addresses as they are showing in the bootlog?, to the R, A and H-regions to start with?:
[    0.640000] 0x000000000000-0x000000050000 : "U"
[    0.650000] 0x000000050000-0x0000001f0000 : "K"
[    0.660000] 0x0000001f0000-0x000000310000 : "R"
[    0.660000] 0x000000310000-0x000000590000 : "A"
[    0.670000] 0x000000590000-0x000000800000 : "H"
 
For example: the mtd2-chunk (A) it would then be:
```
mmc rescan
fatload mmc 0:1 0x???????? mtd2.bin 0x120000
sf probe
sf erase sf erase 0x1f0000 0x120000
sf write 0x???????? 0x1f0000 0x120000
```
My question now is: at what address should I load this file in RAM? What starting address to put in the place of the ????????. In one post I read that 0x80008000. Another says to use the lowmem-area (starting at 0x80000000).Your advise much appreciated. 
Maarten

maartenjd, 2020-03-16:
Ref my previous post: I managed to resolve it and to restore the original mtd-partitions (1=K,2=R, 3=A and 4=H). Instead of mmc from a SD-card (I could fatwrire to it but fatread gave a partition table error) I switched to using kermit transfers via serial.

Make sure your camera is connected via serial. Start kermit in terminal mode and connect to the camera. Power up camera and in terminal press enter within 5 seconds to enter u-boot environment.

In u-boot issue the loadb-command ; the camera then listens on the serial port:
`loadb`
_## Ready for binary (kermit) download to 0xC1000000 at 115200 bps...._ 

You notice that loadb will load the received file in ram starting at 0xc10000000. By the way: that is also the address where u-boot loads the kernel before handing over control to the kernel, so apparently this section of ram is free to use and not occupied by u-boot processes or programs.

Then switch kermit to command mode and send over the mtd-image, e.g: **mtd3.bin**
C-Kermit>`send mtd3.bin`

Kermit will show you a progress window and should finish transfer without errors.
 
Switch kermit back to terminal mode. loadb finishes when transfer complete as follows:
_## Total Size      = 0x00280000 = 2621440 Bytes_
_## Start Addr      = 0xC1000000_

Then in u-boot environment issue:
`sf probe`
_SF:    8 MiB [page:256 Bytes] [sector:64 KiB] [count:128] (XM25QH64)_
`sf erase 0x310000 0x280000`
_[PROCESS_SEPARATORS] sf erase 0x310000 0x280000
Erasing:
    100%_

`sf write 0xc1000000 0x310000 0x280000`
_[PROCESS_SEPARATORS] sf write 0xc1000000 0x310000 0x280000
Programming:
    100%
Verifying:
    100%_
 
To ease the process for when I mess up another time, I concatenated the mtd1, mtd2, mtd3 and mtd4 partitions into one file, so that I can restore the camera with only one file transfer, one erase action and one write action (from 0x50000-0x800000).

I was lucky to have made copies of the mtd-partitions before I started experimenting! See [Paus56](https://github.com/ant-thomas/zsgx1hacks/issues/90?fbclid=IwAR1GtZzmwI56CfEjLsoep6IMWPgipMGhb0vz4BT77tOjrdrV4obJlYmkEns#issuecomment-449437159) post

Catalin84, 2020-03-17:
> touch /etc/watchdog

Can you get us more details? I have the same problem. 
On what console you type?

Anon0ne, 2020-03-17:
> > touch /etc/watchdog
> 
> Can you get us more details? I have the same problem.
> On what console you type?

Look for that command in the start script of the modded firmware. Haven't had a reset since I did this.