ant-thomas/zsgx1hacks #92: SDK Download

alesimoes, 2018-12-11:
Hi, where can I download the goke SDK?

paus56, 2019-01-10:
http://en.pudn.com/Download/item/id/3294300.html

dst0815, 2019-01-13:
You are right, deorder. The file doesnt contain any kernel code, just informations in mandarin, mainly for RTOS, not linux. I uploaded the file to my google drive, its at https://drive.google.com/uc?id=1gMJqD-cdlKxbfHqc8gLws2wpt1hfNs-p&export=download . Also, if anyone needs the toolchain for other platforms, they are at https://launchpad.net/gcc-arm-embedded/4.9/4.9-2015-q1-update . Best regards, John

deorder, 2019-01-13:
Thank you. There is a SDK that is reported to work with the Goke as well. Have not tested it yet. You can find the download at the bottom of: https://github.com/Filipowicz251/mijia-1080P-hacks/wiki

paus56, 2019-07-11:
**[GK710X Linux SDK](https://github.com/ant-thomas/zsgx1hacks/issues/124)**