ant-thomas/zsgx1hacks #94: Managed to disable p2pcam and use another RTSP server with a GM8136/SC1135 camera

CarlosML, 2018-12-24:
Excuse me in advance for posting this as an issue, this repository has been very helpful for me and I want to share this information. I managed to disable p2pcam and use another RTSP server with a GM8136/SC1135 camera, although the SoC is not the same as the ZSGX1 and the sensor is an inferior model the filesystem layout is pretty much the same. The first problem I had was that killin p2pcam triggered the watchdog, so I had to disable it using:

`echo 0 > /proc/watchdog/watchdog_enable`

Then, I found the demo RTSP server from GM8136 SDK here:

https://github.com/willthrom/rtspServer_gm/releases

Killing p2pcam disables the cloud services and I can stream the video feed using the RTSP demo server and change the video bitrate (p2pcam was awfully low). Bsesides, I have a file called "ptz_test" for controlling the PTZ commands.

ant-thomas, 2018-12-24:
I must admit I've not put much effort into this hack lately as I felt I couldn't add much more. But this is a very big find that I'll look to implement very soon.

Does it give an onvif compatible stream or just RTSP?

CarlosML, 2018-12-24:
> Does it give an onvif compatible stream or just RTSP?

Just RTSP, it's a barebones server that comes with the SDK, but at least there is source code available.

deorder, 2019-01-02:
Source code available? I actually looked for this and even tried to reverse engineer the hardware without luck. Good find tnx.

roleoroleo, 2019-01-06:
> Does it give an onvif compatible stream or just RTSP?

I used onvif_srvd with a ezviz camera and it works very well.

cspwcspw, 2019-06-17:
I did some work which I report on at https://github.com/cspwcspw/CloudCameraGK7102 

Perhaps the most interesting aspect is that my cloud camera has both Ethernet and WiFi.  So by ensuring that the Ethernet is attached (or we can fool the camera with an Ethernet Loopback Plug), p2pcam won't bring up the wireless and runs in some *__in-lan__* mode that makes it pretty usable without the cloud.  We can then bring the wireless up manually using our own configuration and use it independently of its attempt to go via the cloud.  

Peter