ant-thomas/zsgx1hacks #95: Found a firmware for guudgo gd sc03

nykroy, 2019-01-23:
I found a firmware of 2017/08. Had someone else probed it before?

https://www60.zippyshare.com/v/2GCs0wEa/file.html

Someone published the link here: 
https://www.mydealz.de/deals/banggood-guudgo-gd-sc03-snowman-1080p-wifi-rj-45-ip-camera-pantilt-1035363#thread-comments

nykroy, 2019-01-23:
@ant-thomas  thank you for your work!!!

Gymnae, 2019-01-29:
Be careful, it might brick your GD-SC03. Do you have a newer one currently running?

Gymnae, 2019-01-29:
Could you do me a favor and check if anyone of those has 8MB Flash size and how big the /home folder is? I bricked my camera with the linked firmware and try to get it back to a working state...

nykroy, 2019-01-29:
Coud you explain me what do you mean with:
`check if anyone of those has 8MB Flash size `
The size of home:
[image]
I have atached the home directory too
[home.zip](https://github.com/ant-thomas/zsgx1hacks/files/2809958/home.zip)
Told me how can i help you with the flash size

Gymnae, 2019-01-30:
> Coud you explain me what do you mean with:
> `check if anyone of those has 8MB Flash size `
> The size of home:
> [image]
> I have atached the home directory too
> [home.zip](https://github.com/ant-thomas/zsgx1hacks/files/2809958/home.zip)
> Told me how can i help you with the flash size

Thank you for attaching the home directory :)
The problem I have: My overall flash space seems to be lower than with other cameras, due to having a different flash chip built in: [GD25Q64C](http://www.elm-tech.com/ja/products/spi-flash-memory/gd25q64/gd25q64.pdf)

```
DRAM:  64 MiB
Flash: 8 MiB
```

Whereas other dumps list a bigger Flash, the following is from the [bootdump](https://github.com/ant-thomas/zsgx1hacks/blob/master/bootdump) found in this repo.
```
DRAM:  64 MiB
Flash: 16 MiB
```

Thus the /home folders with nearly 8MB in size won't fit on my camera.

nykroy, 2019-01-30:
i am sorry but my two cameras has near to 8 mb :(

Gymnae, 2019-02-12:
Dear @purduephotog, were you able to grab your camera? A dump of your mtd blocks would help me, since I now ordered a flash programmer in an attempt to salvage that little camera.

Gymnae, 2019-03-15:
I solved my problem :) See [here](https://github.com/ant-thomas/zsgx1hacks/issues/96#issuecomment-473398375)