ant-thomas/zsgx1hacks #100: Newer firmware - read-only reversible?

josvink66, 2019-01-31:
Hello,

I have a simple question. Is the Newer firmware - read-only reversible?
I want to try this firmware on my IL-HIP291G-2M-AI.

My firmware version is: 3.4.1.1212
So i assume i need the read only version.

Thanks,
Jos Vink

paus56, 2019-01-31:
Yes,  firmware - read-only, Tosc still there:   https://github.com/ant-thomas/zsgx1hacks/issues/90
Can I ask you for a file **p2pcam** from your camera?

[There is about this camera there: ](http://4pda.ru/forum/index.php?showtopic=928641)

josvink66, 2019-01-31:
Yeah i found that topic, so if i insert the SD card i will be running the hacked firmware. But if I remove the SD card i will be back at the original firmware? Is this correct?

paus56, 2019-02-01:
To return to the original firmware, just remove the SD card, you do not need to do anything else.

CSab80, 2019-04-30:
@josvink66 
> My firmware version is: 3.4.1.1212
Hi. Are you able to make un dump (or a copy of each of yours mtdx files located at /dev), please ?

@paus56 
Where can we download your tool to create firmware from mtd files ? 4pda.ru displays 404...

Regards,

CSab80, 2019-04-30:
@josvink66 If you want/can, I also interrested by your file **p2pcam** .
Best regards,
CSab80

josvink66, 2019-04-30:
@CSab80 Hello, currently i'm also trying to replace my p2pcam file so it can work with my DVR (QNAP) https://4pda.ru/forum/index.php?showtopic=928641&st=260#entry82569068

Unfortunately i'm also running in to the  issue of getting the 404 when I try to download the patch.

@paus56 I'm also interested on how to download from 4pda.ru so I can download the patch https://4pda.ru/forum/index.php?showtopic=928641&st=260#entry82569068

josvink66, 2019-04-30:
@CSab80 @paus56 I've managed to download my file. You need to register on the website. Witch is quite difficult with the russian Captcha.

CSab80, 2019-04-30:
@paus56 Hi Paus56. I'm french and we can't register on 4pda.ru, we are not able to solve the russian captcha. Could you use another site (here, Dropbox or GoogleDrive) to share you works, please ?

@josvink66 Hi Josvink66, I've got the same problem with SecuritySpy, a NVR software for Mac. After few chats with its developpers, it seems that p2pcam is not fully ONVIF compliant.

Regards,
CSab80

CSab80, 2019-04-30:
@josvink66 Hi again, I succeed to solve the captcha with this help : https://doorsgeek.blogspot.com/2015/08/4pdaru-loginregister-captcha-tutorial.html

hawaltie, 2019-06-13:
Hi Jos,

Is there a step by step manual for running this firmware on my sd card on the L-HIP291G-2M-AI with firmware version: 3.4.1.1212? I tried to figure out how to do this, but I can't get a complete picture.

Regards,
Hans.

josvink66, 2019-06-13:
Hi Hans.

I don't have a step by step instruction. But it's prety simple.
You just need to download the readonly zip: https://github.com/ant-thomas/zsgx1hacks/raw/master/readonlyhack-v0.1.zip

Extract it and put the content inside of the folder "readonlysd" on the root of the SD card.

Then put the SD card in the camera and power it on.

Greetings,
Jos

hawaltie, 2019-06-13:
Hi Jos,

Thanks for your quick response. I thought I had to use firmware from Paus56. But now it works.

By the way, It would be nice if we could integrate the functionality of https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks/blob/master/integration/homeassistant/homeassistant.md to make it easier to integrate in home assistant

josvink66, 2019-06-13:
Hi Hans,

Yeah would be nice. I'm still trying to connect this camera to my QNAP surveillance station. At this moment the camera just reboots after a few seconds. So that sucks..

kamilmirza, 2019-07-26:
@paus56 here's the p2pcam v3.4.1.1212 file you requested 

[p2pcam_3.4.1.1212.zip](https://github.com/ant-thomas/zsgx1hacks/files/3436578/p2pcam_3.4.1.1212.zip)

I have also this version:
3.4.1.1212
SDK10823
gc1034

hawaltie, 2019-07-26:
@jos

Did you get the camara working on the qnap? I had no problem connecting the camera to synology's surveillance station.

josvink66, 2019-07-26:
@Hawaltie

Unfortunately I still cannot get it to work properly. I have managed to get it connecting but without ptz control and after a few seconds the camera reboots

djansen1987, 2020-04-10:
@josvink66 Hi jos are you avalible to help ?