ant-thomas/zsgx1hacks #101: Edit hack for custom pan/tilt?

techspy2000, 2019-01-31:
Is it possible to edit any code of this before putting on the SD card that will allow a pan/tilt protocol from another camera to control the pan/tilt? As an example, the webui works with this camera. I can pan/tilt. However I am not able to find any compatible RTSP model in my app (IPCamViewer) that allows pan/tilt to work. There are several models that lets you specify the RTSP port and a separate data or http port, of which I assume one or the other is for pan/tilt comms, but none of them work with the pan/tilt using this hack. So if I identify the pan/tilt commands for one of the models in the app, how can I implement that into the code that goes on the SD card?
Thanks

gd-99, 2019-03-09:
Not sure if my reply to [#104](https://github.com/ant-thomas/zsgx1hacks/issues/104#issuecomment-471187234)  is any help to you?

techspy2000, 2019-03-09:
That worked! Thanks!