ant-thomas/zsgx1hacks #102: Help needed on GD-SC01

raas78, 2019-01-31:
Hi there, 

glad to have found this site, to have at least some 'support' and 'options' through a customized firmware. 
The troubles I'm having:

my father bought 4* GD-SC-01 cameras and 1* GD-NR-01 network video recorder. 
All cameras are visible by the NVR (after a 'soft reset' on a pc using ONVIF Device Manager), however, the cameras have the strange behaviour to somehow 'reboot' themself within 24-48 hrs after they were first powered on. 

The NVR then reports a videoloss, because it can't use the stream.
After using a pc using ONVIF Device Manager and 'soft reset' every one of the 4 cameras, the NVR picks up the signal again and starts displaying/ recording. 

Now what I know is... 3 out of 4 cameras experience this behaviour, 1 is staying 'online'. 
What my father did with that one was 'updating the firmware of the camera' using his phone. 
Well I thought, that might be the solution, but my dad has lost all 4 QR codes, so we don't have any serialnumber to add the cameras to the XMEye apps.

So, lot of context. 
2 main questions. 

- can this zsgx1hack be adapted that the camera doesn't reboot itself anymore AND that the onvif 'soft reset' isn't needed so that the NVR picks up the signal in case it does. 
- can I somehow via telnet/ssh obtain the cameras ID / Serialnumber to have them add in the XMEye app to update the firmware. 

Not afraid to open these devices up or reprogram, if you need more information, just let me know. 

hope someone is able to shed some light on these strange issues, if this wouldn't occur, it's an ideal system for the price, 4 cameras and 1 NVR for less that $150

Thanks!

Raas78

paus56, 2019-02-01:
This project discusses a hack for a CLOUD Wi-Fi camera: Guudgo GD-SC01 on a GK7102 processor . 
The application for such a camera is GUUDGO, but not XMEye.

Ilovegym66, 2019-02-01:
Hey, I've installed the same camera with Eyeplus, select +, setup wifi, and select qr-code, hold the phone 10cm before the camera, wait... after 30-50 seconds the camera is in the app.

raas78, 2019-02-01:
thanks both @paus56 and @ilovegym66 for your replies. 
Eyeplus was the application I was looking for. 
Unfortunately I don't have the QR codes anymore, so I'm looking for a way around that.. .

I've been thinking by the way... the only thing I need for my application (and that is that the cameras keep sthowing up in the NVR, not that we have to manually initiate a onvif soft reset).. 
Is the soft reset command.. 

Does anyone happen to know what this command is, and If I can put in in the sd-card. so that it is executed each and every time the camer is rebooted.

Thanks!

paus56, 2019-02-01:
- For this camera, you do not need an QR code from the sticker, register in the program EYEplus, and the phone itself will issue an QR code.
- What is the firmware version of your camera?

raas78, 2019-02-02:
The firmware reported by ONVIF Device Manager is: 3.2.8.0121
Def. going to try the EYEplus thing.
Thanks again! I'm sure we'll finally get there.

Thanks.

raas78

raas78, 2019-02-02:
ok. the Eyeplus thing worked.... AFTER I REMOVED THE NETWORK CABLE... stupid me.. it was already on the network, therefor the Eyeplus app couldn't find it.. now it is registered on my WLAN.

it is running without the SD card at the moment, but when I browsed around through FTP when the card was in I noticed (and has been documented) that the hostfile is filled with a lot of entries to prevent the camera communicating with the cloud. 

I have registered the camera in the eyeplus application AFTER I had applied the hack, I am a bit surprised to see it at https://eyeplus.closeli.com/device/live?deviceId=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

Not that I worry to much about it. but thought let you guys know. 

So now I have the camera running (still needed to apply the soft-reset to make the stream visible in the NVR), now lets see if it survives 24-48hrs without a reboot.. 
Could be, because it wasn't registered through the app, that it rebooted every 24-48hrs..

Let's lee, Keep you guys posted, thanks for all the help and input!

Cheers!

raas78, 2019-02-03:
Just to let you guys know, the cam 'survived' the 1st 24hrs and it didn't reboot!
So there's still a stream available on the NVR!
Thumbs up for the 2nd 24 hrs.. if it 'survives' that I'm pretty sure that it is solid. 

Strange though that for a pure local solution, 4 cams and a NVR the cameras still need to be 'registered' through the app. 

From what I gathered on the internet, people claim that these cameras suck, and don't work with the NVR, perhaps this is the solution, 1st register through Eyeplus app. 

I'm not sure though yet if this sd-card hack is still needed for MY particular application, but I still need 3 cameras to configure, so going to first try it out without the sd card hack and see if it works. 

Kee you posted. 

Ohw.. btw.. I have one lying in bits in pieces on my desk, see if i could find serial port for programming before I found this page. Will post them someday so they can be used for reference.

raas78, 2019-02-04:
> I have registered the camera in the eyeplus application AFTER I had applied the hack, I am a bit surprised to see it at https://eyeplus.closeli.com/device/live?deviceId=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
> 

I think this was a mistake on my part. the sd card wasn't in. 
now it's in and it is not registering on the website. 
thought that the hostfile would be overwritten in the camera's flash itself, therefor I had the card out.
card now in, but not registered on website or in the eyeplus app (which I now believe runs through the website, not local)

need to get the hang of these things to see what's going on. 
Can't get my head around it why they reboot/might even reset every 24/48hrs. 
(tin foil hat on) almost start to thing that it's done deliberately to force people into plans, but then why would they offer a local NVR for which no service is needed (tin foil hat of)

See if I can dig up an old hub somewhere, then wireshark it, see what's going on. 

(Hope no-one minds this thread is about to be a log-book of my trials and errors...)

raas78, 2019-02-06:
still no go, rebooted this morning.
going to examine if I can add the 'soft-reset' to the sd-card starting up.

raas78, 2019-02-06:
All right.. 

With some help from the documented [SOAP interface](https://github.com/ant-thomas/zsgx1hacks/issues/80)  and the use of wireshare I found the content to post to localhost/onvif/device

`

    <s:Envelope
        xmlns:s="http://www.w3.org/2003/05/soap-envelope">
        <s:Header>
            <Security
                s:mustUnderstand="1"
                xmlns="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-secext-1.0.xsd">
                <UsernameToken>
                    <Username>
                        admin
                        </Username>
                    <Password
                        Type="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-username-token-profile-1.0#PasswordDigest">
                        6yCRMPbVNvBsvpqIQql1Zp4RPzY=
                        </Password>
                    <Nonce
                        EncodingType="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-soap-message-security-1.0#Base64Binary">
                        zOLJ7cRm1kiL4f74uG7bVQ4AAAAAAA==
                        </Nonce>
                    <Created
                        xmlns="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-utility-1.0.xsd">
                        2019-02-06T19:21:10.552Z
                        </Created>
                    </UsernameToken>
                </Security>
            </s:Header>
        <s:Body
            xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
            xmlns:xsd="http://www.w3.org/2001/XMLSchema">
            <SetSystemFactoryDefault
                xmlns="http://www.onvif.org/ver10/device/wsdl">
                <FactoryDefault>
                    Soft
                    </FactoryDefault>
                </SetSystemFactoryDefault>
            </s:Body>
        </s:Envelope>
`

now, how to get it in the sd card starup.. 

curl?

raas78, 2019-02-06:
when I tried my first attempt to put a curl command with an envelope send, I failed. 

but I noticed some log files on the sd card:
as you can see, some STATUS_NOACCOUNT messages. and some purchasing server messages. This is with it being completely shutoff from the internet. 
Bastards. do they fail the cameras on purpose without subscribtion. 
What can we disable to overcome this. 

`
---------------------------- Dump of /tmp/closelicamera.log ---------------------------
2019-02-07 02:04:57.878229 [195] [IotCloud.cpp(48) SetParams]Trace: tracepoint
2019-02-07 02:04:57.879289 [195] [IotCloudImpl.cpp(66) SetTransmitParams]Trace: timeout(30)
2019-02-07 02:04:57.890819 [195] [IotParam.cpp(46) SetParams]Trace: tracepoint
2019-02-07 02:04:57.891839 [195] [IotParam.cpp(112) LoadDefaultConfiguration]Trace: cfg path: /home/iot.ini
2019-02-07 02:04:57.892922 [195] [IotParam.cpp(115) LoadDefaultConfiguration]Warn: not set path or file not exit!
2019-02-07 02:04:57.893830 [195] [cc_gateway.cpp(144) CloseliGateway_Configure]Trace: succ(1), err(0x0)
2019-02-07 02:04:57.897690 [195] [cc_gateway.cpp(155) CloseliGateway_Start]Trace: tracepoint
2019-02-07 02:04:57.898639 [195] [IotCloud.cpp(59) Start]Trace: tracepoint
2019-02-07 02:04:57.899431 [195] [IotCloudImpl.cpp(44) Start]Trace: tracepoint
2019-02-07 02:04:57.931370 [195] [cc_gateway.cpp(161) CloseliGateway_Start]Trace: succ(1), err(0x0)
2019-02-07 02:04:57.932886 [234] [LoopThread.cpp(106) ThreadProc]Trace: iot[234] thread begin >>>
2019-02-07 02:04:57.933982 [234] [IotCloudImpl.cpp(340) ThreadProc]Trace: Begin__
2019-02-07 02:04:57.937987 [234] [IotCloudImpl.cpp(78) DetectEquipmentEffective]Warn: username null or not valid!
2019-02-07 02:04:57.947050 [234] [IotCloudImpl.cpp(84) DetectEquipmentEffective]Trace: res(0)
2019-02-07 02:04:57.948145 [234] [IotCloudImpl.cpp(344) ThreadProc]Trace: start to pause thread!
2019-02-07 02:04:57.954183 [195] [CameraApp.cpp(172) LoopRun]Trace: CCameraApp::loop(): entering loop!
2019-02-07 02:04:57.955398 [195] [CameraApp.cpp(175) LoopRun]Trace: tracepoint
2019-02-07 02:04:57.956247 [195] [DeviceBox.cpp(64) Start]Trace: start
2019-02-07 02:04:57.971360 [195] [VideoInput.cpp(81) Start]Trace: tracepoint
2019-02-07 02:04:57.973169 [195] [VideoEncoder.cpp(82) CVideoEncoder]Trace: CVideoEncoder::ctor() >>>
2019-02-07 02:04:57.977927 [195] [VideoEncoder.cpp(107) Start]Trace: CVideoEncoder::Start() >>>
2019-02-07 02:04:57.987451 [235] [OpenCore.cpp(390) IsRegistered]Trace: username()
2019-02-07 02:04:57.979134 [195] [AudioEncoder.cpp(89) Start]Trace: tracepoint
2019-02-07 02:04:57.990777 [195] [Speaker.cpp(60) Start]Trace: tracepoint
2019-02-07 02:04:57.991749 [195] [VideoDetect.cpp(80) Start]Trace: tracepoint
2019-02-07 02:04:57.992581 [195] [AudioDetect.cpp(46) CAudioDetect]Trace: tracepoint
2019-02-07 02:04:57.993570 [195] [AudioDetect.cpp(75) Start]Trace: tracepoint
2019-02-07 02:04:57.994995 [195] [Ptz.cpp(37) Start]Trace: CPTZ::Start() >>>
2019-02-07 02:04:57.995911 [195] [CameraApp.cpp(177) LoopRun]Trace: tracepoint
2019-02-07 02:04:57.997091 [195] [ConfigManager.cpp(150) CConfigManager]Trace: tracepoint
2019-02-07 02:04:57.998146 [195] [ConfigManager.cpp(161) Start]Trace: tracepoint
2019-02-07 02:04:57.999166 [195] [ConfigManager.cpp(209) LoadFromFile]Trace: app path(/tmp) config path(/home)
2019-02-07 02:04:58.001997 [195] [JsonIO.cpp(15) LoadJsonFromFile]Warn: open configure file failed! filename(/tmp/config.default.json)
2019-02-07 02:04:58.005994 [195] [CameraApp.cpp(179) LoopRun]Trace: tracepoint
2019-02-07 02:04:58.007210 [195] [UserManager.cpp(24) CUserManager]Trace: tracepoint
2019-02-07 02:04:58.008234 [195] [UserManager.cpp(34) Start]Trace: tracepoint
2019-02-07 02:04:58.009924 [236] [LoopThread.cpp(106) ThreadProc]Trace: ConfigSaver[236] thread begin >>>
2019-02-07 02:04:58.011229 [236] [ConfigManager.cpp(176) ThreadProc]Trace: thread begin__
2019-02-07 02:04:58.009077 [195] [UserManager.cpp(88) loadUsers]Trace: tracepoint
2019-02-07 02:04:58.012646 [195] [UserManager.cpp(68) checkUserExisted]Trace: tracepoint
2019-02-07 02:04:58.013447 [195] [UserManager.cpp(95) loadUsers]Trace: load users configure failed! try to load from users.default.json!
2019-02-07 02:04:58.017625 [195] [JsonIO.cpp(15) LoadJsonFromFile]Warn: open configure file failed! filename(/tmp/users.default.json)
2019-02-07 02:04:58.018704 [195] [UserManager.cpp(68) checkUserExisted]Trace: tracepoint
2019-02-07 02:04:58.020070 [195] [UserManager.cpp(102) loadUsers]Error: no default users!
2019-02-07 02:04:58.021081 [195] [CameraApp.cpp(181) LoopRun]Trace: tracepoint
2019-02-07 02:04:58.022231 [195] [DetectManager.cpp(390) CDetectManager]Trace: tracepoint
2019-02-07 02:04:58.023235 [195] [DetectManager.cpp(448) Start]Trace: tracepoint
2019-02-07 02:04:58.742675 [195] [EventFilter.cpp(86) Stop]Trace: thread have stopped!
2019-02-07 02:04:58.743734 [195] [CameraApp.cpp(184) LoopRun]Trace: tracepoint
2019-02-07 02:04:58.744784 [195] [WorkLogManager.cpp(129) SetWorkLogPath]Trace: Set work log file /home/work.log:/home/idx.log start!
2019-02-07 02:04:58.748937 [195] [WorkLogManager.cpp(89) GetCheckIdx]Trace: Get idx array [27201 0 0]
2019-02-07 02:04:58.780220 [195] [WorkLogManager.cpp(208) SetWorkLogPath]Trace: Set work log file stop, ret=1, 27283B, version is WorkLog_1.1

2019-02-07 02:04:58.781470 [195] [CameraApp.cpp(188) LoopRun]Trace: tracepoint
2019-02-07 02:04:58.782489 [195] [CameraApp.cpp(192) LoopRun]Trace: tracepoint
[... cut ...]
verifyUserPasswdSimple user  passwd 
the username match!!!
verifyUserPasswdSimple user admin passwd 
the username match!!!
=====>verifyUserPasswdSimple the password is authentication failed default passwd gd20160404!!!!
verifyUserPasswdSimple user  passwd 
the username match!!!
verifyUserPasswdSimple user admin passwd 
the username match!!!
=====>verifyUserPasswdSimple the password is authentication failed default passwd gd20160404!!!!
verifyUserPasswdSimple user  passwd 
the username match!!!
`

raas78, 2019-02-07:
While I had expected that the camera had crashed today, it hadn't. So it's over 24hrs now. let's see if it can surpass 48 hrs. 

Current situation: running of the sd-card and in router/firewall all access from and to the internet on this device is prohibited.

If this would surpass 48 hrs. I think I just have to find out to what cloudservices it more wants to communicate than @ant-thomas  has already exluced with his host file.

raas78, 2019-02-08:
still running, 36hrs exceeded

raas78, 2019-02-08:
and it just crapped out before reaching 48hrs.
So running of SD card and NO connection to the internet still doesn't fix it. 

I think my best guess is with the soft-reset command which I need to implement on reboot.

raas78, 2019-02-09:
Ok, got a bit further, but not quitte where I want it to be, I need some help on CURL on the Device itself.

With wireshark I fetched the command to soft-reset the camera. I put this in a text file: envelope2.txt

`<s:Envelope xmlns:s="http://www.w3.org/2003/05/soap-envelope"><s:Header><Security s:mustUnderstand="1" xmlns="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-secext-1.0.xsd"><UsernameToken><Username>admin</Username><Password Type="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-username-token-profile-1.0#PasswordDigest">GAGEAPb5bV7jH9qUF/xZ7U75mMw=</Password><Nonce EncodingType="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-soap-message-security-1.0#Base64Binary">/utB99pDLk6UPIGCifVKJBQAAAAAAA==</Nonce><Created xmlns="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-utility-1.0.xsd">2019-02-09T18:34:11.661Z</Created></UsernameToken></Security></s:Header><s:Body xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema"><SetSystemFactoryDefault xmlns="http://www.onvif.org/ver10/device/wsdl"><FactoryDefault>Soft</FactoryDefault></SetSystemFactoryDefault></s:Body></s:Envelope>`

on my desktop (win10 +curl for windows) I run this command:

`curl --header "Content-Type: text/xml;charset=UTF-8" --header "SOAPAction:urn:my:service-interaction:soap:soft-reset" --data @envelope2.txt http://10.0.0.58/onvif/device`

This works! it immediately 'soft resets' the camera and the NVR directly captures and records the stream.

Now I'd like to attach this command  to the end of @ant-thomas  its script by doing this:

`# raas78
# calls the ONVIF Soft Reset command for GUUDGO GD-SC-01, which doesn't init on reboot. and reboots after 24-48hrs.
(sleep 60 && curl --header "Content-Type: text/xml;charset=UTF-8" --header "SOAPAction:urn:my:service-interaction:soap:soft-reset" --data @/media/envelope2.txt http://localhost/onvif/device ) &`
As you can see, I directed the path to the textfile to the sdcard and changed the ip with localhost.
but it doen't work.

Any idea's?

Many thanks!

raas78, 2019-02-09:
so I guess curl is not on the camera....

https://github.com/ant-thomas/zsgx1hacks/issues/99

raas78, 2019-02-09:
So..... in our country we have a phrase:

> Als Mozes niet naar de berg komt, dan moet de berg maar naar Mozes komen

freely translated into: 

> If you can't go the way you should, then you should go any way you can

I have instructed my qnap nas to run a sh every minute through crontab with the content above. Works like a charm. Only thing is I'll loose somewhere between 0:30m and 1:30m of footage once every 24/48hrs every time the camera reboots. 

But it doesn't matter to the camera, nor the dvr that the camera gets a soft reset every minute.

Now, I have the identical nas at my parents house as at my house (fiber connected, 2 way instant replica).. So if the script works on my network, it'll work on theirs too.

This will work withouth the sd card hack or disconnection from internet. it's just issueing the soft-reset command every minute which the camera needs after a reboot, initiated with manufacturers intent.
**So. NO fix, but a watchdog/pacemaker on a different always on device on the same network. pretty sure you can turn your router into something that can issue this.**

roleoroleo, 2019-05-02:
If it can help I cross-compiled curl for GD-SC03:
https://repository.ilpuntotecnico.com/files/roleo/public/guudgo_gd-sc03/curl.tgz

david-gv, 2019-08-21:
Hello, 
I can confirm the camera starts failing after the cloud-subscription has finished, and months later it stops working within minutes after conneting it to internet.

I solved this with two files:

/home/hosts.new
`127.0.0.1       localhost
127.0.1.1       goke

127.0.0.1 arcsoft.com
127.0.0.1 xmpp.icloseli.cn
127.0.0.1 relaycn.arcsoftcloud.com
127.0.0.1 update.icloseli.cn

127.0.0.1 esd.icloseli.cn
127.0.0.1 upns.icloseli.cn

127.0.0.1 argus.icloseli.cn
127.0.0.1 relay.icloseli.cn
127.0.0.1 stun.icloseli.cn
127.0.0.1 api.icloseli.cn

127.0.0.1 xmpp.icloseli.com
127.0.0.1 relayus-w.arcsoftcloud.com
127.0.0.1 update.icloseli.com
127.0.0.1 esd.icloseli.com
127.0.0.1 upns.icloseli.com

127.0.0.1 xmpp.icloseli.com
127.0.0.1 argus.icloseli.com

127.0.0.1 relay.icloseli.com
127.0.0.1 api.icloseli.com`

/home/custom_init.sh
`#!/bin/sh

mount -t vfat /dev/mmcblk0p1  /mnt
sh /mnt/debug_cmd.sh

mount --bind /home/hosts.new /etc/hosts`

No other changes. This has been working for me for several months now.